fritz-smh/yi-hack #113: Problem with 12CNxx160418

dmolner, 2016-12-15:
Hello,

I test this hack in some cameras, but now I bought another more and not work.
The seller say the camera is dowgrade to M firmware.
I test downgrade to another firmware but I can't.
The serial number is 12CNYFXXXXXXXX160418

The log.txt file when I put a SD in the camera is this:

```
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=255  (0 is ok)
Wifi configuration answer: Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
Could not read interface ra0 flags: No such device
WEXT: Could not set interface 'ra0' UP
ra0: Failed to initialize driver interface
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 01:00:17 CET 1970
Done
New datetime is Thu Jan  1 01:00:17 CET 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1193 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1220 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1305 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  680 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  945 root         0 SW   [hidog]
  983 root      1164 S    ./peripheral
  984 root      1144 S <  ./dispatch
  985 root      1068 S    ./exnet
 1193 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1220 root      1408 S    ./serverM 80
 1238 root      3360 S    ./record_event
 1239 root     14728 S    ./mp4record 60
 1268 root      1716 S    /usr/sbin/crond -b
 1291 root       904 S    /home/led_ctl -boff -yfast
 1292 root     98.1m S    ./rmm
 1305 root     27252 S N  ./rtspsvrM
 1412 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      3.0M    508.0K  86% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    916.0K  90% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                  7.4G      8.3M      7.4G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                  7.4G      8.3M      7.4G   0% /tmp/hd1/test/http/record

```
These SD works well in other cam.

VaporKing, 2016-12-23:
#94 Take a look at my last entry with this issue

Joeribl, 2016-12-30:
I have the same problem with the CN12 version. The network seems to come up for just a short time but then it goes down again.

Today i connected with a TTL serial adaptor. But it did not give me any clues about what goes wrong....