fritz-smh/yi-hack #114: check_motion.sh Auto-start?

danielolson13, 2016-12-15:
Is check_motion.sh supposed to run on start-up? Mine doesn't start automatically but I can run it with...
`/tmp/hd1/test/check_motion.sh &`

xmflsct, 2016-12-15:
I have the same question. It seems that the line is removed from the most recent startup script, and I don't know why..

danielolson13, 2016-12-15:
Ok got it, it used to be in equip_test.sh, but is no longer there. I can see it on the initial commit. I'm adding it back on my camera and will submit a pull to add the line back.

xmflsct, 2016-12-15:
Just a remark. Today I add the line back remotely, and restart the camera also remotely. Now it hangs, and cannot boot again even replug power... Don't know why. Will post again tomorrow when I have physical access to it.

danielolson13, 2016-12-15:
Huh, I just added it on 2 cameras without issue.

xmflsct, 2016-12-15:
Will see tomorrow, and maybe post the debug log here in case anything interesting.

ygator, 2016-12-16:
You probably forgot the & at the end of the line to start it in the background.

xmflsct, 2016-12-16:
Actually i did. I could not solve the problem even after accessing the camera physically. So I re-format the SD card and update the files again, now it works... Don't know why formatting helps though.

utya1988, 2017-03-05:
hello. motion_detect.sh script doesn't start.

i added `/tmp/hd1/test/check_motion.sh &` in equip_test.sh, but no result. Don't know why

anndrewww, 2017-03-21:
I have tried to get the script to auto-start, to no avail. Anyone have it figured out?