fritz-smh/yi-hack #118: Creating a version for the XiaoFang 1080p camera

no1knows, 2017-01-04:
I'm trying to hack the XiaoFang 1080p camera (i.e. this: http://www.gearbest.com/ip-cameras/pp_487830.html)

I've got root access over serial and worked out what filename to include on the microSD (that gets launched on insert): "snx_autorun.sh" so I can now easily change the root password and launch telnetd. It already runs boa web server (albeit with no content to serve, yet).

fritz-smh - could you give me some pointers on compiling the rtsp binary so I can create a version for the XiaoFang?

fijter, 2017-01-06:
@no1knows good stuff! Not an expert in this but I'd like to check out the fs as well to see what I can do. I've managed to enable telnetd through snx_autorun.sh but changing the root password with `echo "root:testing123" | chpasswd` didn't do the trick for me. Can you share what you did to get in?

no1knows, 2017-01-06:
I originally got in over serial (by soldering leads onto the camera's motherboard) but I managed to brute force the root password yesterday so that should make life easier. It's "ismart12".

fijter, 2017-01-07:
Thanks, great job. This device looks totally different from the Yi if I look at the script placed here. The processes iSC3S (I think for the streams itself) and miio_client (I think for cloud connection) do all the work. The device is constantly connected to a Chinese cloud server even when you are not using it, which is kind of creepy ;)

They didn't even bother to try to hide that this is just a iSmartAlarm Spot system with a slightly custom firmware; Even the hostname still is iSmartAlarm. The hardware is based on the Sonix 98600 SoC (http://www.sonix.com.tw/article-en-958-13487) which just seems to be a ARM based embedded linux platform with some webcam streams. 

I wasn't able to use chpasswd since it's not in the Busybox build; passwd is so that could possibly be used but why bother with the default root password you bruteforced ;) A simple snx_autorun.sh in the root of my SD card with `telnetd&` inside of it was enough to give me root access using telnet.

The rtspsrv binary from this project seems to be some kind of live555 media server based server; I tried to run it but it didn't work because of some missing dependancy on this Busybox installation.

The iSC3S keeps the /dev/video0 device busy, as soon as I kill the process this resource is available again. Once you kill the iSC3S app the device reboots after a moment, I tried killing iSC3S and test_UP but it still rebooted; So in order to produce a RTSP stream this auto reboot needs to be disabled so the iSC3S app can be killed. The Boa webserver is indeed running on it with just a `/cgi-bin/hello.cgi` test script and a symlink to the SD card contents at `/SDCard/`, nothing really special or noteworthy. 

SoniX seems to have a SDK but of course this doesn't seem to be publicly available. This is where it ends for me now since I don't have any experience with building binaries for embedded systems or decompiling/analysing the existing binaries. Hopefully it can be of some help on your quest to get this device to work; I think the hardware is great for the price if we can fully utilise it without having to rely on a third party app and a Chinese cloud to work.

no1knows, 2017-01-07:
Thanks, that's all useful info. It's easy to disable the camera's phone-home and video streaming functionality - simply comment out the last line in /etc/init.d/rcS:
`# /usr/bin/iSC3S/iSC3S &`

...and then reboot. It'll then happily sit there doing nothing, including not connecting to wifi - **so don't do this unless you have serial wires soldered to the board**.

Looks like the Live555 rtsp server code is here: http://www.live555.com/mediaServer/ but I don't have any experience compiling for Arm systems either so will need to do some digging. Judging by some RPi threads, sounds like ffmpeg, cvlc or avconf could also create rtp streams if we can get them compiled and running on the camera. I suspect the issue will be that these need some grunt so unless we can take advantage of the SoC's hardware encoding, the CPU might struggle. 

The SoniX SDK would be useful! I found the programming guide here:
http://pan.baidu.com/share/link?shareid=3532089155&uk=2654552623&fid=285728836228321 (also attached: [SN986 Series SDK Programming Guide.pdf](https://github.com/fritz-smh/yi-hack/files/691581/SN986.Series.SDK.Programming.Guide.pdf))

Paragraph 9.1 looks interesting:
> 9.1 SONiX Galaxy Streaming Sever. SONiX Galaxy is a full function streaming server for both video and audio in SONiX SN986 Serial SDK. Galaxy receives the live encoded video( from the hardware codec) and audio( from SONIX SN986 Serial middleware)as input, and then streams it through RTP over RTSP over UDP/TCP/HTTP. Users can connect to Galaxy and get video/audio stream through the IE browsers or the VLC Media players on the PC、mobile phone or the mobile pad. 

I can't find the SDK or SONiX Galaxy Streaming Sever anywhere... Don't suppose anyone can get their hands on either?

no1knows, 2017-01-07:
I've also found the attached "SN98600 Video Codec Programing Guide".
[SN986 Series Video Codec Programing Guide.pdf](https://github.com/fritz-smh/yi-hack/files/691737/SN986.Series.Video.Codec.Programing.Guide.pdf)

jeffrey-bosma, 2017-01-07:
> The iSC3S keeps the /dev/video0 device busy, as soon as I kill the process this resource is available again. Once you kill the iSC3S app the device reboots after a moment, I tried killing iSC3S and [...]

The iSC3S process interacts with `/dev/watchdog` and the board will reboot itself shortly after the process hangs or dies. I found that killing iSC3S and then unloading the watchdog kernel module (`rmmod snx_wdt`) will stop this reboot behavior.

magnets110, 2017-01-08:
I think I found the SN986 SDK

here: http://pan.baidu.com/share/link?uk=3981671631&shareid=2974078675#list/path=%2F

rehosted on mega https://mega.nz/#!UpUmHDyJ!xGf7OvWNcRp-pXRxQ4e-Ze06WtnimWaB9rhF_ca5ibc

no1knows, 2017-01-08:
Great work! I extracted SN986_1.50_037a_20151022_1049/package/app/galaxy-server.tgz to a microSD card. It also needs the following libraries:
libmtable.so from middleware/video.tgz
libsnx_cap.so from middleware/snx-cap-lib.tgz
libsnxconf.so from middleware/snxconf.tgz

So I put those in the galaxy-server/rootfs/lib folder alongside the existing libserverlog.so and libxmllib.so. 

Once the microSD card is in the camera, telnet into it and then stop Xiaomi's streaming/cloud process hogging /dev/video[0-2] (thanks jeffrey-bosma!):
```
killall test_UP
kill -9 623 # replace "623" with whatever PID is associated with /usr/bin/iSC3S/iSC3S (which you can see by running "ps")
rmmod snx_wdt
```
Then cp the Galaxy Server config files from the microSD card into /etc:
`cp -R /media/mmcblk0p1/galaxy-server/rootfs/root/etc_default/* /etc/`

Then the camera where to find the new libraries, since we can't copy them into /lib as the filesystem is read only:
`export LD_LIBRARY_PATH=/media/mmcblk0p1/galaxy-server/rootfs/lib`

Then, ./sonix-proj finally executes and spits out the following:

```
/media/mmcblk0p1/galaxy-server/bin # ./sonix-proj
Create recording fun[id:1]
Create recording fun[id:2]
Create recording fun[id:3]
init watchdog ok
Log name : console  Maxsize is 1048576
create video source [uuid:38BDB80C-9662-49AE-B53D-A046097950B4] /dev/video0  1280x720 @ 30 with h264
create video source [uuichannel 0 buffer count=2, size=462848
d:B93E7FB4-D37B-404F-9333-7F4E75919F29] /dev/video0  640x480 @ 30 with h264
create video source [uuid:B93E7FB4-D37B-404F-9333-7F4E75919F31] /dev/video0  640x480 @ 30 with mjpeg
create audio source [uuid:E2C4CF08-4BD2-48C8-9EDB-B390BAF5C893] snx_audio_mulaw @ 8000 with mulaw  bitrate:64000
WARNING : snx_conf.c 774:file /tmp/sources.conf is existing
nokey value is video h264 /dev/video0 640x480 30 /dev/video2 640x480 30 bitrate 614400 -1 30
debug: width 640 height 480  scales 1
debug: ispdev /dev/video0 codec_dev /dev/video2
debug:isp fps  30 codec fps 30
sc2135 start streaming
----->has ability to crop!!
cropcap.defrect = (0, 0, 1920, 1080)
----->sussess crop to (0, 0, 640, 480)
snx_audsnx_vc snx_vc: snx_vc_open: Created instance c31d6800, m2m_ctx: c33aa800
io_mulaw, 8000, 64000, mulaw, 0, uuid:e2c4cf08-4bd2-48c8-9edb-b390baf5c893
create audio source [uuid:E2C4CF08-4BD2-48C8-9EDB-B390BAF5C895] snx_audio_alaw @ 8000 with alaw  bitrate:64000
create audio source [uuid:E2C4CF08-4BD2-48C8-9EDB-B39snx_vc snx_vc: snx_vidioc_s_parm: snx_vidioc_s_parm: OUTPUT fps == 30
0BAF5C897] snx_audio_g72snx_vc snx_vc: snx_vidioc_s_parm: snx_vidioc_s_parm: CAPTURE fps == 30
6 @ 8000 with g726  bitrate:16000
snapshot res is 1280x720
snx_audio_alaw, 8000, 64000, alaw, 0, uuid:e2c4cf08-4bd2-48c8-9edb-b390baf5c895
snx_audio_g726, 8000, 16000, g726, 0, uuid:e2c4cf08-4bd2-48c8-9edb-b390baf5c897
isp init ok============
snx_vc snx_vc: s_fmt: Setting format for type 2, wxh: 640x480, fmt: 808596563
640 480  scale == 1
snx_vc snx_vc: s_fmt: Setting format for type 1, wxh: 640x480, fmt: 875967048
<<<snx_vb2_alloc>>> alloc size=925696 reduce size=462848
codec init ok============  CODEC
-----2_h-----
Can find folder rc-snx_vc snx_vc: snx_vc_release: Releasing instance c31d6800
>rc_folder, default folder is /etc/rc
Segmentation fault
```

So close! I have absolutely no idea what the issue is - perhaps a missing app/library? Incorrect config? I attach the folder containing all the necessary files: 
[XiaoFang_Galaxy_Server.zip](https://github.com/fritz-smh/yi-hack/files/692435/XiaoFang_Galaxy_Server.zip)

p.s. to stop iSC3S starting up in the first place, comment out the last line in /etc/init.d/rcS ("/usr/bin/iSC3S/iSC3S &"). This will stop wifi starting up as well but you can make that happen by adding the following into /etc/wpa_supplicant.conf:
```
network={
        ssid="YOURSSID"
        psk="YOURPASS"
}
```
...and the following into /etc/init.d/rc.local:
```
sleep 5
echo "Start wifi..."
ifconfig wlan0 up
wpa_supplicant -B -iwlan0 -c /etc/wpa_supplicant.conf
udhcpc -i wlan0
```

samtap, 2017-01-09:
For some reason the snx_autorun.sh trick doesn't work for me. The script isn't executed or telnetd fails to start, either way I get connection refused when trying to connect over telnet. I still have the 'original' firmware (2.8.0.0, app wants to update to 2.8.3.5 but I'm refusing).

How should the sd card be formatted? Are there any instructions available (a photo would be awesome) of the cable connections for serial console?

fijter, 2017-01-09:
@no1knows Good work on getting rid of the running cloud stuff and @magnets110 great work on finding the SDK. Not sure if Galaxy server is going to do us any good for getting a local RTSP stream up and running, it looks like this service just connects the stream to a configurable cloud server. Too bad the documentation of the SDK is a bit lacking, can't find an example for a RTSP server in there without the cloud connect part; Maybe I should look a bit harder ;)

@samtap Just a FAT32 formatted card, the file should be in the root of that card containing `telnetd &` - Please note that this script doesn't get executed on boot like the Yi device but on insertion of the SD card, so just leave your cam on, remove the SD, replace it, wait until you hear a sound and telnetd should be running. If you reboot the device with the card inserted you need to re-inject it for it to run again.

jonnycastaway, 2017-01-09:
I did the telnet thing so:
First start with the snx_autorun.sh trick.
Then in /etc/init.d/rc.local remove the last line that begins with "cp "
then Append on the end "/bin/sleep 30 && telnetd"

so after every reboot the script waits 30 seconds and then start the telnet daemon
no SD-Card needed. it also stays after press the setup switch and connect to a new network. only a reboot after the successful reconf is needed.

But attention, it will broke by a firmware update.

jonnycastaway, 2017-01-09:
I searched in the SDK to and find the SnOnvif executable. It also will have an extra library. i copy the files SnOnvif and libxmllib.so to my card, add the Library Path as described above and start SnOnvif but get the same error: 
`Segmentation fault`

Nothing more. :-/

samtap, 2017-01-09:
@fijter Thanks, it works when inserting the sd-card after boot. Strange, but anyway it's only needed once, to make the fix in /etc/init.d/rc.local

The SDK actually contains live555 rtsp_server! So I have high hopes we'll get somewhere. I was able to build and run it but unfortunately it also segfaults. But gdbserver is available so next step is creating a debug build.

> /tmp # ./snx_rtsp_server
> Video source = /dev/video1 
> ----->has ability to crop!!
> cropcap.defrect = (0, 0, 1920, 1080)
> ----->sussess crop to (0, 0, 1280, 720)
> 1280 720  scale == 1
> Can find folder rc->rc_folder, default folder is /etc/rc 
> Segmentation fault (core dumped)

petero-dk, 2017-01-09:
I have just purchased a couple of these. I will start by testing the above. This is to let you know there is one more interested party willing to test and contribute. Wish: rtsp streaming, and final hope: Onvif support with events.

no1knows, 2017-01-09:
@samtap - can't believe I didn't see that. I'm trying to cross compile it on Ubuntu 16.04 but struggling. Sounds like you know what you're doing rather better than me... keep us posted!

samtap, 2017-01-09:
It's working reasonably well. You'll need to get middleware/_install/lib/libsnx_rc.so from the SDK as well.

> /tmp # ./snx_rtsp_server_dbg
> Video source = /dev/video1 
> ----->has ability to crop!!
> cropcap.defrect = (0, 0, 1920, 1080)
> ----->sussess crop to (0, 0, 1280, 720)
> 1280 720  scale == 1
> Can find folder rc->rc_folder, default folder is /etc/rc 
> [snx_get_file_value] open md_ratectl_en error
> Start V4L2 Capture... /dev/video1 
> 
> 
> ------- V4L2 Infomation -------- 
> m2m_en: 1
> codec_dev: /dev/video1
> codec_fps: 30
> isp_fps: 30
> width: 1280
> height: 720
> scale: 1
> bit_rate: 1048576
> dyn_fps_en: 0
> GOP: 30
> ds_font_num: 128
> 
> ----------------------------- 
> 
> isp buffers: 1384448 , outpack maxsize : 262144
> lay this stream using the URL: "rtsp://192.168.42.123/unicast"
> H264 thread 1280x720 
> m_a profile-level-id=644029;sprop-parameter-sets=Z2RAKawsqAUAW5A=,aO44gA==
> RTSP DESCRIBE received
> auxLine a=fmtp:96 profile-level-id=644029;sprop-parameter-sets=Z2RAKawsqAUAW5A=,aO44gA==
> 
> RTSP SETUP received
> RTSP PLAY received
> out tv_sec: 1484002726, fps: 1, bandwidth: 10 kbps
> afterGettingFrame1: start playing time 1484002726:873275
> RTSP GET_PARAMETER received
> out tv_sec: 1484002727, fps: 11, bandwidth: 102 kbps
> out tv_sec: 1484002728, fps: 10, bandwidth: 85 kbps
> out tv_sec: 1484002729, fps: 10, bandwidth: 330 kbps
> out tv_sec: 1484002730, fps: 10, bandwidth: 93 kbps
> out tv_sec: 1484002731, fps: 10, bandwidth: 87 kbps
> out tv_sec: 1484002732, fps: 10, bandwidth: 330 kbps
> out tv_sec: 1484002733, fps: 10, bandwidth: 94 kbps
> out tv_sec: 1484002734, fps: 10, bandwidth: 85 kbps
> out tv_sec: 1484002735, fps: 10, bandwidth: 330 kbps
> out tv_sec: 1484002736, fps: 10, bandwidth: 91 kbps
> out tv_sec: 1484002737, fps: 10, bandwidth: 87 kbps
> out tv_sec: 1484002738, fps: 10, bandwidth: 331 kbps
> out tv_sec: 1484002739, fps: 10, bandwidth: 92 kbps
> RTSP TEARDOWN received
> 
> ^CSignal received (2)
> Exiting....  
> ~V4L2DeviceSource
> stopping: (null) 
> Segmentation fault (core dumped)
> /tmp # 

Seems to be limited to 10fps? It drops even lower when I move the cam around a bit.

ykhandler, 2017-01-10:
Hi samtap. I need a hand on xiaomi yi ants 17CN rtspsvr. Can i have your email? Or maybe you can email me on yk_handler@yahoo.com... Thx..

no1knows, 2017-01-10:
@samtap looks promising! can you share the compiled binary for me to test? What system did you compile it on? Any tips/tricks for cross compiling?

jonnycastaway, 2017-01-10:
@samtap did you cross compile it for arm? any special parameters or packages needed? i have a ubuntu install with cross compile things for arm to so i can also test and check this.

samtap, 2017-01-10:
@no1knows I built it on Kubuntu 16.10 x64. There're a couple of small issues to resolve when using the SDK. Is there anything particular you're running into?

Basicly what I did  to build snx_rtsp_server was change a couple of scripts to use bash instead of sh (change #!/bin/sh to #!/bin/bash). Run sdk.unpack. I run setup_env.sh but it's probably safe to omit. Add bin and lib from the toolchain dirs to PATH and LD_LIBRARY_PATH. There's probably a better way to do this with a crosstool command. I had to create an empty file buildscript/include/config/snx_sdk.conf. Also had to mess around in some perl file to make it work with the perl version on my machine. When all is well you should be able to enter the buildscript dir and build the kernel with make oldconfig && make.  This builds a kernel image and sets everything up so you'll be able to build the apps using their individual makefiles. 

Note there's lots of room for tweaking stuff. You can configure the kernel with 'make menuconfig' and there're lots of config files that I didn't touch. I was able to debug with the supplied arm-linux-gdb so that's promising. But eventually you'll probably need knowledge about the low level hardware to tweak the SDK for our specific target. Also the SDK might be a bit outdated (Oct 2015) and not match the latest firmware (though my binaries work on the latest fw). We should also look for a recovery method (I'm sure there's some way to flash it from the sd card, maybe bootloader output gives some clues?). Before messing with the system and risking to brick the device ;-)

petero-dk, 2017-01-10:
I am currently in the process of aquiring the latest SDK for the platform, is any one sure which of the SN986 versions this is?

|Part No|Package Type|Max Performance|DDR|LCD/HDMI|I2S|I2C|SPI|TV-out|MIPI|Serial/NAND Flash|JTAG|
|--- |--- |--- |--- |--- |--- |--- |--- |--- |--- |--- |--- |
|SN98600|LQFP128|1080P15|64MB DDR2 SIP|N/A|N/A|1|1|V|V|Serial/NAND Flash|N/A|
|SN98601|LQFP128|1080P15|64MB DDR2 SIP|N/A|N/A|1|1|N/A|V|Serial Flash|V|
|SN98610|TFBGA292|1080P15|16 / 32-bit DDR2|V|V|2|2|V|V|Serial/NAND Flash|V|
|SN98660|LQFP128|1080p30 + 480p30|64MB DDR2 SiP|N/A|V|2|1|N/A|V|Serial Flash|V|
|SN98661|LQFP128|1080p30 + 480p30|64MB DDR2 SiP|N/A|V|1|1|N/A|V|Serial Flash|V|
|SN98671|QFN88|720p30 + 720p30|32MB DDR2 SiP|N/A|N/A|1|0|N/A|V|Serial Flash|V|
|SN98672|QFN88|1080p30 + 480p30|32MB DDR2 SiP|N/A|N/A|1|0|N/A|V|Serial Flash|V|

jeffrey-bosma, 2017-01-10:
> I am currently in the process of aquiring the latest SDK for the platform, is any one sure which of the SN986 versions this is?
> [...]

That would be great to have! It's SN98660(AFG), someone posted a teardown [here](http://www.geekifix.com/wp/archives/272.html).

no1knows, 2017-01-10:
The teardown is a good find. In case anyone is interested in getting a serial console access, the pads are shown in this photo: http://cdn.geekifix.com/forum/201612/07/172112zf9lv9as3y3kdtzx.jpg (see four empty pads on the right hand side labelled 3.3v, rx, tx, gnd). That's on the back of the motherboard, so you need to remove the two ribbon cables and the motherboard to get at the pads. It's pretty striaghtforward - I've done this on both of my cameras and reassembled them with the wires coming out of the back of the camera through a small hole I cut in the plastic grate section between the speaker and USB port. Baud speed is 115,200.

samtap, 2017-01-10:
@no1knows I assume you hooked it up to a 3.3v ftdi uart adapter? I'll try this evening. A warning for anyone who tries to connect it directly to an USB or serial port: those run on 5V and will probably fry the board...

samtap, 2017-01-10:
RTSP server binaries available here: https://mega.nz/#F!aFZDVAiQ!9DBh2xMR9D_JynwtYSiASQ

jonnycastaway, 2017-01-10:
@samtap Thank you for the rtsp server, it works perfect exept one little thing. The IR-Filter lens is not working anymore. I think it's also controlled by the iSC3 bin. The delay is 1 second in my config, so its ok. Now im searching a way to switch the ir-filter lens.

jonnycastaway, 2017-01-10:
Hey guys,
after some testing i doing this to kill the unwanted things and start the rtsp server automatically:
first i kill all necsessary processes with the following lines in /etc/init.d/rc.local at the end:
`killall test_UP
ps | grep -i iSC3S | grep -v grep | cut -d " " -f 3 | xargs kill -9
/bin/sleep 30 && rmmod snx_wdt`
After that i make a new dir in /etc called rtsp and copy the 2 files from samtap in (wget is your friend). so the files are not gone after a reboot. don't forget chmod a+x snx_rtsp_server! Then also in rc.local i do the stuff to load the rtsp server:
`export LD_LIBRARY_PATH=/etc/rtsp
/etc/rtsp/snx_rtsp_server &`

if you do all right your cam runs the rtsp server ;-) test the stream with
rtsp://yourcamiphere/unicast

attention, after a reboot it needs 1 minute to do all the stuff and start the rtsp server
another thing is that des LED on the back is not lighting up.

navigonic, 2017-01-10:
following this thread like a hawk hoping for a howto from your successes :-)

jonnycastaway, 2017-01-10:
Ok, here is a little summary. I merged the Infos from no1knows, samtap and me.
Now i don't start iSC3S and the watchdog. So the new way comes here:

in /etc/init.d/rcS commend out the following lines with an # at the beginning
`modprobe snx_wdt`
`/usr/bin/iSC3S/iSC3S &`

then in /etc/wpa_supplicant.conf remove all in the network section except:
`        ssid="YOURSSID"`
`        psk="YOURPASS"`
fill YOURSSID and YOURPASS with valid Data for your wlan!

make a new dir in /etc called rtsp and copy the 2 files from samtap in (wget is your friend). so the files are not gone after a reboot. don't forget chmod a+x snx_rtsp_server!

in /etc/init.d/rc.local remove the last line that begins with "cp "
then Append on the end:

`sleep 5`
`echo "Start wifi..."`
`ifconfig wlan0 up`
`wpa_supplicant -B -iwlan0 -c /etc/wpa_supplicant.conf`
`udhcpc -i wlan0`

`echo "Start telnet..."`
`/bin/sleep 30 && telnetd`

`echo "Start RTSP Server..."`
`export LD_LIBRARY_PATH=/etc/rtsp`
`/etc/rtsp/snx_rtsp_server &`

I have 2 Cams running with this adjustments without Problems.

no1knows, 2017-01-10:
Well done everyone! I'll try this out tomorrow and once all tested perhaps get started on an snx_autorun.sh (in a new project) that automates everything to make it easier for others to benefit from all our good work!

@samtap good point - I should have mentioned, no need to hook up the 3.3v - it works fine with just the RX, TX and gnd wires connected. I'm using a cheap universal USB to serial adapter and Putty.

roger-, 2017-01-11:
Could someone rebuild the RTSP server with authentication support? It's really easy to add, [just a couple extra lines in live555](https://github.com/xanview/live555/blob/master/mediaServer/live555MediaServer.cpp#L30).

fijter, 2017-01-11:
Great work all, awesome! :)

In case you didn't find out already, there's an option as well to increase the resolution, bitrate, and enable audio, I ended up with the following command as the last line in rc.local:

`/etc/rtsp/snx_rtsp_server -W 1920 -H 1080 -Q 15 -b 4096 -a &`

This increases the resolution, quality and enables audio. The only thing I'm missing for now is enabling the IR stuff; Hopefully the SDK provides a useful binary for that as well!

jonnycastaway, 2017-01-11:
@fijter Many thanks for share this big quality improvement! I added your line in the summary.
I am searching right now in iSC3S binary for any hints to get ir-lens working.

magnets110, 2017-01-11:
Great work!
Is it running smoothly at 1080p30?

mitchelp, 2017-01-11:
@jonnycastaway regarding the ir-lens, maybe there is a possibility to read the light sensor (if there is one?) and use that to turn on or off the ir-lens?

My cam is currently running with your "little summary" and it's working quite well. Only the cam is not so clear as I hoped. Maybe we could do some tweaking with the bitrate and fps?

fijter, 2017-01-11:
@jonnycastaway @mitchelp please look at `/app/example/src/ir_ctl` in the SDK, this looks like this does what we need (turn on IR/filter when night, turn it off when day, based on GPIO). It just needs a compile (I tried but I have no experience with cross compiling so didn't get very far). If we have this binary it is probably just another rule in the rc.local to turn that script on.

jonnycastaway, 2017-01-11:
@mitchelp 
yes there is a light sensor. this is used by the original programm to switch the lens. but i have to find out 1. how i can read the light sensor and 2. how can i switch the lens.

For the bitrate and fps thing run snx_rtsp_server -h from the console. here you can see certain parameters to play with.

@fijter thanks for the hint, i will look at the file after my lunch ;-)

You all did a great job !

cycloptux, 2017-01-11:
All of this still assumes you're running original 2.8.0.0 firmware, right? I'd have liked to test it out and I'd have no problems getting serial access soldering wires, but I "mistakenly" upgraded the firmware when my cam arrived a couple weeks ago. Are these firmwares availables somewhere/flashable in some way?

petero-dk, 2017-01-11:
I have upgraded one of my cameras to the latest version, and it does the SD card trick just fine. Have not gotten around to testing the binaries.

fijter, 2017-01-11:
@cycloptux don't bother with a downgrade, I had the latest firmware and it works on that.

cycloptux, 2017-01-11:
Nice, I'm at work now but I'll try tonight.

jonnycastaway, 2017-01-11:
@fijter i can not compile the ir_ctrl.c example but i study it and find the following:
in /bin there is a binary called gpio_aud and with this line you can switch the ir leds on:
`gpio_aud write 1 0 1`
and off
`gpio_aud write 1 0 0`

more news to come, i am close to the ir-cut lens ;-)

dmolner, 2017-01-11:
Please could you make instructions for work these in my camera?

samtap, 2017-01-11:
Apparently with the latest firmware, the framerate is no longer limited to 10 fps :-). I'm still on the old version though. It would be nice to dump it so we can revert if needed. 

I'm not sure if we'll be able to create our own image. I can successfully build one based on the SDK, but its for a generic SN98600. The SDK lacks some stuff required for SN98660. Even with that stuff, you'll need detailed info from the hardware and u-boot but then it should be possible to simply flash the device from a file placed on sd-card (There documentation that proves this is possible). 

For the time being I'm not going to risk bricking my device with a custom image. So I'm working on a method to use files placed on sdcard. That way we don't have to mess with the system except for a single rc.local modification. This will include a busybox with lots of goodies like ftpd, netcat and I'll include ir_ctl so somebody can hopefully figure out how it works.

jonnycastaway, 2017-01-11:
Ok i got the ir-cut lens to ;-)
first an init sequence:
`gpio_ms1 -n 2 -m 1 -v 0`
`gpio_aud write 1 1 0`
`gpio_aud write 1 2 1`
`gpio_aud write 1 0 0`

and then ir-cut lens on:
`gpio_ms1 -n 2 -m 1 -v 1`
and off:
`gpio_ms1 -n 2 -m 1 -v 0`

to switch the led on see my last post ;-)
now i come to the light sensor ;-)

jonnycastaway, 2017-01-11:
ok, next thing, the light sensor.
after the init-process in my last post gpio 2 is set to 1 (high)
from now you can read that gpio with
`gpio_aud read 2`
if it is 1 then its day. is it 0 then the dark night rises ;-)

fijter, 2017-01-11:
I don't know how to connect to it yet/what client to use; But I've got two way audio running as well.
The binary is in the SDK (app/two-way-audio/rootfs/bin) and needs libsnxconf.so, I just placed both in /etc/rtsp, did the `export LD_LIBRARY_PATH=/etc/rtsp` and it fired up some audio server on port 8004. If anyone knows how to communicate with it that would be nice. There's a boa binary as well for a webserver which can be nice but for some reason it doesn't run for me.

arhue, 2017-01-11:
Hello!

I got my cam today and the lack of RTSP was a big bummer. Stumbled upon this thread when looking at ways of getting it. I've never played around with modifying firmware on cams so forgive me if I say something stupid.

Would copying files from the Mega link give me telnet access? Would be interesting to see how much you can play around with the wpa_supplicant. Trunked frames over WiFi to untag them on the device would be super great for me. Someone needs to make a small guide or maybe a script to easily execute on the device. Being patient is hard :P

jonnycastaway, 2017-01-11:
i saw an error in my gpio initial post:
`gpio_aud write 1 2 1`
i corrected it to:
`gpio_aud write 0 2 1`

mitchelp, 2017-01-11:
@johnnycastaway very good work!! That's help full!!

Now we should be able to automate things, maybe we could create a little binaries that runs and reads the status of the light sensor? And according that switch ir on and off? I don't have any experience doing that but there should be somebody here that is capable of creating something like that!

mitchelp, 2017-01-11:
Or with simple bash script?

jonnycastaway, 2017-01-11:
@mitchelp yes, thats all possible from now!
a little hint i came across by my testings. the best results comes out when you first take the ir_cut lens on and then the ir_led and vice versa to disable all.

jonnycastaway, 2017-01-11:
And here a little Summary of my work today:

ir init process - checked:
`gpio_ms1 -n 2 -m 1 -v 0`
`gpio_aud write 1 1 0`
`gpio_aud write 0 2 1`
`gpio_aud write 1 0 0`

ir_led - checked:
on
`gpio_aud write 1 0 1`
off
`gpio_aud write 1 0 0`

ir_cut lens - checked:
on
`gpio_ms1 -n 2 -m 1 -v 1`
off
`gpio_ms1 -n 2 -m 1 -v 0`

light sensor - checked:
`gpio_aud read 2`
1 is day, 0 then the dark night rises ;-)

finished for today, i'm going home now

vicfergar, 2017-01-11:
Great work @jonnycastaway!! 👍 

It would be nice to have another daemon that checks continuously the light sensor and set "ir cut lens" and "ir led" according to the returned value. What do you think??

dmolner, 2017-01-11:
Do you thing is possible the image no in BW? all time colour?

no1knows, 2017-01-11:
Judging by the console output of iSC3S, that's exactly what it did - periodically read the light sensor and switch on the IR led and filter when it got dark enough. I'll run a few tests to see if there's enough logging output to work out the default thresholds that iSC3S uses.

GB505, 2017-01-11:
What about being able to change the time embedded into the video files / feed so it's not in Chinese time.
Even enabling NTP so it can update the date and time from a NTP Source ?

jonnycastaway, 2017-01-11:
@no1knows for the lightsensor there is no threshold. iSC3S did the same i did: read the value from gpio and if 1 comes back it is day and otherwise its night. i think in 2 hours i have little time to write a script that we can start from rc.local etc. in background that automatically reads the gpio and if it's night, switch the ir_leds and the ir_cut lens on. i think thats not a big task. read lightsensor, store returned data in a variable, check if it's 0 or 1 and switch the ir-stuff, sleep a time, read lightsensor again, check if the return data is equal the older data and if not switch the ir-stuff again. this is exactly what the ir_ctl example from the sdk did.

samtap, 2017-01-11:
@GB505 ntpdate is already available but there's no zoneinfo. You need to specify a timezone-offset (from China I believe) in /etc/TZ. I'm working on a script to automate it. 

It's probably quite straightforward, given all the example code, to integrate ir-filter control in the rtsp server binary. It would perform much better (low detection time) compared to running some script every five seconds. Did you guys already notice there're two usable video devices? Only one has the timestamp overlay.

jonnycastaway, 2017-01-11:
@dmolner you have a little error in your thinking ;-)
the sensor switches not to BW. It's all time colour. even when the ir_cut lens is active.
the BW colour comes from this:

There is not enough light for the sensor but the sensor can "see" infrared light.
so we power on some ir-leds that lights up things like a flashlight. the ir-spectrum is very broad but the ir-led emits only a very small piece of the sprectrum. and so the sensor "see" only this small piece and thats why ist shows BW. You can test it when you let the ir_led and the ir_cut lens on and then got with a flashlight in the dark room an lighten up anything, its coloured except the rest thats in the dark ;-)

[268 more replies]