fritz-smh/yi-hack #119: Having issues with my timestamp

lund34, 2017-01-12:
Hi!

I used the search function and found people who have had the same issue, and I tried all of their solutions but still cant get it to work. 

My Timestamp still shows the camera being set to 1970/01/01 00:00:00

I'm sure my wifi configuration is ok.  This is what I've entered (numbers are from ipconfig):

> IP=10.10.1.48
> NETMASK=255.255.255.0
> GATEWAY=10.10.1.1 
> NAMESERVER=10.10.1.1

Timezone set to :

> TIMEZONE=CST6CDT,M3.2.0,M11.1.0

NTP :

> NTP_SERVER=0.north-america.pool.ntp.org

I also dont have anything set up in my firewall to block the camera from accessing the internet.

Anyone have any other suggestions of things I can try?

Thanks!

Here is what my log file looks like :

> Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Jan 12  2017 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Jan 12  2017 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Jan 12  2017 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Jan 12  2017 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Jan 12  2017 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr 20:F4:1B:F6:79:43  
              inet addr:10.10.1.48  Bcast:10.10.1.255  Mask:255.255.255.0
              inet6 addr: fe80::22f4:1bff:fef6:7943/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:48 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:10791 (10.5 KiB)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.north-america.pool.ntp.org
':
Previous datetime is Thu Jan  1 00:00:23 UTC 1970
Done
New datetime is Thu Jan  1 00:00:23 UTC 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1232 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1259 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1335 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  681 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  816 root         0 SW   [RtmpTimerTask]
  817 root         0 SW   [RtmpMlmeTask]
  818 root         0 SW   [RtmpCmdQTask]
  819 root         0 SW   [RtmpWscTask]
  949 root         0 SW   [hidog]
  987 root      1164 S    ./peripheral
  988 root      1144 S <  ./dispatch
  989 root      1068 S    ./exnet
 1133 root      3132 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1232 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1259 root      1408 S    ./serverM 80
 1277 root      3360 S    ./record_event
 1278 root     14728 S    ./mp4record 60
 1296 root      1716 S    /usr/sbin/crond -b
 1320 root       904 S    /home/led_ctl -boff -bon 
 1330 root     98.1m S    ./rmm
 1335 root     30568 S N  ./rtspsvrM
 1452 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    796.0K  78% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    916.0K  90% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                 29.7G      9.4M     29.7G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 29.7G      9.4M     29.7G   0% /tmp/hd1/test/http/record

>

ygator, 2017-01-13:
The script does the following:
1. log "Get time from a NTP server..."
2. NTP_SERVER=$(get_config NTP_SERVER)
3. log "But first, test the NTP server '${NTP_SERVER}':"
4. ping -c1 ${NTP_SERVER} >> ${LOG_FILE}
5. log "Previous datetime is $(date)"
6. ntpd -q -p ${NTP_SERVER}
7. log "Done"
8. log "New datetime is $(date)"
9.

My Comments:
3. Your log output has the ': on the following line.  Maybe just a bad message format or the NTP_SERVER variable is messed up somehow.

4. Your log output does not have the ping output.

8. Since ntpd is not run with the -n option the ntpd command returns right away and when it goes to display the "New" datetime it may just have not been set yet.

6. Really should have the -n option.  *** Doing this then brings up the issue of the camera not booting all the ways since it is trying to set the time if the timer server is down or your internet is down and you are not using a local time server.

9. If the -n option is added to line 6 then ntpd should be run again in the background to keep the camera time always in sync.  ntpd -p ${NTP_SERVER}

You can test, just telnet in and try: ntpd -d -n -q -p 0.north-america.pool.ntp.org
This will run with debug output and exit when the time is set.

lund34, 2017-01-13:
Hi ygator, 

Thanks for your help!

I think the part of the log output where the ': is on the following line is just a formatting issue.  

I did manage to telnet into the camera and run this command: ntpd -d -n -q -p 0.north-america.pool.ntp.org.  After I did that, the timestamp on the camera did change.  Although its not the correct time, it is the correct year.   And the camera did start recording with motion detection after I ran this command. (Feels great to finally make some progress!)

I tried to edit the equip_test.sh file to include the -n (file changed to: ntpd -n -q -p 0.north-america.pool.ntp.org), but the camera will not boot at all after that change (which is what I think your comment 6 refers to).

I'm curious why other people arent getting this issue.  Anyway, is the best option to just telnet into the camera and change the time that way and then leave it running rather than trying to edit the equip_test.sh file?

ygator, 2017-01-13:
You did not comment on 4?  Why is there no ping output.

> I did manage to telnet into the camera and run this command: ntpd -d -n -q -p 0.north-america.pool.ntp.org. After I did that, the timestamp on the camera did change. Although its not the correct time, it is the correct year.
That does not make sense.

Try this and post results: date; ntpd -d -n -q -p 0.north-america.pool.ntp.org; date

vosmont, 2017-01-18:
Hello,

after adding the command "sleep", the "ntpd" command works.

https://github.com/fritz-smh/yi-hack/issues/94

Clem-, 2017-10-10:
In order to update this issue, I think that the real problem was the configuration being incorrect.

The NAMESERVER variable should point to a DNS server such as Google's 8.8.8.8
With a valid DNS server, the ntp server address will be correctly resolved and the camera date will be successfully updated.

cdzhx, 2022-12-30:
用中文回答应该也行吧
NAMESERVER= 输入dns地址
设置正确的时区及正确的时间服务器即可正确解析时间。