fritz-smh/yi-hack #120: I have US Version of YiCam need RTSP and app

wyredone, 2017-01-13:
I have been using Chinese YiCams and when the US Version was release for $30 each on amazon prime (I was paying $33-$36 a piece from china and waiting weeks to get them). All of the firmwares I find are modified Chinese which I've used in the past with success however I want to be able to A. Use the YiCam App so I can do 2 way audio/etc. and B. Have RTSP streams so I can link it up to my Synology NAS Surveillance Station and record them that way as well. A bonus would be the ability to block YiCam app from checking for new firmware updates although I think that is done on the app side and if so cannot be blocked from the camera. Any help would be much appreciated. Im currently running 1.8.6.1B_201604071154 official - I got xiaomiYiAntsCameraFirmware_1.8.6.1B_rtspfix to partially work but would not connect via the app (likely because that was modded firmware for the chinese version to add rtsp).

death2all110, 2017-01-14:
You can use this custom firmware with 1.8.6.1B, but make sure to edit the equip_test.sh file  to use M for the HTTP server on the B firmwares. If you add `./p2pserver &` and `./cloud &` back to equip_test.sh after `./mp4record 60 &` then you can get the app to work. Make sure the app is already connected to the camera beforehand though. The only issue I had was that Alerts from the app did not work like this and my camera kept disconnecting. So I'm back on official fw for now. Though I miss the FTP for being able to copy the files from SD to my homeserver remotely.

death2all110, 2017-01-16:
Update!

See my commit [here](https://github.com/death2all110/yi-hack/commit/89549759e5fb4167c95efd96d45ee66ddb1033cf) for enabling cloud with working alerts.

lolento, 2017-03-18:
Hi Death2all110,

I have both RTSP and App running fine.  But any idea on how to get the recorded timeline on the App to work?