fritz-smh/yi-hack #122: Upgrading to 1.8.6.1B

mattbuford, 2017-01-16:
I'm using the US/international version of the camera (from Amazon).  After playing around a bit with the camera's original firmware, I installed this hack, and it works except there are a couple showstopping issues:

- IR LED comes on at night and my camera is inside a window, so the camera just blinds itself with reflection.  Before this hack, I used the app to disable the IR light.  Using the hack, I re-enabled cloud services and the app could see the camera, but the IR disable option was hidden.  Further research found that this feature was added in 1.8.6.1B.
- The picture at night was incredibly dark, even with the IR light on.  I had used the camera for a week on the original firmware, so I'm familiar with how the picture should look in the dark with IR off, and I was not getting that in the hack.  I can see outside clearly with original firmware, but barely see even the streetlight outside using the hack.

So, I went to Yi's site and downloaded the latest firmware, then copied that onto the "home" file on the sdcard.  I booted into this and it worked, however it was starting the "K" rtspserver, which would crash immediately upon a user connection.  I changed the equip_test.sh script so that it would start rtspsvrM, and that one doesn't crash.  I connected with VLC and success, I see a nice well-lit scene!

However, after a few seconds it became apparent that it only gives me a single frame of video each time I connect.  Data keeps flowing, but VLC never shows any further frames.  VLC's statistics show the frame count increasing and data flowing, but the on-screen picture never changes.  Pressing stop and play on VLC updates the picture to the current image, but then gets stuck again after only a single frame.

Also, starting cloud services doesn't seem to work.  The app is never able to connect to the camera.

So, has anyone made any better progress on this?

In the end, I just want to have the IR light off, the video to work, and the picture to not be crazy dark.  I don't really care what version of the firmware is used if I can make these things happen.

Gilgameshismist, 2017-01-16:
It looks like the script is using the IR filter wrong.
During daytime the IR filter should be before the lens, blocking IR light from the sun and the leds should be off, during nighttime the IR filter should be moved from the lens which will let sensor use the IR light from the LEDs (which should be on during nighttime).
What seems to happen is that the IR filter comes before the lens during night blocking the light from the leds, making it "crazy dark".

ygator, 2017-01-16:
IR LED
See thread "turn off the IR leds" for information on disabling IR Leds at boot time by changing /etc/ui.conf.  I am on 1.8.6.1B and I know it works for that.
VLC
I believe VLC has a problem with the stream when it is a night.  If you watch the stream on a mobile device with MX Player or RTSP Player it is fine.  It also works fine on a computer in chrome.  See thread "Live stream view on camera home page".

mattbuford, 2017-01-16:
Gilgameshismist:  In other comments, I found a command to enable/disable the IR filter.  When I did that (on the original hacked firmware), I did hear a click, but then the video would freeze up.

ygator:  I did have ui.conf configured to disable the IR LEDs all along (the setting seemed to be preserved from my original use of the camera on the non-hacked firmware), but that wasn't honored when the hacked firmware booted into 1.8.5, which is why I went to 1.8.6 + hack.  Anyway ... Aha!  It looks like you are right about VLC.  I just tried mx player and it plays video without getting stuck like VLC.  So, I guess all my video freezing issues were just VLC not liking something about the night video.  Thanks, that helps a lot.

Hmm, something is very wrong with the audio under mx player though.  I'm getting something like 5 seconds of audio, 10 seconds of silence, 5 of audio, 10 of silence, and so on.  Even more strange, the video continues uninterrupted during the audio silences.  This has resulted in my video being ~3 seconds behind realtime while my audio is 5-10 minutes behind realtime.  Hah!  Anyway, I guess I'll play with this some more tomorrow and see what I can figure out.

death2all110, 2017-01-16:
@mattbuford for enabling cloud services with latest firmware, see my commit [here](https://github.com/death2all110/yi-hack/commit/89549759e5fb4167c95efd96d45ee66ddb1033cf).

hyptechdev2015, 2017-01-26:
@death2all110 Thank you so much for the update on Yi-Hack. It working perfect only 
on Yi Home with IR (US version, purchased from Amazon), firmware 1.8.6.1B_201604071151

1. I setup Yi Home by follow official setup
2. Formware Sd Card
3. Download "death2all110's Yi-hack" package, and extract to SD card.
4. Follow death2all110's instructions
5. Replace replace "home" with official "home" firmware from https://www.yitechnology.com/firmware/index/class/home
6. It work perfectly with Cloud enabled and alerts

technophile1, 2017-01-30:
Seconded, I have the cams working perfectly with the cloud enabled patch from @death2all110 .  Thanks for all the amazing work to everyone!

My wishlist is:
Enable the OSD date/time with latest firmware (the one with bottom left "YI" watermark)
MQTT/ONVIF motion detection notification
Tiny SNMP server
Way to prevent user inadvertently upgrading firmware from app
Timelapse snapshots?

jimiamoore, 2017-02-17:
Has anyone tried the hack along side the current firmware -- 1.8.7.0A_201702081101?

death2all110, 2017-02-17:
I have not. I'll take a look this weekend.

ygator, 2017-02-17:
They are basically the same except for the following:
1. goolink removed
2. new p2p_tnp
3. new oss (and hi_cipher.ko kernel module)
4. Some tweaking of wifi tx power settings

death2all110, 2017-02-18:
According to the official changelog (that shows in app) it made some changes to night mode.

I updated my fork for the new firmware.

alfa756, 2017-02-21:
Followed the instructions. My camera is ver. 1.8.7.0A_201702081101
Works fine with original firmware but gets stuck on steady yellow light when booting custom firmware from microSD. Any suggestions?

alfa756, 2017-02-21:
Actually re-downloading all files corrected the error and got me to a point of "searching for wi-fi". However after about 45sec of blinking blue it turns to blinking yellow which I assume failure to connect to wi-fi. As i am a noob, could you please clarify if you script supports PSK2-AES and do I keep or remove the quotations marks around the SSID and password?

alfa756, 2017-02-21:
Does this look right: 
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid=Guest1
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk=jfneoijf77wos
}

death2all110, 2017-02-22:
@alfa756 your SSID and psk needs to be in double quotes

like 
```
...
ssid="Guest"
...
psk="mypskgoeshere"
```

alfa756, 2017-02-22:
Tried this but no luck. The blue light blinks then goes steady for 3 secs of so and back to blinking yellow.

Here is the script:

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid=“Guest1”
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk=“usmloaner”
}

This is what the router setup looks like (see attached)

Thank you for your help.

Anton

death2all110, 2017-02-23:
@alfa756 your double quotes are messed up. Its using curly/non-ascii characters.

I've attached the file with the information you have above.

If you're using textedit on a mac, hit cmd+a on the keyboard then go to edit>substitutions and click on "Smart Quotes" to de-select it. Then re-type the double quotes.

alfa756, 2017-02-23:
Followed your instructions exactly. Same thing: blinking blue for about 45 sec then steady blue for 3secs, then yellow flashing. Any other suggestions?

alfa756, 2017-02-23:
I tried something else also without success: I created a network with the name "your_wifi_ssid" and password "your_password" and just copies the original file I git from the site. Still not success. Could there be something wrong with one of the other files?

alfa756, 2017-02-23:
Got it, DHCP was not enabled by default in yi-hack.cfg
Now struggling with getting rtsp stream to show in blue iris

alfa756, 2017-02-23:
Done, really appreciate your help! is there any way to disable the blue light with the added script?

jimiamoore, 2017-02-23:
To confirm. This hack provides rtsp access and still allows you to access via the Yi Home app, correct?

alfa756, 2017-02-24:
I never ended setting up the Y home app so I can not confirm this, but it works very well with Blue Iris and it's mobile app.

death2all110, 2017-02-24:
@jimiamoore this version does not, but my fork does.

@alfa756  in the config should be able to set the LED_WHEN_READY to -boff -yoff

alfa756, 2017-02-25:
Thank you that worked great! Another question: is it possible to turn off infra-red with another script change - have camera monitoring through the window and need to disable it to avoid glare?

alfa756, 2017-03-03:
Any ideas on how to disable IR with a script?

death2all110, 2017-03-04:
I'm not sure. I'd check out the ui.conf file

alfa756, 2017-03-04:
Hm, there is no such file and any of the directories

tony-gutierrez, 2017-03-08:
So is @death2all110 's fork the best place to start now? Is it being merged into here?

tony-gutierrez, 2017-03-08:
@death2all110 Might want to modify your readme git clone to point to your own repo...I just did all that work for the wrong fork :)

raidenii, 2017-03-13:
@technophile1 
Not sure how to enable date & time OSD, but the watermark can be removed by following the guide here:
http://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/

Just extract (or use paint to create) two bmp files named hight_1555.bmp and low_1555.bmp to /home/hd1/test/, add cp -f /home/hd1/test/*.bmp /home/ to equip_test.sh (I added to the part after the camera finishs initialization), and that's all you need.

technophile1, 2017-03-13:
@raidenii Thanks!

death2all110, 2017-03-30:
@tony-gutierrez Fixed! Sorry about that!

brumsterino, 2017-05-09:
Can someone confirm if this works with the International version: 1.8.7.0A_201702081101

I have a camera, I've copied @death2all110 fork to the SD card, reset the camera and then booted. The camera boots and then goes through the waiting for connection process as if its ignoring the SD Card.

wpa_supplicant.cfg looks like this:

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="my-ssid"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="my-wifi-passwd"
}

and I have DHCP set.

Am I missing something?

Thanks.

galadril, 2017-05-10:
@brumsterino I've received my yi cam (1.8.7.0A_201702081101, international) and I can't get it to work with these packages.

brumsterino, 2017-05-10:
Thanks @galadril 

I've got a couple of queries out elsewhere. If I get anywhere I'll post back.