fritz-smh/yi-hack #124: Xiaomi 1080p Dome 360 Camera - Black colour

batrarobin, 2017-01-18:
[image]

[image]

[image]

I bought this camera from China and would like to use it overseas.  When I connect to it it starts saying "This camera can only be used in China".  I have been looking into various hacks that have been done to previous models but none of them have seemed to work on this one.

I have connected fly leads to the serial port and have connected via serial link.  I have tried to start "telnetd" service but it gives me an error saying "not found".   

Is anybody able to help with providing "telnetd" file from the previous model cameras so I can atleast get the telnetd access.

death2all110, 2017-01-23:
Can you get a dump of the file system? I tried and failed a couple times at extracting the data from the firmware image. Appears it is encrypted.

shadow-1, 2017-01-26:
I have managed to unpack the firmware file for a number of firmware images for the Yi Dome and Yi Dome 1080p cameras.

I have uploaded them at the link below.

[https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr](url)

---

With regards to getting telnetd working on the Yi Dome and Yi Dome 1080p. The issue is that the version of BusyBox included on these cameras is severely limited.
BusyBox which has been included on the Yi Home cameras has many more tools included.

The Yi Home, Yi Dome and Yi Dome 1080p cameras are all based on the HiSilicon Hi3518 chip. So it should be possible to copy BusyBox from the Yi Home and use it on the Yi Dome cameras. Alternatively BusyBox can be compiled from source. I managed to find the HiSilicon Hi3518 SDK online.

death2all110, 2017-01-26:
@shadow-1 can you explain how you were able to get the dome camera firmware extracted?

shadow-1, 2017-01-27:
@death2all110 
I got access to the Xiaomi Dome 1080p via serial port. From there I was able to examine how the camera unpacks the firmware and was able to replicate it. The steps are as follows:

1. Strip off the first 22 bytes from the firmware file. The first 22 bytes contains the firmware version.

2. Strip off the following 1344 bytes from the firmware file. This gives you an encoding key.

3. On the camera, there is a program to decode the encoding key file. Run the program rsa_pub_dec passing the encoding key as the parameter. I didn't look into the algorithm used to decode the file.

4. Append the decoded key (rsa_pub_dec program creates a file in the same directory where you run it) onto the firmware file after Step 2 (after stripping off the 1344 bytes).

5. Strip off the first 33 bytes from the firmware file after Step 4. This gives you the MD5 hash for the archive which we will get at the end.

6. Strip off the following 33 bytes from the firmware file. This will give you the password for the archive which we will get at the end.

7. Strip off the following 22 byes from the firmware file. This gives you the firmware version again (I guess to check that the firmware version from the initial firmware file matches after decoding).

8. The rest of the firmware file is a password protected 7zip archive.

---

The 7zip file can be extracted normally using any program compatible with 7zip archives.
You can verify that the archive hasn't been corrupted by comparing MD5 hash of the 7zip file against the MD5 hash extracted from the firmware file.
The password for the archive is 32 characters long. It has been extracted after Step 6.

shadow-1, 2017-01-27:
@death2all110 I currently have the Xiaomi Yi Dome 1080p camera which is region locked to mainland China. When I boot up, I get the message "This camera can only be used within China". The usual hacks I managed to find online do not seem to work with this camera (I still want to use the Xiaomi app).

If you have the same camera, do you happen to have firmware version 1.9.2.0A_201609281601 installed. If you do, can you share the filesystem from your camera or the firmware file if you have it saved?

I want to downgrade my firmware and see if I can defeat the region ban on the older firmware. However I can't seem to find the firmware file for download anywhere online.

felixekman, 2017-02-10:
So I have a full dump of the flash for the Yi Dome (not 1080p). I'm impressed @shadow-1 that you reversed that for the update file. Did you look at extpkg.sh for this or found it out some other way? As you mentioned the firmware files provided by Yi only update "home" which is related to the cloud services and other Yi services. It does not touch the busybox system or auxiliary at all. 

I've also found the current busybox distribution on the cameras severely limited as many functions are not there including telnet. I've tried to replace the entire system with my own compiled version of busybox with no success/fails to boot and drops to U-boot. Same with any form of RTSP server. So if anyone can compile for the chips mentioned above I could give it a try. 

For anyone exploring that does not want to fiddle with soldering leads to TX/RX pins. I can say that I've noticed that the system looks for /test/factory_test.sh on the SD card at boot. So a similar non-intrusive hack as this was originally should be possible as well. The camera also looks for these files at boot.
u-boot_v201
env_v201
conf_v201
kernel_v201
rootfs_v201
home_v201

For people who want to dump the entire flash of the camera this is the pattern of the SPI flash that I found.
mtd0: 00040000 00010000 "boot"
mtd1: 00010000 00010000 "env"
mtd2: 00010000 00010000 "conf"
mtd3: 00190000 00010000 "os"
mtd4: 00140000 00010000 "rootfs" //BUSYBOX
mtd5: 00cb0000 00010000 "home" //The "Yi" part - what is touched by the firmware updates.
mtd6: 00010000 00010000 "vd1"
mtd7: 00010000 00010000 "ver

So if anyone as questions/suggestions, drop a line here and I'll se what I can help with. Because I also really want to get this thing working. As for region block, I have a EU/US version so I haven't seen that issue. But I can see that functions dispatch and cloud does check for region and ban.

/////More data dump if it helps anyone:

## Booting kernel from Legacy Image at 81000000 ...
   Image Name:   7518-hi3518c-kernel
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1384824 Bytes = 1.3 MiB
   Load Address: 80008000
   Entry Point:  80008000

Linux version 3.4.35 (chenshibo@ANTS-SH-SV02) (gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300) ) #1 Wed May 25 19:18:17 CST 2016

CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
Machine: hi3518ev200

Kernel command line: mem=22M console=ttyAMA0,115200 root=/dev/mtdblock4 rootfstype=jffs2 mtdparts=hi_sfc:256k(boot)ro,64k(env),64k(conf),1600k(os),1280k(rootfs),12992k(home),64k(vd1),64k(ver)

ominivision ov9732 sensor 720P30fps

####

felixekman, 2017-02-16:
Anyone have some guidelines for combining the RTSP server for a new platform chip? I might have made progress with the dome camera.

phucn, 2017-02-17:
Dear shadow-1, 
could you share sensor chipset of  Xiaomi Yi Dome 1080p ? and dump firmware?

shadow-1, 2017-02-20:
@felixekman Yes I studied extpkg.sh to determine how the firmware file is being unpacked.
I managed to borrow a Yi Home camera. The busybox implementation on the Yi Home has all the tools included. This includes telnetd, ftpd, uname etc.

I have cheated a bit and transfered busybox from the Yi Home and found that it works perfectly on the Yi Dome/Yi Dome 1080p. I have placed a copy of this implementation of busybox on the link I posted above.

I am also trying to develop some programs on the Yi Dome. I suspect that the SDK I currently have is not compatible with the Operating System installed on the camera. I have tried finding newer versions of the SDK, however I don't think I found the appropriate cross-compiler yet. I have also added the SDK's I managed to find on the link I posted above.

I am looking for the arm-hisiv300-linux cross-compiler as this is what is required to develop applications for the Yi Dome 1080p. I suspect this is required for the Yi Dome too.

felixekman, 2017-02-20:
@shadow-1 how did you go about actually transferring the busybody from the regular home to the dome? I've tried a few ways but have been met with kernel panic upon boot and thrown back out to u-boot. 

I don't know if you tried these but they might give you some directions:
https://github.com/hi35xx/hi35xx-buildroot
https://github.com/hi35xx/hi3518e-buildroot
I've tried a few of these and Hi3518_SDK_V1.0.7.0 with no success.

shadow-1, 2017-02-20:
@felixekman I haven't tried replacing busybox on the Yi Dome. I just ran the tools manually from the SD card and confirmed that they work properly. For example, telnetd works perfectly.

felixekman, 2017-02-20:
Ah alright, makes sense. I've tried that as well with success, but not replacing the running system.

shadow-1, 2017-02-21:
@felixekman I managed to find the proper SDK for this platform. The chipset for the Yi Dome and Yi Dome 1080p is the HiSilicon Hi3518E V200.
However most of the SDK's available online are for the HiSilicon Hi3518E V100.

The SDK I got includes full documentation, including a guide on how to compile and deploy BusyBox.

The cross-compiler that is required to be used is arm-hisiv300-linux.

---

I am currently uploading the SDK on the same link as above to share. However my Internet upload speed is very limited and it will take a little while to upload. Box.com has a maximum file size of 250mb on their free account, so I had to split the archive into six pieces.

xcray, 2017-02-25:
maybe useful: just copy telnetd file from yi home 720p camera, it could works fine.

xcray, 2017-02-27:
Another way to extract the firmware via serial port is:
1, dd if=/dev/mtdblock? of=/tmp/sd/???
all the partitions can be copied to sd card or nfs directory (mount in advance).
the result is the images of each partition, of witch the image of rootfs and home is in jffs2 format.
2, mount the jffs2 images in a Linux pc, all the files are available now.
modprobe  mtdblock
modprobe mtdram total_size=image size
dd if={the image file} of=/dev/mtdblock?
mount -t jffs2 /dev/mtdblock? /{mntnode}

phucn, 2017-03-30:
@felixekman 
could you sent some info of umap devices?
cat /proc/umap/...

shadow-1, 2017-03-30:
@phucn 
What information are you after with regards to the umap devices?

---

@batrarobin @felixekman 
By the way, I have managed to create a custom firmware for the Yi 1080p Dome (black) that seems to work well. At the moment I have Telnet Server, FTP Server, Web Server implemented and have a working workaround for the region ban (for those with cameras region locked to mainland China). That is the main reason why I created the firmware.

The firmware is available here:
https://github.com/shadow-1/yi-hack-v3

I can adapt the firmware for the original Yi Dome (white) and the Yi Home 1080p v1 with a little help. They are all very similar cameras based on the same chipset. However I can't do any testing as I don't have these cameras.

I also have some ideas on getting RTSP streaming working at the same time as the Xiaomi cloud service. However I haven't had much time to develop it at this stage.

phucn, 2017-03-30:
@shadow-1 , i try to know sdk version of the camera, since have a lot of hi3518eV200 SDK

shadow-1, 2017-03-30:
@phucn I have successfully used Hi3518E_V200R001C01SPC030 SDK. I have compiled many programs for Hi3518EV200 chipset without issues. I think Hi3518E V200R001C01SPC040 has been released. However I don't have a copy of it.

phucn, 2017-03-30:
@shadow-1 , pls help me to find sdk version by cat /proc/umap/venc. do you know the IPC sensor chipset?

shadow-1, 2017-03-30:
[VENC] Version: [Hi3518EV200_MPP_V1.0.4.0 B050 Release], Build Time[Nov 25 2016, 21:56:23]

-----MODULE PARAM--------------------------------------------------------------
     VencBufferCache
                   0

-----VENC CHN ATTR 1-------------------------------------------------------------
      ID   Width  Height  Type  ByFrame     Timeout    Sequence   LeftBytes     LeftFrm  CurPacks  prio
       0    1920    1080    96        Y           0     2368476           0           0         1     0
       1     640     360    96        Y           0     2369043           0           0         1     0
       2    1920    1080    26        Y           0           0           0           0         0     0
       3     320     192    26        Y           0          43           0           0         0     0

-----VENC CHN ATTR 2-------------------------------------------------------------
   VeStr  OsdStr   SrcFr   TarFr     Timeref  PixFmt PicAddr
       Y       Y      15      15     4737906  YUV420  0x8313be00
       Y       Y      15      15     4737906  YUV420  0x83434000
       N       N      -1      -1           1  YUV420  0x       0
       Y       Y      -1      -1           1  YUV420  0x83532000

-----VENC CHN RECEIVE STAT-------------------------------------------------------------
      ID       Start     StartEx    RecvLeft     EncLeft
       0           1           0           0           0
       1           1           0           0           0
       2           0           0           0           0
       3           0           0           0           0

-----VENC VPSS QUERY-----------------------------------------------------------
      ID       Query     QueryOk     QueryFR       Invld        Full      VbFail   QueryFail     InfoErr        Stop
       0           0           0           0           0           0           0           0           0           0
       1           0           0           0           0           0           0           0           0           0
       2           0           0           0           0           0           0           0           0           0
       3           0           0           0           0           0           0           0           0           0

-----VENC SEND1----------------------------------------------------------------
      ID     VpssSnd     VInfErr     OthrSnd     OInfErr        Send        Stop        Full     CropErr    DrectSnd     SizeErr
       0     2368378           0           0           0     2368378           0           0           0     2368378           0
       1     2368954           0           0           0     2368954           0           0           0     2368954           0
       2     2368378           0           0           0           0     2368378           0           0           0           0
       3       82897           0           0           0           0       82897           0           0           0           0

-----VENC SEND2----------------------------------------------------------------
      ID     SendVgs     StartOk   StartFail       IntOk     IntFail      SrcAdd      SrcSub     DestAdd     DestSub
       0           0           0           0           0           0           0           0           0           0
       1           0           0           0           0           0           0           0           0           0
       2           0           0           0           0           0           0           0           0           0
       3           0           0           0           0           0           0           0           0           0

-----VENC PIC QUEUE STATE------------------------------------------------------
      ID    Free    Busy     Vgs
       0       6       0       0
       1       6       0       0
       2       6       0       0
       3       6       0       0

-----VENC CHNL INFO------------------------------------------------------------
      ID         Inq       InqOk       Start     StartOk      Config     VencInt  ChaResLost    OverLoad
       0    12917580     2369952     2369952     2369952     2369952     2369952           0           0
       1    10852621     2369448     2369448     2369448     2369448     2369448           0           0
       2    15793662           0           0           0           0           0           0           0
       3      552650           0           0           0           0           0           0           0

-----VENC CROP INFO------------------------------------------------------------
      ID  CropEn  StartX  StartY   Width  Height
       0       N       0       0       0       0
       1       N       0       0       0       0
       2       N       0       0       0       0
       3       N       0       0       0       0

-----VENC STREAM STATE------------------------------------------------------------
      ID     FreeCnt     BusyCnt     UserCnt     UserGet     UserRls    GetTimes    Interval   FrameRate
       0           6           0           0     2607054     2607054     2368476       58099          15
       1           1           0           0     2606460     2606460     2369043       60062          15
       2           0           0           0           0           0           0           0           0
       3           1           0           0          86          86          43    69800990           0

phucn, 2017-03-31:
thanks @shadow-1 . the IPC is using SDK Version Hi3518EV200 SPC. V1.0.4.0

diogoribeiro09, 2017-04-05:
Is there any Discord channel or something that we can openly talk about and maybe cooperate? Really looking forward to have this hack working on my dome.

batrarobin, 2017-04-05:
@diogoribeiro09 

Have you tried shadow-1`s hack for 1080p dome camera? It's a working hack and quite simple to flash.

batrarobin, 2017-04-05:
https://github.com/shadow-1/yi-hack-v3

amitnavik, 2017-04-07:
To all those who have been working hard but I need help soo plzz I m noob I need guidance plzz hope people and community are here to help .... So now listen my brother brought from China we want to use this in India  cn anyone help me how to unlock the firmware or get compatible toh my region plzz thanks in advance it will of total waste of money plzz.... Hope any one understand it's Yi dome camera white colour asap

batrarobin, 2017-04-07:
@amitnavik
So we can help you, you have to help us first. On your white dome 720p camera,  you will need open it and attach wires to its serial links to copy some files and send us. 

Read @shadow-1's last comment in this post:
https://github.com/shadow-1/yi-hack-v3/issues/1

You will need to attach wires as shown in photos here:
https://github.com/fritz-smh/yi-hack/issues/124

diogoribeiro09, 2017-04-07:
Thanks for the help @batrarobin. Either way i was expecting no wiring involved. Thought it would be as easy as copying file to the SD card like other hacks.

shadow-1, 2017-04-07:
@diogoribeiro09 
Installing the firmware is as easy as copying files to the SD card. However I don't have this particular camera. Without a backup image of the camera, I can not create a custom firmware that is compatible with the Yi Dome (white).

If you can help. I have detailed the initial steps that need to be done in my post below:
https://github.com/shadow-1/yi-hack-v3/issues/1#issuecomment-292124058

amitnavik, 2017-04-08:
What further  procedure to do after soldering all detailed steps

amitnavik, 2017-04-08:
@batrarobin what after wiring stuff...? Explain in detail which os to use Linux windows and all that memory card stuff rootfs kind n ol
..rply

batrarobin, 2017-04-08:
@amitnavik 
Have a good read of that above thread from @shadow1, he will provide instructions on what needs to be copied from the camera. 

No Linux required, just a Windows computer with PuTTY installed will work.

amitnavik, 2017-04-08:
@shadow1 can u explain me how can I or what cn I do to make camera work in India here how to do all that 8 points strip 22 bit from here I just don't get it

shadow-1, 2017-04-08:
@amitnavik 
After wiring the serial port, you need to connect the camera to your computer using a USB to Serial converter.
You need to use a terminal program to access the camera through the USB to Serial converter.
All operating systems are compatible. However I like to use PuTTY on Windows.

I will write the following instructions as if you are using PuTTY on Windows.
1. Connect the USB to Serial converter to your computer and to the camera.
2. Start the camera with a microSD card inserted.
3. Find out which COM port your USB to Serial converter is using. This is found in the Device Manager.
4. Start Putty. Enter the following options
- Connection Type: Serial
- Serial Line: **\<Your COM Port\>** _Example: COM3_
- Speed: 115200
5. Press Open. _You will now see lots of text on the screen come up. If not, try swapping your Rx and Tx connections as it is the most common issue. Press enter a few times to see the command line._
6. Type the following commands into the terminal:
`dd if=/dev/mtd0 of=/tmp/sd/mtd0
dd if=/dev/mtd1 of=/tmp/sd/mtd1
dd if=/dev/mtd2 of=/tmp/sd/mtd2
dd if=/dev/mtd3 of=/tmp/sd/mtd3
dd if=/dev/mtd4 of=/tmp/sd/mtd4
dd if=/dev/mtd5 of=/tmp/sd/mtd5
dd if=/dev/mtd6 of=/tmp/sd/mtd6
dd if=/dev/mtd7 of=/tmp/sd/mtd7`
---
Once this is done, you will need to send me the files mtd0, mtd1, mtd2, mtd3, mtd4, mtd5, mtd6, mtd7 from your microSD card. From those files I can create a custom firmware compatible with Yi Dome (white).

justinchuu, 2017-04-15:
Hi! I have the Yi Dome Camera 1080P. Can you help me update/ have a firmware that will work in the philippines? I also have the same problem as you. camera says "This camera can only be used in China" Thanks in advance!!

batrarobin, 2017-04-15:
@justinchuu

Time to do a bit of a reading:
https://github.com/shadow-1/yi-hack-v3/issues/1

justinchuu, 2017-04-15:
@batrarobin thanks! will try reading it tomorrow. its late here. Didn't understand this thread that much. sorry for being noob :) just 1 question. can I do it without rewiring? :)

batrarobin, 2017-04-16:
@justinchuu

Yes, no wiring required.  It's a simple process to flash the custom firmware, all you need is a microSD card of capacity 16GB or less.

justinchuu, 2017-04-16:
@batrarobin Got it working! Thanks!!! :)

closed 2017-04-16 by batrarobin

amitnavik, 2017-04-23:
i soldered now what??????????? after i connect this usb it show usb device  not recognized and will i have to buy serial to usb converter
[image]

amitnavik, 2017-04-23:
@shadow-1  i soldered now what??????????? after i connect this usb it show usb device not recognized and will i have to buy serial to usb converter
[image]

shadow-1, 2017-04-23:
@amitnavik 
I can't see the USB to serial converter in your cable? The pads on the camera are for serial, they are not USB.

---
However someone who had terminal access helped and I created a custom firmware for this camera. It is available here:
https://github.com/shadow-1/yi-hack-v3

Make sure you do some reading as some configuration is required to defeat the region ban.

amitnavik, 2017-04-23:
@shadow-1  
as i cn se u have created this custom firmware hope this might work  cn i try this i downloaded this zip what next ?/???/?? what is rootfs all i have to do is extarct the zip in memory card and paste and then boot camera is it  like that  ??????????

amitnavik, 2017-04-23:
all these thing is damn confusing what is all rootfs 20 home and all that stuff plzzz make me understand and how will i access all these things can any1 make video on this or anything plzzzz itz has to be like ato z how when  like this...!plzzz and thaks for support each and every one

danishno1, 2017-04-24:
I think @amitnavik is right can you plz make a video for demonstration because we are new and stuck buying this product although it was best in china but sucks in india its a humble request to you guys plz make a video or something easier to follow so that any one can follow and make is camera work again and thank for this much help

shadow-1, 2017-04-25:
@amitnavik @danishno1 
I have updated the readme file on my Github page to make it much clearer what steps need to be followed.
https://github.com/shadow-1/yi-hack-v3#getting-started---step-by-step-guide

Feel free to have a go. Be sure to read each step carefully.

Unfortunately at this stage I can not make a video demonstration. However perhaps someone else will in the future.

johnnyxvii, 2017-05-22:
Hi @shadow-1 ,

Thank you so much for taking the time to hack this and provide the solution for everyone else. It took a few tries, but I was able to get this working.

shadow-1, 2017-05-27:
@johnnyxvii 
Please go to my Github page and post an issue there rather than this closed thread.
A step-by-step guide has been written for this firmware. It is available here:
https://github.com/shadow-1/yi-hack-v3#getting-started---step-by-step-guide

When posting an issue, can you explain exactly at which step you are having trouble with.

baiyunfei00, 2017-06-19:
Hi @shadow-1 , i having issue in finding ProxyList , it cannot be connect.  How to know when it connected?

batrarobin, 2017-06-19:
@baiyunfei00

I can only tell you that you're posting in a wrong thread that was closed a long time ago.

wabesamis, 2018-10-18:
@shadow-1 - i have an international version of YI 1080p Dome 360  Camera i want to convert it to Chinese version to support the auto save video to my mi router 3. is it possible to convert it?