fritz-smh/yi-hack #126: Issue connecting - Wifi or hardware? (US model)

kasperb2, 2017-01-22:
Hi guys, 

After going through all the issues on the forum, I'm still not able to use this hack for my YI Home Camera as it doesn't seem to connect to the Wifi. Note that I did get the blue led earlier this week, but was not able to view the RTSP stream, or open the camera in my browser through the IP. So my assumption is that something was wrong at that time as well.

I'm trying to connect to my 2.4GHZ wifi, with WPA/WPA2 security. I'm using the US version of the camera, and have flashed the latest US firmware. 

What I've tried:
- Flash an older version of the firmware, and try to install the hack.
- Set a static IP for the camera
- Checked for IP address conflicts (there are none)
- Double checked my settings in the wpa_supplicant.conf and yi-hack.cfg
- Tried different versions of the hack that were posted in the forum; none have worked (i.e. hermitcrabs)

The only thing that worked is the firmware from http://xiaoyi.querex.be/. Unfortunately, with this version I have to reconnected the camera once you unplug it, and doesn't allow me to modify some other settings (LED, etc.). 

Could it be that this is a newer version of the camera, with different hardware and the hack therefore doesn't work?

Find below the log file. Thanks in advance for your insight.

--- 
Starting to log...
Start telnet server...
MTK 7601
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.6.1Q_201607271501
    [kernel]
    version=3-0.8
    provider=1
    devtype=1
Firmware letter is : 'Q'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Sep  9  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Sep  9  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Sep  9  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Sep  9  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Sep  9  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=255  (0 is ok)
Wifi configuration answer: Successfully initialized wpa_supplicant
Line 4: failed to parse ssid '“MY SSID”'.
Line 4: failed to parse ssid '“MY SSID”'.
Line 10: Invalid PSK '“MY PASSWORD”'.
Line 10: failed to parse psk '“MY PASSWORD”'.
Line 11: failed to parse network block.
Failed to read or parse configuration '/home/wpa_supplicant.conf'.
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:96:76:E2  
              inet addr:192.168.0.190  Bcast:192.168.0.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe96:76e2/64 Scope:Link
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:161 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:42744 (41.7 KiB)  TX bytes:0 (0.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 00:00:58 GMT 1970
Done
New datetime is Thu Jan  1 00:01:29 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1778 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1805 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1908 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  615 root         0 SWN  [jffs2_gcd_mtd4]
  619 root      1712 S    /bin/sh /etc/init.d/rcS
  621 root         0 SW   [mmcqd/0]
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1716 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  682 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  821 root         0 SW   [RtmpTimerTask]
  822 root         0 SW   [RtmpMlmeTask]
  823 root         0 SW   [RtmpCmdQTask]
  824 root         0 SW   [RtmpWscTask]
  954 root         0 SW   [hidog]
  992 root      1164 S    ./peripheral
  993 root      1144 S <  ./dispatch
  994 root      1068 S    ./exnet
 1778 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1805 root      1408 S    ./serverM 80
 1823 root      3360 S    ./record_event
 1824 root     14732 S    ./mp4record 60
 1853 root      1716 S    /usr/sbin/crond -b
 1902 root       904 S    /home/led_ctl -boff -yfast
 1903 root     94444 S    ./rmm
 1908 root     27252 S N  ./rtspsvrM
 2019 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    844.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.1M    800.0K  91% /home
tmpfs                    32.0M     48.0K     32.0M   0% /tmp
/dev/hd1                 62.5G     13.2M     62.5G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 62.5G     13.2M     62.5G   0% /tmp/hd1/test/http/record

---
NETWORK CONFIGURATION:

### Network configuration
# Don't forget to also fill the file wpa_supplicant.conf for the wifi configuration
IP=192.168.0.150
NETMASK=255.255.255.0
GATEWAY=192.168.0.1
NAMESERVER=192.168.0.150

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid=“MY SSID”
scan_ssid=1
proto=WPA2
key_mgmt=WPA-PSK
pairwise=CCMP
group=CCMP
psk=“MY PASS”
}

txparadise, 2017-01-22:
I also have the US version of the yi-cam which i recently purchased of Amazon and was having similar issues , follow instructions below and you should be good to go ( have this working on two cams ) ,my wifi is running WPA2-psk , AES . 

1) go to ( sitehttp://xiaoyi.querex.be/firmwares/ )  and flash ur cam with (  1.8.5.1N_201601071352 )  the **Original International Version** 
2) once you've done that now flash this guys hack ( https://github.com/hermitcrabslab/yi-hack/blob/master/README.md ) 

3) Be sure to have your ssid,password, ip address,gateway and name server are input correctly. ( name server address is the same as your gateway address ) see my examples 

( IP=192.168.1.113
NETMASK=255.255.255.0
GATEWAY=192.168.1.1
NAMESERVER=192.168.1.1 ) 

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="Verizon-345LVW-767C"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="16240899"
}

let me know how this works for you.

kasperb2, 2017-01-23:
Thanks for the suggestion. I followed all steps (twice!), but unfortunately still receiving the same error. I also tried to flash the latest US firmware, and install the hack, which also didn't work.

I double checked my wifi as well; WPA-psk, AES. I also made my Wifi password shorted (<10 characters), in case that would make a difference (it did not).

The following does happen:
- Startup music
- Waiting to connect...
- Connecting to Wifi... blinking blue light
- Blue light is on for ~5 seconds.
- Orange light starts blinking again
- Nothing

Updated logs:

Starting to log...
Start telnet server...
MTK 7601
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.6.1Q_201607271501
    [kernel]
    version=3-0.8
    provider=1
    devtype=1
Firmware letter is : 'Q'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Sep  9  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Sep  9  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Sep  9  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Sep  9  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Sep  9  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=255  (0 is ok)
Wifi configuration answer: Successfully initialized wpa_supplicant
Line 4: failed to parse ssid '“MY SSID”'.
Line 4: failed to parse ssid '“MY SSID”'.
Line 10: Invalid PSK '“MY PASS”'.
Line 10: failed to parse psk '“MY PASS”'.
Line 11: failed to parse network block.
Failed to read or parse configuration '/home/wpa_supplicant.conf'.
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:96:76:E2  
              inet addr:192.168.0.125  Bcast:192.168.0.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe96:76e2/64 Scope:Link
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:112 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:26328 (25.7 KiB)  TX bytes:0 (0.0 B)
    
Do network configuration 2/2 (DNS)
[... 92 lines cut ...]
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 29.8G     14.1M     29.8G   0% /tmp/hd1/test/http/record

---

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid=“MY SSID”
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk=“MY PASS”
}

---

Network configuration
# Don't forget to also fill the file wpa_supplicant.conf for the wifi configuration
IP=192.168.0.125
NETMASK=255.255.255.0
GATEWAY=192.168.0.1
NAMESERVER=192.168.0.1

death2all110, 2017-01-23:
Hi @kasperb2 

The issue is your SSID and PSK lines.

```
...
ssid=“MY SSID”
...
psk=“MY PASS”
...
```
The quotes around them are causing the issue.
Delete the quoutes and re-type them.  Instead of being slightly slanted they should be straight up and down.

```
...
ssid="MY SSID"
...
psk="MY PASS"
...
```

kasperb2, 2017-01-23:
@death2all110 Thanks so much - that resolved the issue. The camera is now successfully connected.

mrgeophysics, 2017-11-02:
are you using the 720 p or the 1080 p version?