fritz-smh/yi-hack #127: Disable Motion detection/Recording

Savkin, 2017-01-23:
I think an option to disable motion detection and video record might be usefull. I am using Xiaomi cameras together with Zoneminder and Synology Surv station where all the motion analisys is performed centrally by software.
Is there already such a capability to disable recording?
The hack is brilliant nevertheless!!!

death2all110, 2017-01-23:
There is not an option to turn it on and off, but you can probably comment out the ./record_event & and the ./mp4record 60 & lines

Savkin, 2017-01-23:
Where do i find these lines?

death2all110, 2017-01-23:
Those are found in sd/test/equip_test.sh

cheahkhing, 2017-01-30:
@Savkin I am looking for ways to utilize my Synology NAS as well, can you share with me on how you use Zoneminder and Synology Surv station?
Sorry for hijacking this thread. Anyway, this repo of hacks for Yi camera is awesome!  thanks.