fritz-smh/yi-hack #128: "Waiting to connect" Orange light blinking

cdb8457, 2017-01-26:
I apologize this is my first time setting this thing up ever, and on top of that with a hack.

These are the steps I took:

Formatted my SD card

Downloaded the zip file

put 

[image]

SD file on the SD card

Changed the SSID and Password and Static

[image]

[image]

Saved

Then I took the SD card out put it in the YI cam. Plugged it in. Orange Blinking light

Version is YHS -113-IR

hyptechdev2015, 2017-01-27:
[image]

and "home", and test folder to newly formatted SD card

cdb8457, 2017-01-29:
OK, those are in the same file behind that Test file.  

If I have 

[image]

Do i need to change it from TKIP to WPA Personal?

[image]

cdb8457, 2017-01-29:
If its yellow blinking does mean it cant connect? I if figure this out. I might make a YouTube video explaining it to my dumb self.

cdb8457, 2017-01-29:
It starts as yellow solid light. Then starts blue blinking. Two Clicks, blue blinking, Blue solid for like a few seconds, then finishes with yellow blinking. Any help?

asmile60, 2017-01-30:
same with me here at first is ok nothing problem after use for few days and then suddenly the image is upside down, and I restart the camera, and now yellow blinking after solid blue.

qelanhari, 2017-01-31:
I don't know if it's your problem, but remember that the camera only connects to 2.4ghz wifi, not 5ghz.

cdb8457, 2017-01-31:
Thanks for the responds. I have only been connecting to the 2.4ghz. I am going to try again sometime soon.

death2all110, 2017-02-02:
Can you post a copy of the log file? Should be on the SD in the test folder.

cdb8457, 2017-02-04:
[image]

[image]

[image]

Sorry for the late response. I appreciate the help!

death2all110, 2017-02-04:
@cdb8457 I see the issue.

in wpasupplicant.conf you need single quotes around the SSID and PSK
Example:

```
...
SSID='ManCave'
...
PSK='Vikings@58'
...
```

cdb8457, 2017-02-05:
Well...That is slightly embarrassing . I will Try that today......

cdb8457, 2017-02-05:
Well I have given up. I have done everything correct. It get to the last blue light. Hangs for a little bit then goes to yellow blinking.

death2all110, 2017-02-05:
Can you post the log? Are you using the same equip_test.sh in this repo, or my fork, or an edited version? You can post the log to gist.github.com and then share the link here, that will make it easier.

cdb8457, 2017-02-07:
<script src="https://gist.github.com/cdb8457/c8b483d14584c075bf9a3d6bd00eebc0.js"></script>

I believe I am using the one that is in the zip on this git hub page.

cdb8457, 2017-02-07:
[c8b483d14584c075bf9a3d6bd00eebc0-b864f8e44dcf2627857b7662848a323f1cd380d2.zip](https://github.com/fritz-smh/yi-hack/files/756373/c8b483d14584c075bf9a3d6bd00eebc0-b864f8e44dcf2627857b7662848a323f1cd380d2.zip)

cdb8457, 2017-02-07:
I apologize with trying to get this log on here. Can you view it?

[log.txt](https://github.com/fritz-smh/yi-hack/files/756377/log.txt)

death2all110, 2017-02-08:
Sorry for the late reply!

I actually advised you incorrectly on you wpa_supplicant.conf. My aplogies.

Try using quotes instead of single quotes.

```
...
SSID="ManCave"
...
PSK="Vikings@58"
...
```

cdb8457, 2017-02-09:
I tired it and thought it was going to work .....it did the blue blinking for a lot longer this time. I will keep trying. Now I have never flashed anything on it. All I have ever done is put those files on the sd card.

cdb8457, 2017-02-09:
Is there supposed to be talking to me while this sets up? There is none.

death2all110, 2017-02-09:
@cdb8457 it didnt play any sound at all during boot?

Remove the files from the SDCard temporarily, re-format to fat32 and place the files back on and put it back in the camera. 

Also, just to make sure, you did double quotes right? because the actual quotation marks `"` is seen as 1 character where as two single quotes `''` is seen as two.

cdb8457, 2017-02-09:
I used Shift + '  ="   I used " that.

death2all110, 2017-02-09:
@cdb8457 ok good, just making sure.

So at this point, what I recommend doing is: 

- unplug the camera
- remove the sd from the camera
- move the files from the SDCard temporarily
- re-format the sd to fat32 
- place the files back on the sd 
- put the sd back in the camera
- plug the camera in.

If that still doesn't work, please attach the log file as you did before.

cdb8457, 2017-02-09:
[log.txt](https://github.com/fritz-smh/yi-hack/files/765230/log.txt)

No sound when starting up. Just Clicks noises. Just did what was above

death2all110, 2017-02-09:
What color is the LED?

According to the log, the wifi is connected.

cdb8457, 2017-02-09:
It starts as orange solid, Clicks, blue blinking, clicks twice. blue blinking for a bit. Then solid blue for 8 seconds, then goes back to blinking orange forever.

death2all110, 2017-02-09:
Now that I think of it, has the sound worked at all? Because the path to the voice files in equip_test.sh is actually incorrect it all points to /home/hd1/voice instead of /home/hd1/test/voice.

2 options. move the voice subfolder to the root of the SD, or correct the paths in the script by following [this commit.](https://github.com/fritz-smh/yi-hack/pull/106/commits/d549a48a726a5e6a718e1b17a8430f865b894806)

cdb8457, 2017-02-09:
[image]
 This is what Happens after I start up. New files pop up.

death2all110, 2017-02-09:
I'm guessing the reason for the orange blinking is that the last ping command is failing. 

In equip_test.sh you will see the following lines:
```
### Check if reach gateway and notify
ping -c1 -W2 $(get_config GATEWAY) > /dev/null
if [ 0 -eq $? ]; then
    led $(get_config LED_WHEN_READY)
    /home/rmm "/home/hd1/voice/success.g726" 1
else
    led -boff -yfast
fi
```

go ahead and change to the following:

```
### Check if reach gateway and notify
###ping -c1 -W2 $(get_config GATEWAY) > /dev/null
###if [ 0 -eq $? ]; then
    led $(get_config LED_WHEN_READY)
    /home/rmm "/home/hd1/voice/success.g726" 1
###else
###    led -boff -yfast
###fi
```
Also, to fix the voice you have 2 options. move the voice subfolder to the root of the SD, or correct the paths in the script by following [this commit.](https://github.com/fritz-smh/yi-hack/pull/106/commits/d549a48a726a5e6a718e1b17a8430f865b894806)

cdb8457, 2017-02-09:
[image]
 hmm"Else is not even on mine.

death2all110, 2017-02-09:
Theres actually 2 ping sections the one I'm referring to is the second one near line 399. The section you screenshot is the first one.

cdb8457, 2017-02-09:
k, fix both things!

death2all110, 2017-02-09:
Awesome! Glad to hear it man! You should see a button to mark the issue as closed. Have a great day!

cdb8457, 2017-02-09:
How did those files get changed?...Did i do it?

cdb8457, 2017-02-10:
Thanks for the Help!

closed 2017-02-10 by cdb8457

death2all110, 2017-02-10:
@cdb8457 Nah, the equip_test.sh has the wrong paths for sound to begin with.

Structure3creative, 2017-02-24:
Had same issue. Did the above with audio of successful pairing. Solid blue LED, webpage won't load. No ping.

ITSolution4U, 2020-12-11:
Please watch this video for Solution.
https://www.youtube.com/watch?v=ZaKGDeC4_oM