fritz-smh/yi-hack #129: MQTT

kdudkov, 2017-01-27:
Hi,

I think it would be great if yi could post mqtt message in case of motion detected.

technophile1, 2017-01-30:
That would be amazing!  I've been looking into DVR based solutions, but mqtt or simply ONVIF motion would be ideal.

yajrendrag, 2017-02-14:
i came here today to specifically ask for this so pleased to pile on... i hacked together a solution to do this, but it's clearly circuitous... i copy the motion video files from the camera to a separate ftp server and watch for new files being added to the ftp server using inotifywait.  the inotifywait function is in a bash script on the machine running ftp... when it identifies a new file being created, it publishes an mqtt message...

i tried to figure out how to install the mosquitto_pub client on the camera, but couldn't figure out / couldn't find any documentation on how to build or install...

airdrummingfool, 2017-02-15:
I've added POST notifications from the camera to a LAN HTTP server on motion detected/not detected (code hasn't been cleaned up or pushed yet). I'm not very familiar with mqtt, but if there is an HTTP interface to it (or an easy way to add one) then I should be able to integrate it in.

kdudkov, 2017-02-15:
If this cam has lua interpreter (as many routers/cams do) it's easy to find simple lua script for mqtt publishing. But HTTP POST is a good option too.

yajrendrag, 2017-02-16:
@airdrummingfool - i can work with http POST - am integrating to homeassistant and it has a rest API that i could point the POST to - and an api endpoint that will publish the POST data as mqtt, so would work well.  so, would be great to have it integrated...  

@kdudkov - doesn't seem to have a lua interpreter

airdrummingfool, 2017-02-18:
Does the POST body need to have any specific structure? Right now it's just plaintext `motion=yes` or `motion=no`, but I'm open to changing that up a bit if necessary.

yajrendrag, 2017-02-21:
Ideally, yes.. the user would likely want to customize their own message, e.g., adding date, time &/or a camera ID/location....  In the preceding days, I was able to figure out a way to do this using netcat, however it doesn't work with https - only http.  It's a bit brittle tho' since you have to first determine what is passing back and forth (e.g, using curl with trace on a fully featured Linux).

Is your solution more robust?

airdrummingfool, 2017-02-24:
I should be able to add the current time and possibly a camera ID (customizable in the config file) to the payload. My solution is built into an improved version of the `check_motion.sh` [script](https://github.com/fritz-smh/yi-hack/blob/master/sd/test/check_motion.sh).

yajrendrag, 2017-02-25:
ok - i did same - just using netcat to generate the http POST... so, think i am good for now, thanks.

gl-lamhnguyen, 2017-04-24:
@airdrummingfool how/where do you implement the POST? I'd like some kind of push notification to another home server so I can automate things.