fritz-smh/yi-hack #130: Disable Startup Sound(s)

CaptInsano, 2017-02-01:
Yi Cam 1 720p US model
Firmware: 1.8.6.1B_201604071151
Yi-Hack version: github.com/death2all110/yi-hack as of 1st Feb 2017

Is it possible to disable the startup music on yi-hack?

I have looked through the equip_test.sh and I saw these lines which I thought were relating to sounds only:

`### Let ppl hear that we start
/home/rmm "/home/hd1/test/voice/welcome.g726" 1
/home/rmm "/home/hd1/test/voice/wait.g726" 1`

So I tried to comment out the first line to match the following

`### Let ppl hear that we start
#/home/rmm "/home/hd1/test/voice/welcome.g726" 1
/home/rmm "/home/hd1/test/voice/wait.g726" 1`

While this **does** remove the startup music on booting the camera, the camera now does not successfully connect to wifi. Is there more to the /test/voice/*.g726 files than audio?

Is is possible for an option in yi-hack.cfg to remove the boot sounds? 
(Disable Music/Disable Wifi anouncements/Disable all sounds)

death2all110, 2017-02-02:
Hey @CaptInsano can you post the contents of the log.txt file in the test directory on the SD card?

CaptInsano, 2017-02-03:
Apologies,

I reformatted my SD card and retried the above with success

Issue Closed (probably user derp!)

Sorry again

closed 2017-02-03 by CaptInsano