fritz-smh/yi-hack #131: RTSP streams stop working after a while

lazarevski, 2017-02-05:
Somewhat similar (in terms of when it happens) issue with #85, with the difference being that everything else works except the rtsp streams.

Namely, the camera rtsp streams stop working after a random time. But I can ping the camera, connect via telnet/ftp, http server also works.

When i try to open any of the three streams in VLC, i get:
```
VLC is unable to open the MRL 'rtsp://192.168.100.75:554/ch0_0.h264'. Check the log for details.
Your input can't be opened
```

I have the EU version of the camera (22EUT4A00S). Any thoughts on why this happens?

Not sure if this is the correct log, but here it is:
```
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:8E:C9:92  
              inet addr:192.168.100.75  Bcast:192.168.100.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe8e:c992/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:84 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:16840 (16.4 KiB)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
PING 0.uk.pool.ntp.org (178.79.160.57): 56 data bytes
64 bytes from 178.79.160.57: seq=0 ttl=49 time=67.386 ms

--- 0.uk.pool.ntp.org ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 67.386/67.386/67.386 ms
Previous datetime is Thu Jan  1 01:00:24 CET 1970
Done
New datetime is Thu Jan  1 01:00:24 CET 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1245 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1272 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1346 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  682 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  816 root         0 SW   [RtmpTimerTask]
  817 root         0 SW   [RtmpMlmeTask]
  818 root         0 SW   [RtmpCmdQTask]
  819 root         0 SW   [RtmpWscTask]
  949 root         0 SW   [hidog]
  987 root      1164 S    ./peripheral
  988 root      1144 S <  ./dispatch
  989 root      1060 S    ./exnet
 1123 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1208 root      1708 S    ntpd -q -p 0.uk.pool.ntp.org
 1245 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1272 root      1408 S    ./serverM 80
 1318 root      1716 S    /usr/sbin/crond -b
 1331 root       904 S    /home/led_ctl -boff -boff
 1341 root     97472 S    ./rmm
 1346 root     30892 S N  ./rtspsvrM
 1437 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    864.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.1M    844.0K  91% /home
tmpfs                    32.0M     28.0K     32.0M   0% /tmp
/dev/hd1                 29.0G      9.5M     28.9G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 29.0G      9.5M     28.9G   0% /tmp/hd1/test/http/record

```

TomFaulkner, 2017-02-06:
I haven't looked at logs, but I have tried two of these and can't get them to stay online for very long. Maybe a few hours at best. Most of the time VLC won't even play the video beyond ten seconds, though audio continues.

lazarevski, 2017-02-07:
Few hours ago i implemented the fix mentioned on #108. So far no issues. I'll report back in a few days or sooner if the streams stop working.

lazarevski, 2017-02-09:
3 days up-time of the camera so far. It seems that the fix on #108 did the trick.

lazarevski, 2017-02-14:
I'm afraid i need to reopen this issue since the problem still persists. I've experienced the same issue three times since my last comment. I can still ping the camera, telnet/ftp/http all work except the rtsp streams.

Any ideas? Anyone willing and able to write a cron job (if at possible) to restart the camera at specific intervals?

lazarevski, 2017-02-19:
[image]

keith721, 2017-05-27:
Try some variation on [this script](http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/#rtsp_check) from BobbyRomeo