fritz-smh/yi-hack #132: How to view rstp steam?

cdb8457, 2017-02-10:
I have no clue on how to look at this Yi cam. 

[image]

I get to this, but none of the blue links work.

closed 2017-02-10 by cdb8457

Structure3creative, 2017-02-27:
Did you ever figure out how to view the rtsp stream or get the links to work? Same issue here.

joozy, 2017-02-28:
You'd usually use a video viewer, VLC, is a popular one.

Open Network Stream
Enter rtsp://cameraip:port/ch0_0.h264 or ch0_1.h264 or ch0.264.

Structure3creative, 2017-02-28:
I've tried the [VXG Player](https://www.videoexpertsgroup.com/o/chrome-media-player-plug-in/) and the furthest it has gone is an image still. Will [VLC](http://www.videolan.org/vlc/) work similarly?
 
The server page loads, and none of the links connect to the stream, motion, or record.

joozy, 2017-03-01:
I'm getting a live video stream from mine, throug VLC.
Have not tried VXG.

I've noted there was an issue, from this github, that the live stream will "terminate" after X minutes eventhough the camera was still running (the user could telnet into the camera).