fritz-smh/yi-hack #133: How about I could downgrade my yi1080p(version:2.0).

Lingdu0, 2017-02-10:
How about I could downgrade my yi1080p(version:2.0).I used this tool to try to hack my camera,but i found it doesnot work.

littleben, 2017-02-11:
+1.
Or any solution for version2.0?

Borillion, 2017-02-23:
For what its worth the chip is slightly different. Same Hi3518 ERBCV200 in this one vs ERBCV100 in the older. Also binwalk results in no output when run on the firmware file, perhaps its encrypted.

xmflsct, 2017-03-03:
@Lingdu0 @littleben Make sure you are referring to the second version, but not first version with 1080p upgrade. For hacking the second version, please refer to [yi-hack-v2](https://github.com/niclet/yi-hack-v2).

@Borillion The home image of this hack can be unwrapped as a uImage, then extract as a jffs2 image.

Lingdu0, 2017-03-04:
@xmflsct first version with 1080p upgrade。the firmwave maybe have been encrypted。

xmflsct, 2017-04-10:
720p RTSP streaming is possible now. [xmflsct/yi-hack-1080p](https://github.com/xmflsct/yi-hack-1080p)

For the other approach of hacking this camera, please refer to the acknowledgement section in above project.