fritz-smh/yi-hack #135: How do I restore to original state?

kingamoon, 2017-02-15:
I followed the instructions to uninstall the hack. I then removed the SD card, and reset the camera. When I attempt to re-setup the camera with the Yi App, it initially works, then it just gets stuck at 50%, then times out when attempting to load the camera on phone. I just want to go back to using their app and reset everything back to factory original. I can't see to get that working on any of my 3 cameras.

jritza, 2017-02-26:
I am also experiencing this issue. 

I installed the hack. The hack was installed perfectly and functioned well, however I decided to restore the camera until I can have a more formal recording setup for the camera. 

As the readme suggests, I removed my SD card from my Yi camera and replaced the text within the "equip_test.sh". Not touching any of the other files, I placed my SD card back into the camera and booted it up. Upon testing, I became unable to connect to the hack's IP page or the official Yi camera app. 

So I checked out the SD card on my computer once again but this time I formatted the SD card and then booted the camera with the empty SD card. The camera booted as normal, but still no connection to the hack or official app. The official app oddly reported having only connected at 50% but doesn't proceed further. 

I have tried to reset the camera itself and start the connect sequence all over with the official Yi home app. The camera now speaks Chinese oddly enough, before it spoke English, and it doesn't work with the app. 

Please, how do I restore my camera to how it was before the hack?

nehaleem, 2017-03-12:
Maybe thats not this case, but i saw similar behavior with newest version of app. I will scan 2D barcode and camera will not speak whole line and app will timeout again and again. I downloaded then older version of apk (4.0.8) from internet (i think its better version!) and camera will speak properly after scan 2D code and connect after ~30 seconds (initial setup).

Note: I never inserted hacked sd card.. just FYI

issagak, 2017-04-10:
Thank you so much Jritza - You steps resolved my problem.

wizard7926, 2017-05-25:
@jritza I took similar steps to your solution, and I can use the app now.. but I'm finding bugs, like I get "old" notifications at all times of the day, and I also can't go backwards and examine non-notification movements, it just zaps me to the live video feed. Any of these issues present for you?

juan22a91, 2017-06-07:
Hi @jritza , I am trying to go back to original so I can connect the camera to the Yi Home app. I've done all the steps you describe but still can't connect to Wi-Fi. When the original firmware boots I follow the process to connect to Wi-Fi and after some time it sends the message Wi-Fi password error. I know for sure the password is OK. With the hack I wasn't able to connect either after I tried to flash back to original firmware.

When you changed the equip_test file did you erase everything from it and paste only the code in the readme file?

tvincent87, 2018-06-16:
It's working! It's working! I've been trying to view this camera through hass.io on and off for 3 weeks easy. I went to 0.1.5 (not supported) without realizing it. Then I had problems going back. I kept finding logs on the sd after booting and it failing soon after scanning the QR code and starting all over. After resetting with & without sd and countless reformatting, the firmware finally took! I just formatted the sd card, copied rootfs_y18 and home_y18 for 0.1.4 beta2 (the root was 936 kb and the home was 7314 kb if that helps) directly to the freshly formatted SD card, ejecting, putting it in the cam, plugging it in, letting it try to boot up, failing, pressing the reset button on the back, and it booted up like a charm! I didn't wake up the wife with my "YES!!!" like I did when the tasmota firmware successfully flashed to a sonoff, but I'm pumped! If I can assist, let me know! Gotta do the same to my other yi now.