fritz-smh/yi-hack #140: During startup LED starts blinking orange.

Structure3creative, 2017-02-24:
Trying this code out and and I'm not sure if I'm adding the network info correctly. The startup process starts to cycle through the blinking blue, and to a solid blue, and then goes to a blinking orange - and stays there. Any ideas as to what I should try to get it working?

nauska, 2017-02-24:
I just encountered the same problem, and you need to change the time server. If you look at the log you can see that the previous datetime and the new datetime are the same. Go to [http://www.pool.ntp.org/zone/@](url) and pick a new one that is close to you, and change it in the yi-hack.cfg.

`Get time from a NTP server...
But first, test the NTP server '0.us.pool.ntp.org':
Previous datetime is Wed Dec 31 17:00:30 MST 1969
Done
New datetime is Wed Dec 31 17:00:30 MST 1969`

Structure3creative, 2017-02-24:
I tried using  'pool.ntp.org' as recommended on http://www.pool.ntp.org/zone/@ ... audio says "you can start using your camera now" and the webpage loads - though no vids are coming through on the streams, API, or record. Ping works.

nauska, 2017-02-26:
I had the same problem as you, and I did the changes in the thread you link to (basically move the voices folder to root instead of test and comment out a couple of lines around line 399) and I changed the NTP server to one that gave me the correct time, and then it worked. Sorry, but right now I don't know what might be wrong.

death2all110, 2017-03-02:
Are you using DHCP or Manual config?

Structure3creative, 2017-03-02:
DHCP

closed 2017-03-30 by Structure3creative