fritz-smh/yi-hack #141: Progress with xiaoyi ants yi 1080p home camera, not version 2

xcray, 2017-02-25:
I've got a xiaoyi 1080p webcam (CN version), it's upgrade of version 1 (not version 2!), looks very same as the 720p type (version 1) and with the price of only RMB169. 
This is the international version (or same as CN version?): https://www.yitechnology.com/yi-1080p-home-camera
[image]

The progress now:
After check inside, it's based on Hi3518 chip, almost the same as 720p version, even the linux kernels they used are quite same.
With a TTL cable, it can be logged in as root without password automatically.
From the bootlogs and /proc/mtd, we're sure that:
  - the 16MB flash is splitted to 8 partations as follow:
> 0x000000000000-0x000000040000 : "boot"
> 0x000000040000-0x000000050000 : "env"
> 0x000000050000-0x000000060000 : "conf"
> 0x000000060000-0x0000001f0000 : "os"
> 0x0000001f0000-0x000000330000 : "rootfs"
> 0x000000330000-0x000000fe0000 : "home"
> 0x000000fe0000-0x000000ff0000 : "vd1"
> 0x000000ff0000-0x000001000000 : "ver"
Copy the Code
- the hashed password is :$1$$qRPK7m23GJusamGpoGLby/, anyone could help to make a bruce force?  - the rootfs partition is quite limited, there is no telnetd and very poor left space.
  - if copy the telnetd file from 720p cam, it works fine.
  - the firmware (upgrade file of home partition, named home_y20m) is encrypted.
  - at the beginning of boot, it will looks sd card for updates of uboot, env, conf, kernel, rootfs and home, i.e. this is one possible hacking method.
  - the timer of "Hit any key to stop autoboot:" is set to 0, i.e. we can't goto the uboot interface. Is there a simple way to modify it?
  - it seems that there is watchdog to monitor changing on some files. I had replace the file /etc/passwd with blank root password and had copied telnetd to /home/app and had loaded it successfully, telnet to the camera seems quite well, but the watchdog caused continues reboot.
  - nfs is not enabled, while cifs is available to backup onto xiaomi routers.

Target:
  - the root password (only can be set to blank via TTL interface now);
  - find the methed to load telnetd from sd card during boot automatically;
  - rtsp, nfs, record to xiaomi routers just after motion detected, and so on;

could anyone give some help or hints? thanks

xcray, 2017-02-27:
Restore the flash with dd could be help if the camera fall into continues reboot.

After inspection in the home filesystem, I'd made another progress:
there is an _init.sh_ file under /home (the root of home file system), it will call _/home/app/script/factory_test.sh_,and _/home/app/script/factory_test.sh_ will check if there is the file _/tmp/sd/test/factory_test.sh_, if yes then load it and follows _sleep 1000_.

xcray, 2017-02-28:
Trying to use busybox from 720p home camera will damage the rootfs and home partitions, this will push the camera into continuous reboot! Reset or firmware update can't repair.

atmirr, 2017-02-28:
Hi @xcray, 
Do you find any solution for Yi Home 1080p that camera working outside China?

xcray, 2017-02-28:
@atmirr I am regret for your question. I am using the camera in China, thus don't have any experience.
But I have checked the firmware version, it seems that there's only one version for both Chinese and International, so I guess this type could work abroad.

xmflsct, 2017-02-28:
@TommyChausson oh yes, now I see the difference between those two. So V2 has a special speaker area on the back side if you look at the picture from the other project, which the one I have (1080p) visually is identical to the 720p one.

xcray, 2017-02-28:
@TommyChausson beside the differences of appearance, the chipset is also different!
"yi 1080p home camera" is based on Hi3518 chip, while "yi 1080p home camera 2" (we call it V2) is based on Ambarella "Cortex-A9-600MHz: S2LM" chip, and V2 is more expensive (more than double, RMB 399 vs 169).

TommyChausson, 2017-03-01:
@xcray @xmflsct 
Thank you for these details ! Now i know and i'm sure that i've got the V2 ambarella based !

xmflsct, 2017-03-02:
@xcray may I ask how did you get to the shell? I soldered the serial ports, and I can read debug outputs only, and it doesn't go into any shell. By the way, I have 18CN so things might have changed..

xcray, 2017-03-02:
@xmflsct (-: Just press return. In fact, serial port is in the shell, but there are too many messages make confusion.

xmflsct, 2017-03-02:
Hmm, the serial doesn't seem to grab any input, very weird.. I am using CoolTerm and the serial monitor of Arduino, both with no luck.. I am trying different methods now. Will keep posted.

xmflsct, 2017-03-02:
@xcray so I have made some progress. There is a fairly easy way to change the bootdelay.
[mkubootenv](https://github.com/tklauser/mkubootenv)
This tool provides a bidirectional converting uboot boot flags into an image file. I reverse my mtdblock1 dump (with dd) to a text file; change the bootdelay to any number you like; encode it into a same size image; then dd back to mtdblock1. I have succeed in having a boot delay of my camera. Sadly due to unknown reasons, I still cannot send any information to the camera... I suspect it is my TTL board's issue.

xcray, 2017-03-02:
@xmflsct Thanks for your comment!
btw, I also think there is something fails on your TTL board.

xmflsct, 2017-03-03:
@xcray Welcome :) I actually try with another TTL board, but still fails. Now assuming that they modify the hardware connection on 18CN version. Will use a scope on Saturday to figure it out.

xmflsct, 2017-03-06:
@xcray to update you, I have made some progress.
The most recent version of busybox won't work, giving segmentation fault for all commands. Then I notice that the version of busybox used by Hi3518 SDK is actually 1.16.1. So I downloaded this version from the official website, and it works! Both telnetd and ftpd work perfectly. I have forked this project to another one specializing in 1080p version, [yi-hack-1080p](https://github.com/xmflsct/yi-hack-1080p).
What it does till now is that, it can take over the boot procedure from /home/app/init.sh, which then reads wifi info from our own wpa_supplicant.conf and successfully connect to wifi, open up telnet and ftp server, without the need to scan anymore QR code. However I have encountered these questions below which are not solved yet.

1. After connecting to wifi through our own factory_test.sh, it seems like the /home/app/dispatch will re-run this process again (the QR code method). I can kill it, but then the watchdog will bark. Also, it seems like the camera module is loaded by it, different than 720p version. Am I correct?

2. Controlling LED is different than 720p, at least the led_ctl won't run. I have tried to use the v2 approach from writing to GPIO directly, but it seems the mapping is different.

3. rtsp server provided in both v1 and v2 projects won't work. Any ideas how to get going?

xmflsct, 2017-04-10:
720p RTSP streaming is possible now. [xmflsct/yi-hack-1080p](https://github.com/xmflsct/yi-hack-1080p)

For the other approach of hacking this camera, please refer to the acknowledgement section in above project.

xcray, 2017-04-15:
Thanks to @xmflsct , busybox compiled including telnetd and ftpd, thus this issue could be closed and merged into https://github.com/xmflsct/yi-hack-1080p.

closed 2017-04-15 by xcray

junhau89, 2017-06-09:
Hi guys, I seem like flash with wrong firmware. Now the Yi camera once power on and stay and turn on with yellow light. Even press reset button also no help. I already try re-flash back with firmware 1080p version firmware but can't. I guess I wrong flash with 720p firmware in 1080p camera.