fritz-smh/yi-hack #143: Include secure Curl

utya1988, 2017-03-07:
Hello. A lot of thanks for this hack. I would like to how i can include secure curl in home file. O maybe you can make branch with this feature. Vey Thanks.

karepiu, 2017-03-15:
Use this http://nemon.org/ipcam-ipr1631x/#SDK to compile curl from scratch. You will need also openssl. I can confirm that curl-7.53.1 and opessl-1.1.0e works fine. In 3 weeks I hope to branch fritz code adding curl, lighttpd, samba, smartthings capabilities.

utya1988, 2017-03-19:
@karepiu , maybe can you complie ffmpeg. I tried but failed.

karepiu, 2017-03-20:
HI, compiled it with --enable-shared --disable-yasm. Have not tested it but should work. Working dit should be /tmp/ffmpeg.  Good luck!

https://www.dropbox.com/s/6s009nfchafndbh/ffmpeg-3.2.4-arm.tar.gz?dl=0

utya1988, 2017-03-20:
@karepiu, thanks.

closed 2017-04-03 by utya1988

gl-lamhnguyen, 2017-04-25:
@utya1988 did you solve the issue with secured curl? I kept getting ssl not supported in the current version

karepiu, 2017-04-26:
Compile openssl first, then point compilator for curl to its dirs and it works like a charm.

gl-lamhnguyen, 2017-04-26:
@karepiu thx for the info, I'm not really sure how to compile it? is there a binary somewhere I can download and copy it to the SD card?