fritz-smh/yi-hack #147: How to control cam from telnet

karepiu, 2017-03-18:
HI, 

does anybody have idea how to control cam from within the filesystem without loosing cloud. I know that I can kill all services and just start and stop record_event and mp3record but I am rather looking hot to mimic the configuration from cloud locally? Any idea ?

jav413, 2017-04-17:
I want to know how to access and make changes to it without going through the hassle of removing the SD card.

querex007, 2017-04-17:
Same problem over here.
When you login with user root by telnet, you don't have permissions to modify the WiFi config file (ssid / password)

karepiu, 2017-04-17:
@querex007 - I just modified wpa_supplicant.conf without any issue as root. Not sure what you are trying to do but as a root you have all rights. You may not be able to modify file if it is locked by some running process but you can change anything else. Be carefull though with Wifi config changes. My suggestion would be to prepare a config outside of the dir, copy original config as backup somewhere and right a script that will copy your config from memory card on boot. This gives you option to go back to old config on reboot when something will go wrong. 

@jav413 - what do you mean ? You can insert memory card with a script that will permanently add telnet option to cam ( it modifies content of cam memory ) and you can do whatever you want without memory card then. This is how I am manipulating the cam. 

This cam is normal just a linux with busybox and arm processor. You can do whatever you want there. The only missing part that I have is how rmm service is running cause any modification around this forces cam to reboot. Otherwise you can do everything.

jav413, 2017-04-17:
I thought telnet was built in already without a script? I used putty and it seemed like I was in. I am starting to get familiar with the camera. Can you point me in the right direction with telnet? Also, have you added the script to delete videos within a certain amount of time?

karepiu, 2017-04-17:
Telnet is built it but there is not init.d script to start this functionality, at least on most of the firmware versions. If you have the hack installed the telnet is being started there.

jav413, 2017-04-18:
I want to learn how to change some settings via telnet but I would appreciate a jump start in the right direction. Which command do I use to see the files which can be edited? I can figure the rest out over time. Thanks again

karepiu, 2017-04-18:
This is more question on how to use linux. There is ton of materials about this online. Not the best place to discuss this. 

Just to give you jump start:

ls - like dir in Windows shows content of dir
cd - change dir 
vi - text editor 
chmod - changes permission to file ( use carefully - modifying the rights to sys file may cause that you may never be able to start cam again )

closed 2017-04-18 by karepiu