fritz-smh/yi-hack #149: Modify "Activity Zone"

anndrewww, 2017-03-22:
I have been taking a look in /etc/ui.conf to try and modify the "Activity Zone" of the motion detection. Is this possible with "Yi-Hack"? I have identified some possible associated lines.

```
bitmapX=20
bitmapY=20
``` 
(I think this is the location of the "Yi" watermark on the screen - but a good possible reference point.

```
osd=1
osd_pos=1
osd_xpos=56
osd_ypos=30
osd_time=1
osd_flow=0
osd_chnno=1
osd_frameRate=0
osd_info=Õ®µ¿1
```

utya1988, 2017-03-23:
I modified the email push string but not result.

airdrummingfool, 2017-03-28:
I was able to modify the `motion_[left/right/top/bottom/sensitivity]` values using `vi` over telnet. Make sure that you `sync` and then `reboot` before connecting/viewing properties with the mobile app or they seem to get overwritten.
The resolution appears to be `1280x720`, so your max value for `motion_right` is `1279` and `motion_bottom` is `719`.

anndrewww, 2017-03-30:
I was able to modify it as well, but it doesn't look like it has any effect.

ujnatg, 2017-04-07:
could you please share more details about versions of firmware ? Fox example, I've no motion_* options predefined in ui.conf file

notdone, 2017-04-12:
please share how to do that

airdrummingfool, 2017-04-12:
@anndrewww As long as I sync and reboot after modifying the values (before connecting with the app), the app shows that the activity region has been moved. I'm able to trigger motion alerts in the newly-defined area, and alerts do not seem to trigger with motion in the old area.

@ujnatg I'm running `1.8.5.1N_201601071352`, which was the version I had to update to in order to be able to pair with the Yi Home app. I can confirm that they were not present when I was running `1.8.5.1M_201512011815`, the default version included in this hack (see #152, my PR for updating to the app-required firmware).

@notdone 
1. `telnet` into the camera
2. `vi /etc/ui.conf`, modify the values as you see fit (to skip to the motion section, you can search "motion_" by typing `/motion_` and hitting enter)
3. save and quit vi
4. `sync` to persist changes to flash memory
5. `reboot` to restart the Yi and apply the changes

anndrewww, 2017-04-12:
@airdrummingfool 

Thanks, I'll try to test that out and see if I can get it working.

notdone, 2017-04-14:
@airdrummingfool
oh my cam running 1.8.6.1B so dont have "motion_mod..." to modify the values.

karepiu, 2017-04-18:
Have you been able to control cam being on/off using similar approach ?

anndrewww, 2017-04-19:
@airdrummingfool Still no luck. What values are you using for motion_mode; motion_resolution; and motion_sensitivity?

closed 2017-04-19 by anndrewww