fritz-smh/yi-hack #157: YI 1080p Home Camera

martyandhope, 2017-04-05:
Any chance there will be an update to include this new camera?

Thanks

shadow-1, 2017-04-07:
Which version of the Yi 1080p Home camera do you have? In addition what features are you looking for?

There is another repository working on the Yi 1080p Home [here](https://github.com/xmflsct/yi-hack-1080p) along with my own custom firmware [here](https://github.com/shadow-1/yi-hack-v3).

xmflsct, 2017-04-07:
@shadow-1 Just a thought, how shall we proceed further? I can close my project and/or refer people to your project for the releases. Or I can take the foundation from your project, and publish any further changes specifically for v1 1080p in my current project. What would you prefer?

martyandhope, 2017-04-07:
[image]

This is all the info on my camera.  I tried to email it but it didn't work.

xmflsct, 2017-04-07:
@martyandhope It is a US version. And why do you still want to hack it?

martyandhope, 2017-04-07:
To be able to control which cloud it writes to .   I would prefer it write to mine.
Thanks

xmflsct, 2017-04-07:
@martyandhope Write to own cloud, namely RTSP stream is still under development.

martyandhope, 2017-04-07:
Thank you so much.  Also looking at GitHub for information.  Do I need telnet access to the camera to install you hack?   Do I need a cable attached to the camera?

xmflsct, 2017-04-07:
All hacking methods for this type of camera now just require a micro SD card, and that's it.

martyandhope, 2017-04-08:
I don't understand what this is saying.
Download a release firmware binaries from Github. Two firmware files are required. One firmware file is for the rootfs partition on the camera whilst the second firmware file is for the home partition on the camera.

Where do I find the binaries.   I find the src files, etc.  But no binaries.
From other release I am to copy the sd folder to the root of the SD card.   Does this work the same?
Thanks again.

| rootfs_y20 | home_y20 |

martyandhope, 2017-04-08:
Sorry for being so dense ... I found everything and all is working great.
Thanks go much.
As I tinker with the camera I will try to submit useful updates.

Marty

xmflsct, 2017-04-08:
@martyandhope Wonderful!

shadow-1, 2017-04-08:
@martyandhope 
Great news! Looks like I need to update the documentation to make a few things clearer. I have no idea why the Releases link on Github doesn't show up when using the mobile web page!

@xmflsct 
I'm not fussed how we go about the firmware. I believe the method of running a script from microSD card is no longer the best approach anymore. Every firmware I have looked at besides the original Yi Home 720p has disabled running scripts from microSD card with an update.

I am quite happy to merge our projects together and create one firmware for all Hi3518ev200 devices.

martyandhope, 2017-04-09:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; } I will send you some suggestions on the documentation later on today 
Thanks for all you help
Marty

xmflsct, 2017-04-10:
720p RTSP streaming is possible now. [xmflsct/yi-hack-1080p](https://github.com/xmflsct/yi-hack-1080p)

junsheng89, 2017-04-11:
Hi all, i have just looked through most of the threads.  I cannot seem to be able to make my yi home 1080 v1 work with the mi home app, it just keeps complaining, "this camera cannot be used outside china". 

my QR code starts with 18CNYXXXX

Things i tried so far:
xmflsct/yi-hack-1080p
shadow-1/yi-hack-v3
https://www.youtube.com/watch?v=SsAMkIqUZLQ

I am hoping that experts here can lend me a helping hand. Thanks!

xmflsct, 2017-04-11:
@junsheng89 You probably did not read `readme` carefully of those projects. Use with Mi app is fully supported and tested with [shadow-1/yi-hack-v3](https://github.com/shadow-1/yi-hack-v3). For any further issues, please direct to that project.

junsheng89, 2017-04-11:
@xmflsct, hi zhiyuan, i was able to successfully flash rootfs partition & home partition, however, i am stuck at the configuration part. what changes are to be made on the ProxyChainNg configuration? can you provide some guidance?

xmflsct, 2017-04-11:
@junsheng89 [https://github.com/shadow-1/yi-hack-v3/issues/1#issuecomment-293198769](https://github.com/shadow-1/yi-hack-v3/issues/1#issuecomment-293198769)

junhau89, 2017-06-09:
If guys, I seem like flash with wrong firmware. Now the Yi camera once power on and stay and turn on with yellow light. Even press reset button also no help. I already try re-flash back with firmware 1080p version firmware but can't. I guess I wrong flash with 720p firmware in 1080p camera.

alextimario, 2017-07-12:
Hi, I have China version Yi Home 1080p with 48CNYxxxx version. Can anyone help me to have a custom rom, because I can't use it outside China. Thanks in advance!

shadow-1, 2017-07-19:
@junhau89 @alextimario 
Feel free to try my firmware which supports the official app via Proxy server:
https://github.com/shadow-1/yi-hack-v3

Make sure you read the step-by-step guide carefully.

alextimario, 2017-07-26:
@shadow-1 
Thank you very much! My camera is working now! Good job! I really appreciate your help. Thanks again!

junhau89, 2017-07-28:
@shadow-1 , Thanks and it work for my camera. But the video record can't save on the memory card and play back? Any idea on this?

moeboid, 2017-12-20:
Hi there, 

i chanced upon this thread and don't mind me jumping in. 
I have a YI 1080P home camera with serial number that starts with 48CNYxxxxx. 
I have tried the above yi hack v3 method but however, when my camera was successfully linked to my wifi, it immediately said that this camera can only be used in china. 
I know that the next step is really to open the browser and connect to the ip address of my camera which i found via network scanner. 
The issue here is i am not able to open up the ip address config page and by this time, my camera has already turn off (meaning no more lights). 

Is there any steps i am missing or is the device is not reading the firmware correctly? 
Please help! 
Thanks in advance!

moeboid, 2017-12-20:
i figured that i didn't manage to flash it successfully. attaching a photo of  my camera details. It cn v2.2, not sure if this matters. any advice is greatly appreciated. 
thanks! 
[image]

moeboid, 2017-12-22:
I have solved the issue of flashing it but now the other problem is it can't open the freevpn.ninja website. 
Not sure if it's block in my country, Singapore. 
Help anyone?