fritz-smh/yi-hack #159: OpenHAB interface to Yi-Hack

rbpaul, 2017-04-09:
Greetings all.

I am not much of a hacker, but I really like what is going on here.  I am using death2all110’s version of the hack because it seem to be most up to date.  I am leaving my comments/questions here because this is where I see the most activity.  I think the Yi Camera is a big improvement to the video capabilities of the Raspi Cameras that I have been using.  

I use openHAB home automation software on a raspberry pi.  I have been using Raspi Cameras and USB Cameras to stream video which I access on the web interface of openHAB.  I also use PIRs on the Raspberry Pis to capture motion events to trigger photographs which the Raspberry Pi then emails, and I can access these photo through the web interface of openHAB.  I can also manually trigger these event from the web interface of openHAB.

I have modified airdrummingfool’s check motion script to get the Yi Camera to signal openHAB when motion is detected. OpenHAB then SMS’s me of the event.  There is an acceptable few second delay form the Yi Camera motion alert vs my Raspberry Pi SMS.  (I still have the Yi Camera app up, but expect to shut It down soon). 

My Raspberry Pi is also triggered to run a script that uses the Yi Camera’s ftp_upload script to save the video file to my NAS.  From there the Raspberry Pi pulls and emails a still image from the saved video and also loads it to openHAB for viewing.  The Raspberry Pi also cuts a 3 second video clip from the longer Yi Camera Video and emails it.  This lag on receiving these emails is about 2 minutes after the Yi Camera detects motion. I can work with this.  

What I cannot do, but would like:

1.  I would like to somehow trigger the Yi Camera to begin recording video upon a openHAB command (i.e., a Raspberry Pi motion event or I just want to save that particular video moment).  If someone out there can see how to do this, please share. 

2. I would like to have the RTSP stream available through openHAB.  The web interface on openHAB doesn’t seem to be able to display a RTSP stream.  I am using VLC for the time being.  The MJPEG streams from the Raspberry Pi camera and USB Cameras work pretty good.

3. I would like to have access to the 3 second video clips the Raspberry Pi makes on the web interface of openHAB.  I can view them find from the email or on OSX with quicktime, but I don’t know how to get my openHAB web interface working with these files.

4. I would also like to know if anyone is having success with the audio interface with this hack.  I can see a doorbell capability with that feature.

Any suggestions or directions would be appreciated.  

Thanks to death2all110 and airdrummingfool.  Keep up the good work.

Old man in Alabama.

airdrummingfool, 2017-04-12:
Hello @rbpaul, I'm glad my code has been useful to you. Regarding 1, the `/home/motion_detect_record` executable may be useful for what you're trying to accomplish. The usage is `/home/motion_detect_record [/path/to/output.mp4] [length to record in seconds]`. Note that it seems to save a few (3-4) seconds of video from before the command was issued, which in my opinion is a plus (but you might want to extend the record time a few seconds to compensate).

Of course you currently can't trigger it over the network, but perhaps you can modify someone else's existing network command code to add it in. I'm also considering taking a swing at web control at some point, but it may not be anytime soon.

The audio interface works for me via the official app, but I haven't seen anything in the hack that accesses the audio.

rbpaul, 2017-04-12:
Thank you very much airdrummingfool.

I have been able to create a bash script on my raspberry pi which automatically telnet's into the camera; executes the /home/motion_detect_record executable; records a 5 second mp4; then the bash script uses ftp to download the mp4, and I have it on my NAS 9 seconds after triggering the script.

I can also access this script through the web interface of openHAB.  I can automatically trigger this script with raspberry pi PIR events.  With that, I don't have a problem with triggering it over the network/web.  

I do have a lingering question about what will happen if the camera tries to execute a record event while I am also using this script to record.  I anticipate some sort of error.  Will report it as I am able to identify it.

Thanks again.
rbpaul

modidep, 2017-04-26:
Hi rbpaul,

I am using OpenHAB as well. Can you share how you integrated Xiaomi Yi with OpenHAB? Thanks in advance,

Regards,

modidep

rbpaul, 2017-04-26:
modidep,

Not sure what you are asking that I didn't address above.  Can you be more specific? 

I would not say I have integrated the Yi camera with OpenHAB so much as I can use BASH scripts to command both the Camera and OpenHAB to a point where the two become useful for my purposes.  
The biggest shortcoming is not being able to view the Yi Camera's MP4 steams in the openHAB interface the way I can view my other MJPEG streams.

Is there something specific you are trying to do?  I might be able to help with that questions.

modidep, 2017-04-26:
Thanks for the prompt reply rbpaul,

I want to use motion detection if the hacked yi as a trigger in openhab to switch lights on. Can you let me know how did you integrate yi camera and openhab. Thanks,

Modidep

rbpaul, 2017-04-26:
Greetings Modidep,

airdrummingfool’s improved check_motion script has included support for POSTing a JSON message to notify a local/internet server (e.g. openHAB) about motion status. This will allow the Yi to be used as a motion sensor.  

I am not good with JSON messaging, so I modified his script to POST a text/plain type message which updates my openHAB item when it is triggered. 

Rather than using the JSON POST line airdrummingfool’s check_motion script, which is:

`/home/curl -H "Content-Type: application/json" --data "$notification" "$1" --silent --show-error --stderr - && echo "$1 notified"`

I am using this line:

`/home/curl -H "Content-Type: text/plain" --data "1" http://myopenHAB_IP:port/rest/items/my_openHAB_item`

After this trigger, the openHAB rules attached to this item executes on the item being updated.

Hope this helps.

modidep, 2017-04-26:
Thanks a lot rbpaul,

It is great help. I just can't seem to find
airdrummingfool's check_motion.sh script. I think I will be able to figure
out more if I can find the script.

Regards,

Modidep

modidep, 2017-04-26:
Found it :)
https://github.com/airdrummingfool/yi-hack/tree/ceffb8b8c3a733eb2b853a340ccdd0208d13473d

modidep, 2017-04-29:
Hi all,

After fiddling with it for sometime, I was able to trigger a recording input Surveillance Station on my QNAP and a switch in my OpenHAB whenever there is a motion seen by Xiaomi Yi Home camera with death2all110's fork of fritz's Xiaomi Yi Home hack,  airdrummingfool's check_motion.sh script and rbpaul's modifications and guidance. How I did it is: 

I created a custom HTTP event called "motion" in my QNAP Surveillance Station:

Go to "Event Management" in Surveillance Station, toggle "Advanced", click the small plus near  "User-defined Event", type motion in the "Event name" and click "Apply" then while motion is selected click "Add", select Action Type "Recording", check Ch-01 and then click "Apply"

and creating a switch called "Patio_Motion" in my OpenHAB:

add the below to my OpenHAB items file:
Switch Patio_Motion <present> {autoupdate="false"}

My main challenge was the shell script since my QNAP recording and OpenHAB switch on kept being triggered all the time. Finally, I got it working.  (For easier follow up my NAS ip is 192.168.1.30 my OpenHAB ip is 192.168.1.40).

Replace the below line in  airdrummingfool's check_motion.sh script:

/home/curl -H "Content-Type: application/json" --data "$notification" "$1" --silent --show-error --stderr - && echo "$1 notified"

With the below lines:

if ($has_motion); then
			/home/curl http://192.168.1.30:8080/cgi-bin/logical_input.cgi?name=motion --silent --show-error
			/home/curl -H "Content-Type: text/plain" --request POST --data "ON" "http://192.168.1.40:8080/rest/items/Patio_Motion" --silent --show-error

		else
			echo $"GRS-NAS & OpenHAB trigger not executed."
		fi	

Now everything works thanks to everybody: fritz, death2all110, airdrummingfool, rbpaul :) I thought someone else might use it.

Cadsters, 2017-04-30:
Thanks modidep for the Qnap <-> openHAB explanation, and all who contributed to the y-hack!