fritz-smh/yi-hack #161: Cam with multiple IP addresses

jav413, 2017-04-14:
I have an interesting issue. The hack works fine on one camera and tinycam pro picks it up easily. Now that camera has 2 IP addresses. I can access it from the browser using either IP address or tiny cam using either IP address. I want to set up another cam. I installed the hack the same way as the other. Tinycam picks it up but no pic or anything. I can even access the 2nd cam from the browser and get the yi hack page but nothing recorded. The interesting thing is if I have both cams plugged in then only one has 2 IP addresses. But if I unplug either camera and leave only one on, then that camera has 2 IP addresses. The address that keeps repeating ends in xxx.xxx.xxx.128. Any help would be appreciated. Thanks

airdrummingfool, 2017-04-23:
See my PR #108 for a fix - apparently `eth0` is always configured to use a .128 address, so I disable it during bootup.

querex007, 2017-04-23:
Many thanks, had the same annoying issue over here...
Now it's fixed, works great!

jav413, 2017-04-23:
Many thanks. I saw your post but it does not say how to disable etho?

scadaguru, 2017-07-05:
Thanks, it worked!