fritz-smh/yi-hack #166: US version firmware 1.8.7 cant hack for RTSP

dwr555, 2017-04-21:
i got the cam with firmware 1.8.7 tried all solutions found to change the firmware to RTSP support but couldnt make it work...
when i turn the cam on with the SD + the firmware files on root... it keeps go to "Waiting for conection" i have reset several time but with no luck to update the firmware
anyone had a success to make yi home works on with RTSP when it has the 1.8.7 firmware?
(i couldnt even downgreade the firmware)

fredanaman, 2017-04-25:
Hi
i have exactly the same issue. Hope it gets resolved.

brumsterino, 2017-05-09:
Same here. no updates?

keith721, 2017-05-30:
The 1.8.7 firmware doesn't appear to activate the script /test/equip_test.sh on the SD card.  Additionally, the U-Boot  distributed with this firmware doesn't allow interrupting the boot process.   If you change the filename on the SD card from 'home' to 'home_y18' it WILL downgrade the firmware, but still doesn't execute the equip_test.sh script to perform the hacks. :(

AdySan, 2017-05-30:
@keith721 I just bought a 4 pack of these. Don't know what firmware version is on them yet. Hope you can figure out some way to downgrade and make RTSP work.

dwr555, 2017-05-30:
I am trying to add it to synology Survilliance  and seems that only rtsp will work...
I hope this thread which opened awhile ago will come up with solution as the yi cam seems like a good product for fair cost.

keith721, 2017-05-30:
In order to hack these, we'll need someone with telnet/ssh access to their camera, and a fairly recent 1.8.5 or 1.8.6 version of the firmware with the hacks working.  Then, they'd need to use the 'dd' utility to dump all the MTD partitions to the SD card.  From there we'd update the other non-'home' partitions to restore the equip_test.sh script execution.  Then use the U-Boot 'mkimage' command to package the partition into a U-boot wrapper, which the firmware loader expects.

ekos2001, 2017-05-30:
I was able to load 1.8.6.1B_rtspfix from http://xiaoyi.querex.be/firmwares

keith721, 2017-05-30:
> I was able to load 1.8.6.1B_rtspfix from http://xiaoyi.querex.be/firmwares

How can you download anything from that site without getting the ad.fly advertising infection?

AdySan, 2017-05-30:
nothing seemed to persist on my computer, unless I'm missing something.. after the few seconds of wait, the button on top right starts the download.

dwr555, 2017-05-30:
You managed to use this firmware with the normal load? (Reset,sd,boot)
Are you using the US or international based model?

-

ekos2001, 2017-05-30:
> How can you download anything from that site without getting the ad.fly advertising infection?

After some time "Please wait" in the top right corner will change to "Skip Ad" and you should be able to download it.

> You managed to use this firmware with the normal load? (Reset,sd,boot)

I followed their instruction page: http://xiaoyi.querex.be/instructions/

> Are you using the US or international based model?

US version (sn: 22US…)

haideramn, 2017-06-08:
keith721 I have access to a camera that took the hack. I bought a 4 pack from Amazon and one of the 4 cameras took the hack....the others dont. 

Anything I can help with providing from my working camera?

AdySan, 2017-06-09:
@haideramn thats interesting, and can you confirm the firmware versions that were on these cameras before you applied the hack?

haideramn, 2017-06-09:
1.8.7.0A_201702081101

haideramn, 2017-06-09:
CORRECTION to my original post. The Hack did NOT Work with any of the 4 PK
cameras whereas it worked with my single white pack. I have ordered 1 more
SINGLE pack of the YI camera White to see if I can make it work again as I
am very confident that the black US version and the 4PK US versions will
simply not take the hack....whereas the single white camera seems
to.....not sure.

Will update.

On Fri, Jun 9, 2017 at 11:19 AM, Mohammed Haiderali <haiderali1211@gmail.com

> 1.8.7.0A_201702081101
>
>
>> @haideramn <https://github.com/haideramn> thats interesting, and can you
>> confirm the firmware versions that were on these cameras before you applied
>> the hack?
>>
>>
>
>
>
> *-Mohammed N. Haiderali*

haideramn, 2017-06-09:
Confirming the hack works on single White YI Camera from Amazon. It does
not work on 4PK or Black Camera -- no clue what the difference is, but the
white ones immediately take the hack.

On Fri, Jun 9, 2017 at 11:20 AM, Mohammed Haiderali <haiderali1211@gmail.com

> CORRECTION to my original post. The Hack did NOT Work with any of the 4 PK
> cameras whereas it worked with my single white pack. I have ordered 1 more
> SINGLE pack of the YI camera White to see if I can make it work again as I
> am very confident that the black US version and the 4PK US versions will
> simply not take the hack....whereas the single white camera seems
> to.....not sure.
>
> Will update.
>
> On Fri, Jun 9, 2017 at 11:19 AM, Mohammed Haiderali <
>
>> 1.8.7.0A_201702081101
>>
>>
>>> @haideramn <https://github.com/haideramn> thats interesting, and can
>>> you confirm the firmware versions that were on these cameras before you
>>> applied the hack?
>>>
>>>
>>
>>
>>
>> *-Mohammed N. Haiderali*
>>
>>
>>
>>
>
>
> *-Mohammed N. Haiderali*

daweeze02, 2017-06-11:
I have 2 white ones from Amazon with 1.8.7.0A and have tried the hacked but it does not work.  I've tried with the original WPA settings and adjusting them to match my WPA2 AES encryption but I just end up with a flashing orange light after the solid blue for a few seconds

daweeze02, 2017-06-11:
UGH after an hour I found out my issues.  My SSID is in all caps and but I didn't type it that way in the config FML

haideramn, 2017-06-12:
Great! Just did the hack on the 3rd white camera also. Works great!

AdySan, 2017-06-12:
@haideramn just to confirm, it's still not possible to downgrade from 1.8.7.0A, or were you able to?

haideramn, 2017-06-12:
Correct not able to downgrade, but able to run hack on the white cameras
that were sold in single-pack running 1.8.7.0A.xxxxxxxxxx

AdySan, 2017-06-12:
@haideramn thanks, that is good to know. Will try. since there's a bunch of variations now, can you please describe the exact process you followed, did it involve renaming any files, modifying anything?

haideramn, 2017-06-12:
No change to any filename, except your SSID/network config in TEST>WPA SUPPLICATION.CONF & Yi-Hack settings such as no cloud, no led, etc in TEST>YI-HACK.cfg.

I used these specific files each time and it worked like a charm. 

[Cam Files.zip](https://github.com/fritz-smh/yi-hack/files/1068840/Cam.Files.zip)

AdySan, 2017-06-13:
@haideramn unfortunately these files also make the camera go directly to "Welcome to Yi Home Camera....Waiting to connect"

haideramn, 2017-06-13:
Yes that is what happens for me also, it goes to that and runs through the
prompt then connects up fine without any app and in addition, I block all
outbound IP communication for the camera IP address, but it joins my wifi
and works on LAN just fine.

I am about to set the hack on my 3rd camera today.

AdySan, 2017-06-13:
@haideramn oh that's interesting. Can you tell me roughly how long should I wait or how many time does it say Waiting to Connect before it finishes applying the hack? Any specific LED activity to look for?

haideramn, 2017-06-13:
It's a few seconds but are you powering camera on with sd card and hitting
reset button untill you here reset successful?

Only after that does it take the hack

AdySan, 2017-06-13:
@haideramn perhaps my WPA2 settings are wrong. @daweeze02 can you share yours please?

ygator, 2017-06-18:
`if [ -f "/home/hd1/test/equip_test.sh" ]; then
	/home/hd1/test/equip_test.sh
	exit
fi
`
The above is in init.sh on all versions of firmware I have including home.1.8.7.0A_201702081101us

dwr555, 2017-06-18:
i have tried the files from @haideramn  powerede on the camera and hold the reset button but it keeps saying waiting to "connect".
mine is the single cam US version...
any idea how do force the firmware update?

ccantill, 2017-06-24:
I just got my camera today and it had the same firmware. I tried renaming the 'home' file to 'home_y18' but it looks like that bricked my camera. Now if I turn it on without an sd card the led is stuck to orange. I've tried other firmwares on the sd card (as home_y18) and that just turns the orange led blue occasionally, but doesn't seem to be doing much else. Any ideas anyone?

imrich82, 2017-06-26:
now i;m having the same issue like @ccantill, i rename the version 1.8.5.1K_201508311131 home into home_y18 and insert the sdcard, how i'm stuck to orange too. please help me fix this problem?

haideramn, 2017-06-26:
It's bricked

dexy86, 2017-06-26:
anything on US version ?

jun3280net, 2017-07-29:
Hello there. Hopefully, this can useful to some of you.  While I can not comment specifically on THIS RTSP fix, I was successful in loading a different one (probably an older version) onto 5 x Yi Home 720P Cameras (US model).  I used the hack made for the Chinese model noted here : [http://en.miui.com/thread-196713-1-1.html](http://en.miui.com/thread-196713-1-1.html).

Supplemental details can be found here : [http://xiaoyi.querex.be/instructions/](http://xiaoyi.querex.be/instructions/)

The instructions and process are pretty straightforward, but this is the process I used.

**Downgrade firmware :** 
1.  Use your Yi Home camera to format an SD Card (Yi Camera doesn't always agree with the Windows formatting).
2.  Once formatting is complete, unplug the power to the camera (do not reset the camera).
3.  Remove the SD Card from the Yi Home Camera and insert in your computer.  
4.  Delete (not format), the files and folders from the card.
5.  Copy the firmware onto the SD Card and rename the file to "home" (all lowercase).
6.  Plug the SD Card back into the camera, and plug in the power.
7.  The camera's yellow light should start blinking for a few minutes (as it is loading the firmware).  When it completes, it blue light should turn on, and you should be able to connect to the mobile app and confirm that the firmware updated.
8.  If u are unsuccessful loading 1.8.5, try loading 1.8.6 onto the camera first, and then load 1.8.5 afterwards.  I needed to do this for one of my cameras. 

**Load RTSP Hack**
1.  Complete Steps 1 - 4 from Downgrade firmware section.
2.  Copy the folder "test" from the test-rtspfix-07.zip file onto the SD Card.  I did not modify any of the code for the timezone in the .SH file as suggested.  Mine worked fine and I'm in UTC - 7.
3.  Plug the SD Card back into the camera, and plug in the power.
4.  The camera's yellow light should start blinking for a few minutes (as it is loading the firmware).  When it completes, it blue light should turn on.
5.  From your computer Telnet the IP of the camera.  If you are successful, it should ask you for your login (root) and password (1234qwer).
6.  Once logged in, change the root password from Telnet typing "passwd" 
7.  You should be able to see your RTSP feed
Main stream: rtsp://ipaddress:554/ch0_0.h264
Minor stream: rtsp://ipaddress:554/ch0_1.h264
Audio stream: rtsp://ipaddress:554/ch0_3.h264
8.  If you are loading multiple cameras - repeat steps 1-7, as reusing the SD Card sometimes doesn't work.

With RTSP enabled, I have all 5 Yi Home Cameras running through Blue Iris :
[image]

I've included all files I used for your convenience.  

Regards,

J

[Yi Home Firmware Older Versions.zip](https://github.com/fritz-smh/yi-hack/files/1185266/Yi.Home.Firmware.Older.Versions.zip)
[test-rtspfix-07.zip](https://github.com/fritz-smh/yi-hack/files/1185267/test-rtspfix-07.zip)

mdisieno, 2017-08-09:
@jun3280net I tried both 1.8.5 and .6 and sadly my Yi doesnt blink amber, just blinks blue for a second or two. My version # is 1.8.7.0A_201702081101

jun3280net, 2017-08-09:
Just to confirm, did you format the card in the Yi can first. Power off the
unit.  Then remove SD card and put in computer. Delete contents (do not
format). Copy the firmware to the card and rename to "home".   Then put the
card bad into the camera and power on?

mdisieno, 2017-08-09:
@jun3280net Yes I did. The only peculiarity might be I did this all on a linux laptop, but that shouldnt really matter.

jun3280net, 2017-08-09:
Hmm.  Have you had any success changing any of your firmwares?

I had one camera which was a bit troublesome, and I had to downgrade to a
newer firmware before I was able to downgrade to an older firmware.

The last firmware I downgraded from was 1.8.7.0A_201702081101.

wrongbongo, 2017-08-10:
I believe the core issue of your procedure to downgrade is that the camera's newer firmware no longer looks for "home" to boot. There is all a line to which you could feasibly download to given the change in hardware.

Hokiecow, 2017-12-02:
Has anyone been able to get the black Yi Home 22US working with Blue Iris?

haideramn, 2017-12-02:
Not yet, all my successes have been with the Yi White only and those that
are sold in single kits not the three pack white ones.

mdisieno, 2017-12-18:
Forewarning, the newest revision software updates itself after a while. I was in A revision and now I'm in D :(

SteveAnthonyFM, 2026-11-10:
Confirming this thread's method still works in 2026 on a **22US Yi Home 720p (YHS-113-IR) starting from firmware 1.8.7.0F_201809191400**, which is newer than anything reported here.

What worked:
1. Downgrade with `1.8.5.1N_201601071352home_us` (from the *Older Versions* zip attached above), renamed to `home` on a blank FAT32 card. It took several minutes of blinking amber, then solid amber. Confirmed via `version=1.8.5.1N…` in the `log_*_yi_login.tar.gz` the camera writes to the card, since the app couldn't show it.
2. The `test` folder from `test-rtspfix-07.zip`, **without `test/server`**. Windows Defender flags that binary as `Trojan:Linux/SAgnt!MTB`, and RTSP doesn't need it. `rtsp://<ip>:554/ch0_0.h264` (720p + AAC) and `ch0_1.h264` (360p) came up about 8 minutes later.

I also wrote two follow-up hook scripts: one turns telnet off and makes FTP read-only on the recordings folder, and one blocks internet on the camera (so no auto-updates), with a status file readable over FTP. The guide, scripts and checksums of the files from this thread are here: https://github.com/SteveAnthonyFM/yi-home-720p-22us-rtsp

One gotcha worth knowing: while `test/equip_test.sh` exists, the camera runs it *instead of* its normal startup, so any hook script must end with `reboot` or the camera sits on amber with no Wi-Fi.

Huge thanks to everyone here, especially whoever posted the working US-model steps and `test-rtspfix-07`. None of this would have happened without this thread.