fritz-smh/yi-hack #168: Camera goes off line AND Filesystem set to read-only?

rbpaul, 2017-04-27:
Sometimes my camera goes off line - no RTSP stream, no telnet, no ftp connections available.

Sometime it will come back online without intervention. When it does, I can see it has been able to record video while offline.  I think this may be a wifi issue - not sure.  Sometimes I will have to reset the camera by disconnecting the power and re-powering it.

Sometime the camera seems online, but the “Filesystem has been set read only”  I can see a RTSP Stream, but the check_motion script will not trigger.  I can telnet into the camera and “dmesg” tells me:

`FAT-fs (mmcblk0p1): error, fat_free_clusters: deleting FAT entry beyond EOF
FAT-fs (mmcblk0p1): Filesystem has been set read-only`

I thought this indicated a bad SD card, but then found out I can reboot from telnet and it will sometimes return to normal operations, but eventually it happens again.

Any thoughts of how to correct these type issues would be appreciated.  

Thanks

rbpaul, 2017-04-27:
``

modidep, 2017-05-03:
Hi rbpaul,

I had the same problem when I disconnected the camera. What i think is if the camera is disconnected while recording on the sd card the file system is left in intermediate state. If you connect your sd card to a computer you'll see it asks if you want to fix the sd card like you disconnect a usb drive while writing to a file.  If you don't fix it you can not even delete the record folder. So file system gets corrupted. As for solutions I don't have any :)