fritz-smh/yi-hack #171: Flashing / WiFi connection problem

dkoguciuk, 2017-05-01:
Hello,

A few days ago I've received my Xiaomi Home Camera (720p, model: YHS-113-IR), which I've ordered on aliexpress, since then I'm trying to flash my camera, but no luck so far. Steps I made:

- reset the camera with "RESET" button
- format the microSD card (GoodRam, 16GB) to fat32 using gparted on ubuntu 16.04
- update the wpa_supplicant.conf with ssid and psk fields:

> ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="ERR_INVALID_SSID"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="snow_nows_nothing"
}

-update yi-hack.cfg with IP configuration: 

> IP=192.168.0.242
NETMASK=255.255.255.0
GATEWAY=192.168.0.1
NAMESERVER=192.168.0.1

- copy the **home** file and **test** directory to the SD card
- insert sdcard in the camera and power it on 
- heard the "Welcom to Yi home camera" and "Waiting to connect" messages, led is blinking yellow
- powering off the camera and plugging the sd to the pc shows no other files (no test/log.txt file or anything else)

Could you please help me somehow? How do I know flashing was successful? How do I know the network configuration is fine?

Thanks!
Daniel

corollas, 2017-05-04:
Having same issues with connection. I was under the assumption camera has to be downgraded to an earlier firmware (which I did), connect using Yi Home app (which it did successfully) and then apply this hack. But its not working

plexdev768, 2017-05-13:
i have the same problem.  I have version 1.8.7 installed.  Did anyone ever figure this out?

0sm3l, 2017-05-17:
I have two of them working perfectly fine and one that can't get downgraded. Not sure what's going on. The two that work came with the white base, not sure if that makes a difference.

dkoguciuk, 2017-05-20:
@0sm3l have you downgrawded camera firmware before installing the hack?

0sm3l, 2017-05-20:
@dkoguciuk I've tried it with some old firmwares but it seems to ignore the card completely.

corollas, 2017-05-20:
mines working good now. Downgraded to custom firmware which has similar features to this hack (cloud disabled, RTSP, motion detection recording, loop recording etc.)

juan22a91, 2017-06-07:
@corollas can you please post the link for this hack please?

lstarrett, 2017-06-08:
I cannot get my camera to recognize the SD card and boot with the custom firmware and scripts. I reset it without the SD card installed, then inserted the prepared SD card and powered on the camera, and it still boots with "Waiting to connect" repeated forever until I set it up with the Yi app. It seems to totally ignore the SD card. I really want the RTSP functionality, but I'm at a loss. Does anyone have advice? Thanks much.

ontdamian, 2017-06-08:
The answer is in other threads. You probably have 1.8.7.0A firmware which do not recognize 'home' and other files. Just follow [this](https://github.com/fritz-smh/yi-hack/issues/174) thread.

lstarrett, 2017-06-08:
@ontdamian confirmed that I do indeed have the 1.8.7A firmware on this camera. I read through that other thread and it appears that so far no one has gotten a downgrade path to work to get cameras with this firmware to load custom firmware and the hack files. Is this still the case as far as you know?

ontdamian, 2017-06-08:
@lstarrett, By "answer" I mean hint, that files without "_y18" are not read at all. Camera do not ignore SD card indeed, but files are wrong. We should follow main thread and waiting for fix.

lstarrett, 2017-06-08:
@ontdamian Understood, thanks for the clarification. I'll keep following that thread and hopefully someone more clever than me can crack it soon!