fritz-smh/yi-hack #172: Anyway to improve wifi in the yi camera.

bronzerain, 2017-05-04:
The original YI small ants 720p camera seems to have weak WiFi capabilities. The Xiaofang camera has a separate wifi antenna  and seems to have a good signal. What is the best firmware version to use for WiFi? Is there something I can modify in the script to improve this?

bronzerain, 2017-05-05:
Is there a way to add a external antenna like the Xiaofang 1080p  camera and/or add something to the case to pick up WiFi better?

keith721, 2017-05-30:
The Yi camera DOES have a U.FL snap-on connection for the internal antenna. This would allow a bit of creative engineering to install an RP-SMA connector on the stand for higher-gain (3, 5, 9 dBi) antennas.

RiRomain, 2017-08-18:
Could do something analog to this:
https://github.com/samtap/fang-hacks/wiki/External-Wi-Fi-Antenna

Maybe fixing the connector to the camera support.