fritz-smh/yi-hack #173: Can not boot into hack (one of my camera)

cnberg, 2017-05-09:
Hello, firtz-smh, THANKS for your excellent work!

I have four 2nd hand cameras, two of them can boot with your hack (using the same configuration) working perfectly, and one of them uses https://github.com/hermitcrabslab/yi-hack .

But the last camera, I cannot boot it into any hack (by you or Hermitcrabslab), I have modified yi-hack.cfg & wpa_supplicant.conf, but it always boot into build-in system. **NO log left on the sd card**, only some record mp4 file, I also did reformat the SD card and change another SD card which works with other cameras.

> Camera version: YHS-113
> Build-in firmware version: 1.8.6.1C_201605191103

Heeere-GittyGitty, 2017-05-21:
Hi, having the same issue here. How can I see my firmware version? and Camera version? where do you see that?

cnberg, 2017-07-03:
This camera I have is 17CN version, so it's not compatible with this hack.