fritz-smh/yi-hack #174: Working firmware for 1.8.7.0A_201702081101 ?

galadril, 2017-05-10:
Is this fimware working for the Yi version 1.8.7.0A_201702081101??

eeztrbo, 2017-05-12:
Galadril did you get the hack to work on 1.8.7 firmware?

galadril, 2017-05-12:
nope

plexdev768, 2017-05-13:
broke for me as well. i tried everything.

0sm3l, 2017-05-17:
do you guys have the white or black camera? I was able to install on two white cameras and failed on two black. I wonder if there's something different in the firmware.

eeztrbo, 2017-05-17:
Black.

galadril, 2017-05-17:
White

0sm3l, 2017-05-20:
So I guess the color didn't matter. It must be just newer versions then.

keith721, 2017-05-27:
I should have a late-model Yi arriving from Amazon this afternoon.  I'll try to determine the firmware version and research it.

0sm3l, 2017-05-27:
I contacted support about a week ago and so far they can't figure out why these two cameras can't downgrade but the other ones can. They even sent me an older firmware and it didnt work either :(

keith721, 2017-05-27:
I'd guess they removed the console connector simply to save two cents per camera.  My problem is trying to solder wires to those tiny traces is next to impossible. :(

keith721, 2017-05-28:
Got a serial console connected tonight, and found the newest version of U-Boot is looking for a file named 'home_y18' instead of 'home'.  Unfortunately, U-Boot is not allowing an interrupt to change boot arguments.  But it does appear we can overwrite/update the 'home' filesystem again. :)

0sm3l, 2017-05-29:
I saw your post on slickdeals. Nice progress! I found some info about it here but sadly it is in Russian:

https://4pda.ru/forum/index.php?showtopic=638230&st=9740

I tried downgrading and sadly it seems it is not booting anymore lol. Oh well 😆

keith721, 2017-05-29:
I've noticed the camera has a tendency to just not initialize properly occasionally.  Unplug the power, press the reset button once or twice, then wait five or ten seconds, and try again.  Might work!

FYI, [here's a helpful link](https://yadi.sk/d/MpaQaTC1mU2kQ) that contains older versions of the camera firmware.

0sm3l, 2017-05-29:
Thanks for the link! I've tried everything and nothing works. It blinks blue twice and then the light stays on (orange). Reset button does nothing either. It seems like it has the same issue as this guy:
http://bbs.xiaomi.cn/t-13123979 (Chinese)
 No worries, it's going back anyway 😆

keith721, 2017-05-30:
Can anybody with a fairly recent firmware (1.8.5.x or 1.8.6.x) and telnet access provide a dump of their MTD flash RAM partitions?

    0x000000000000-0x000000040000 : "boot"
    0x000000040000-0x000000050000 : "env"
    0x000000050000-0x000000060000 : "conf"
    0x000000060000-0x0000001f0000 : "os"
    0x0000001f0000-0x000000330000 : "rootfs"
    0x000000330000-0x000000fe0000 : "home"
    0x000000fe0000-0x000000ff0000 : "vd1"
    0x000000ff0000-0x000001000000 : "ver"

Thanks!

AdySan, 2017-05-30:
@keith721 I have 1.8.6.1B_rtspfix from here http://xiaoyi.querex.be/firmwares/, would that help in anyway?

keith721, 2017-05-30:
@AdySan Thank you, but I'm afraid that would only be the 'home' partition.

AdySan, 2017-05-30:
@keith721 OK, I meant my camera came with 1.8.6.1B and I installed this version of the firmware to get rtsp working. Still wouldn't help? If it would, I'd appreciate instructions on how to dump the firmware via telnet.

keith721, 2017-05-30:
Oh!  Yes, that would be helpful.  Essentially, you use the Linux 'dd' command to read the MTDRAM block device and output to a file on the SD card.

    dd bs=512 if=/dev/mtdblock1 of=/tmp/sd/mtd1.img
    dd bs=512 if=/dev/mtdblock2 of=/tmp/sd/mtd2.img
    dd bs=512 if=/dev/mtdblock3 of=/tmp/sd/mtd3.img

AdySan, 2017-05-30:
Sorry, never played with Telnet. What am I doing wrong here?

```
pi@AdyPi3:~ $ telnet 192.168.1.128
Trying 192.168.1.128...
Connected to 192.168.1.128.
Escape character is '^]'.

(none) login: root
Password:
Welcome to HiLinux.
None of nfsroot found in cmdline.
# dd bs=512 if=/dev/mtdblock1 of=/tmp/sd/mtd1.img
dd: can't open '/tmp/sd/mtd1.img': No such file or directory
#
```

AdySan, 2017-05-30:
Here you go: https://www.dropbox.com/sh/peffv41syexua7l/AAD7bHYBFeUfW1AdT-L_zdwPa?dl=0

I have three files in here, is that all?

keith721, 2017-05-30:
Thank you, @AdySan !! I'll have to look at these after I get finished with a few conference calls.  It will be a couple of hours before then, though. :(  I'd much rather dig here than listen to these calls.

keith721, 2017-05-30:
If you could also dump /dev/mtdblock4 "rootfs" that might be helpful.  That would include everything in the "/" top-level/root directory which doesn't reside in the "/home" directory path.  I'm guessing that includes the original call-out to the /tmp/sd/test/equip_test.sh script that's been disabled in the 1.8.7.0A firmware.

/dev/mtdblock5 "home" is what we already have in all the various firmware files.  No need to dump this.

/dev/mtdblock6 "vd1" and /dev/mtdblock7 "ver" might be interesting/helpful as well.  Guessing they're more configuration/version information, or they might be blank/empty, as well.

I'll use the U-Boot 'mkimage' command to wrap these dump files, and attempt to flash/burn them on my camera using the SD card at boot-up time.  If that succeeds, I'll be able to hack my camera as we all expected in the first place. :)

AdySan, 2017-05-30:
@keith721 ok, three more files added to same link. There is no 

https://www.dropbox.com/sh/peffv41syexua7l/AAD7bHYBFeUfW1AdT-L_zdwPa?dl=0

BTW, there is no mtdblock7, here's the directory listinghttps://gist.github.com/AdySan/d0cecbba65845ce203a4ef75f827407f

keith721, 2017-05-30:
Thank you so very much, @AdySan :)  Funny there's no mtdblock7, as the kernel boot log clearly shows eight MTD flash RAM partitions, with the eighth one named "ver".  What I'm attempting right now is to build the U-boot image for the mtd1 "env" partition, with the bootwait value updated to '3'.  If that is successful, then I should be able to interrupt the U-boot sequence and add the 'single' argument to the bootargs variable, leading to a root shell.

bunver, 2017-05-30:
@keith721 go for it, Keith! My camera also has the same problem and I'm following you guys.

keith721, 2017-05-30:
I've got the env_y18 uImage file copied to my SD card . . . I'm about to find out if it works or not.  Thanks for the support!!

keith721, 2017-05-31:
Quite frustrating, honestly.  I'm currently using this mkimage command syntax

    mkimage -A arm -T filesystem -C none -a 4000 -e 0 -n '7518-hi3518-env' -d mtd1a.img  env_y18

and receiving the following lovely output from the U-boot startup process:

    2: home_y18!
     find index 5: home_y18
    
    u-boot_y18 not found!
    reading env_y18
    
    magic 0x27051956 0x27051956
    arch 0x2 0x2
    size 0x20000 0x40
    type 0x7 0x2
    crc 0xa351d2af
    name 7518-hi3518-env
    idx1: old : ffffffff, new : 592dd44d
    reading env_y18
    **write_len too big**

0sm3l, 2017-05-31:
Would it help to have the files from 1.8.7?

keith721, 2017-05-31:
It might, but I honestly don't think it will make any difference.  I just need to get this silly file size issue cleared up, then it should be good.  I chopped all the null/FF bytes from the end of the image file, and it still gave the "write_len too big" message.  That tells me something about the input file just isn't quite right.  

And while we're reviewing it, it would not hurt my feelings in the least if someone with a bit more knowledge comes in here and says, "Hey, dummy - it's THIS easy:   xxxxxx yyyyy zzzzz".  ;^P

0sm3l, 2017-05-31:
Here you go:

https://www.dropbox.com/s/mpdn83rhuq4qphb/mtd.zip?dl=0

Let me know when you download it so I can remove the file 👍

keith721, 2017-05-31:
Got it - THANKS!!

AdySan, 2017-05-31:
Dang just got a pack of 4 cameras with firmware 1.8.7.0A. Fingers crossed. No pressure.

keith721, 2017-05-31:
I changed the "-a 4000" value to "-a 0" so it should write the image at the start of the partition.  Even though that partition begins at 0x004000, it's still 0x0 or the beginning of the partition.  It still barfs.  So, I've got everything correct except for the size for some reason I haven't quite figured out, yet.  /me scratches head (a lot.)

keith721, 2017-05-31:
Hrmmm . . . the partition map from the boot log says:

    0x000000040000-0x000000050000 : "env"

which is a length of 0x10000 bytes, or exactly half what the image file says at 0x20000 

Maybe I should chop off 64K of the image file and try again...

0sm3l, 2017-05-31:
maybe @lambokini may know what's going on. He was working on it a couple of weeks ago but different camera. https://github.com/xmflsct/yi-hack-1080p/issues/3

bunver, 2017-05-31:
Help us @lambokini !

keith721, 2017-05-31:
ANNNNNNDDDDDDDD my Yi camera is permanently b0rken.  When trying to resolder it, the traces on the board just lifted right off, leaving no place to solder/connect. :( :( :(   Looks like it's time to grab another one from those Amazon folks. :(

AdySan, 2017-05-31:
@keith721 I'd be happy to try, just a few questions though
- Will this possible premanently brick my camera? If not how do I recover from a broken state?
- How do i flash this partion? some `dd` command?
- Do I need the serial wires soldered? Asking because, I was planning to return the pack or 4 cameras, opening and soldering stuff onto it might not be a good idea in that case.

0sm3l, 2017-06-01:
@keith721 sorry to hear about the camera!  I tried the file on my camera (the one that doesn't boot anymore) and it flashes blue twice every 7 seconds or so. Sadly I dont have a way to see the logs so I have no idea what's going on. 

@AdySan  you should try to downgrade the firmware first, if you are able to you don't have to worry about any of these workarounds :)

keith721, 2017-06-01:
@AdySan:

- brick the camera?  Not likely.  It will either load and take longer to boot, or not load at all.  You'd need a serial console connection to determine the actual status.
- how to flash?  Simply by placing it on the root directory of the SD card.  The U-boot loader will recognize it and attempt to load it.
- A serial console is really a must-have when debugging things at the U-boot level.

Thank you for offering, but I'll get back to testing it Thursday evening.  Thank goodness for Amazon Prime next day delivery! X^D

AdySan, 2017-06-01:
@0sm3l the tittle of this issue says, cannot downgrade 1.8.7.0A, is it still worth trying?

0sm3l, 2017-06-01:
@AdySan two of my cameras (white) were able to downgrade just fine. The other two (black) were not. So I would give it a try anyway.

keith721, 2017-06-01:
While waiting for the replacement camera to arrive this afternoon, I thought I'd describe my approach to getting 1.8.7.0A firmware to update.  

1. Enable BOOTDELAY=5 in the env_y18 partition image so we can interrupt the U-boot loader.
2. Append 'single' to the end of the BOOTARGS arguments to leave the console logged in, in single-user mode.
3. Determine where in the 1.8.7.0A firmware startup they've disabled/removed the call to the equip_test.sh script.
4. Re-enable the equip_test.sh script call, and create a new root_y18 image file.
5. Package and post the env_y18, root_y18, and home_y18 image files to enable the hacks moving forward.

keith721, 2017-06-01:
New camera arrived, and surprise, it has the serial port connector installed.  Plus, the firmware appears to be older, and not the 1.8.7.0A stuff.  Nevertheless, it will still allow me to test the env_y18 file I created. :)

keith721, 2017-06-02:
Where's the emoticon/emoji that looks like my avatar?  [This post](https://github.com/fritz-smh/yi-hack/issues/141#issuecomment-283756414) describes the correct utility for creating the env_y18/env partition image.  Doh!!

AdySan, 2017-06-02:
@keith721  `orz.. ` not exactly an emoji but a person on their knees and head on the floor. 

Looking forward to your progress!

zackoch, 2017-06-02:
I'm afraid to try and solder my jtag to these pins. Holy tiny batman...

keith721, 2017-06-02:
They're not JTAG, they're serial/console port.  And yes, they are ridiculously tiny.

zackoch, 2017-06-02:
haha I meant to say uart. I was thinking about jtag for some reason.

I know what I'm doing, I promise!

keith721, 2017-06-02:
Don't feel bad about it.  I was beginning to worry JTAG wasn't a thing any more.  But seriously, there is a large group of pogo-pin contact points near the CPU chip.  If you really wanted to JTAG it, you probably could. ;^)

keith721, 2017-06-02:
The output of the mkubootenv program from that other thread was rejected by the U-Boot loader, so it looks like I'll STILL have to prefix it with a 64-byte U-boot image header.  To make matters more entertaining, my USB-to-Serial cable is acting flaky tonight. :(

AdySan, 2017-06-02:
If you had a RPi handy you could just look at the serial output directly,
no need for the USB-Serial converter. Good luck.

keith721, 2017-06-02:
I just hope the micro 3-pin connectors I ordered off eBay will fit the connectors on these two cameras.  The solder pads just lifted off the one I was using Thursday night and tonight, so I'm onto my third camera.  It also has the 3-pin micro connector for the console, so I'm expecting older firmware on it, as well.

zackoch, 2017-06-03:
A few questions -

Where do you see env_y18, home_y18 for example? I see the partitions are called env, home, sans the "y18" bit. I don't see where you're seeing 
> Got a serial console connected tonight, and found the newest version of U-Boot is looking for a file named 'home_y18' instead of 'home'. 
All I'm seeing is uboot start and then boot the OS lol
```
System startup
=========1
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=0

 do_auto_update: detect flash size: 0x1000000
### UBOOT:test pin high! ###
Hit any key to stop autoboot:  0
16384 KiB hi_fmc at 0:0 is now current device
```

Are you basing this looking for ****_y18 stuff off of pulling apart the env image of 1.8.6 from mtd1 and seeing something interesting?

Is uboot configured to look for partition images at boot just sitting on the SD card and just flash them? I don't see how else you're able to even write to a partition without shell or JTAG. 

If I can understand a little bit more about this I might be able to help more.

keith721, 2017-06-03:
Actually, it looks like someone else updated the /test/equip_test.sh script at [this parallel project](https://github.com/death2all110/yi-hack/issues/29#issuecomment-305961612).  You all should try that script on your SD card and see if it then updates the hacks as expected.

Regarding env vs. env_y18 and home vs home_y18, it seems the very latest versions of this camera (without the 3-pin serial port socket soldered in place) have a newer U-boot bootloader loaded that's looking for the _y18 variant of the file names.  I found this out when first attempting to hack the new camera.  Here's the output of the serial console on that camera:
    
    System startup
    =========1
    hello, open console
    
     do_auto_sd_update: detect flash size: 0x1000000, gsdexist=1 
    Interface:  MMC
      Device 0: Vendor: Man 035344 Snr 09d21606 Rev: 8.0 Prod: SS04G
                Type: Removable Hard Disk
                Capacity: 3781.5 MB = 3.6 GB (7744512 x 512)
    Partition 1: Filesystem: FAT32 "NO NAME    "
                test/
      8402712   home_y18 
        13218   equip_test.sh 
    
    2 file(s), 2 dir(s)
    
    1: home_y18!
    
     find index 5: home_y18 
    
    2: equip_test.sh!
    
    u-boot_y18 not found!
    
    env_y18 not found!
    
    conf_y18 not found!
    
    kernel_y18 not found!
    
    rootfs_y18 not found!
    reading home_y18
    
    magic 0x27051956 0x27051956 
    arch 0x2 0x2 
    size 0x8036d8 0x40 
    type 0x7 0x2 
    crc 0x8fcfcdc7 
    name 7518-hi3518-home 
    idx5: old : 56a895ec, new : 589a8f5b
    reading home_y18
    þ

zackoch, 2017-06-03:
haha I wonder if the SD card needs to be installed - no wonder why I wasn't seeing the y18 stuff...

> 
System startup
=========1
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=0

 do_auto_update: detect flash size: 0x1000000
### UBOOT:test pin high! ###
Hit any key to stop autoboot:  0
16384 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 81000000 ...
   Image Name:   7518-hi3518c-kernel
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1385848 Bytes = 1.3 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0
Linux version 3.4.35 (chenshibo@ANTS-SH-SV02) (gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300) ) #3 Tue May 31 17:35:37 CST 2016
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: hi3518ev200

It goes from uboot to linux without checking when there's no SD card. I was like uhhh what is this guy seeing, maybe I'm blind.

AdySan, 2017-06-04:
So.... with [this](https://github.com/death2all110/yi-hack/pull/33) PR, can we consider this problem solved for the 1.8.7.0A firmware version? I'll give it a try as soon as I'm able to.

bunver, 2017-06-04:
Not actually,  I tried and it didn't work for me. I used two camera and no success.

zackoch, 2017-06-04:
I tried, I also tried to flash and I screwed one of my cameras now. It hangs while it loads the kernel :-( I'll troubleshoot some more tomorrow

[63 more replies]