fritz-smh/yi-hack #176: what's the range of your YI's WIFI connection??

resident12, 2017-05-15:
Somehow a free space range of just 5meters seems to be the range limit for the distance between my Yi and my WIFI modem. Beyond this range the camera is deregistered and never reregistered again. Rather a disappointment, for my cell phones WIFI easily ranges up to 30meters. What range is your Yi reliably capable of and what could I do to extend this range?

Yi:                  type=22EUTX8GAB, model=YHS-113-IR, firmware version=1.8.5.1M_201512011815
Wifi modem: brand=Fritz!box, type=7390

bronzerain, 2017-05-17:
resident12: I own 7 of the yi cameras. I have one working outside in a metal security light case and it never has a connection problem, even in the metal case it is in. Two of the other yi cameras have WiFi problems when they are more than 5 meters from the WiFi access point. I wish there were a way to increase the WiFi capabilities of the yi cameras. The XiaoFang cameras have a separate antenna but the yi cameras seem to be more reliable. Does anyone know of a small camera that has RTSP already enabled out of the box? It also would be nice to be able to use an Ethernet connection  for reliability.

keith721, 2017-05-27:
The Yi camera DOES have a U.FL snap-on connection for the internal antenna.  This would allow a bit of creative engineering to install an RP-SMA connector on the stand for higher-gain (3, 5, 9 dBi) antennas.

RiRomain, 2017-08-18:
Same here, 5 meter away from the access point, and the camera cannot connect.
Phone and other device don't have problem at 15+m even on another floor!

Since it uses a U.FL connector, this hack should also work:
https://github.com/samtap/fang-hacks/wiki/External-Wi-Fi-Antenna
The antenna attachment might, however, not be as aesthetic!