fritz-smh/yi-hack #177: Blurry Video

0sm3l, 2017-05-16:
Are there any options to set advanced video parameters such as brightness/sharpness/bitrate etc? I tried changing the file /etc/ui.conf but it doesn't seem that this file does anything at all anymore.
I've seen many scripts for the Yi action camera that seems similar to what I'm trying to accomplish but nothing for the "home/ant" cameras. Any help would be appreciated.

keith721, 2017-05-27:
Is it perhaps a focus/lens issue?  If so, disassemble the camera, remove the glue from the lens threads which locks it in place, and while viewing live video from the camera, gently turn the lens in either direction until you get the most clear image.

0sm3l, 2017-06-17:
I just tried and it makes a HUGE difference. Thanks for the tip!

[image]

closed 2017-06-17 by 0sm3l

querex007, 2017-06-18:
Damn what a hudge difference, thanks for the tip!