fritz-smh/yi-hack #178: How can I convert own .g726 audio file?

alpozdemir, 2017-05-18:
Hi All,

I want to translate xiomi yi ant audio file to other languages. Im trying to convert from mp3 to g726
But unfortunately yi ant not correct playback my converted audio files. Any idea?

keith721, 2017-05-30:
It looks like [the Audacity program](http://forum.audacityteam.org/viewtopic.php?f=28&t=72608) can do what you want.

andy2301, 2017-08-06:
@alpozdemir @keith721 were you able to play the converted g726 file? I followed the instructions but the converted g726 file won't play in the camera.

keith721, 2017-08-06:
Did you try the ffmpeg command in the link I provided?

    ffmpeg -i test.mp3 -acodec g726 -ar 16k -ac 1 test.wav

That should take an mp3 file and output a wav file encoded with g726.

andy2301, 2017-08-07:
@keith721 yes of course I used the command provided (had to change -ar to 8k in order for it to work). I heard static noise when running "/home/rmm test.wav 1" on the camera, while running "/home/rmm /home/welcome.g726 1" I could hear the official welcoming voice. 

I also downloaded the welcome.g726 file from camera to my pc, but the audio file doesn't play at all in all my audio players. "avprobe" returns "invalid data found when processing input". 

It almost looks to me that the .g726 audio file used in the camera is some kind of encrypted audio format that only /home/rmm can decrypt and play. Anybody has any insight into it?