fritz-smh/yi-hack #179: MOTION_NOTIFICATION_URL

theflarenet, 2017-05-21:
Apparently it's not working for me. I've changed it from:

`MOTION_NOTIFICATION_URL=`
to
`MOTION_NOTIFICATION_URL=http://www.myaddress.com/webpage_to_accept_json.php
`

It doesn't seem to execute at all and using wget to that address worked fine.

Any ideas?

theflarenet, 2017-05-21:
Ok. Found that it was an issue due to the host blocking the request:
```
{"armed": true, "motion": true, "last_motion_check": 1495355441, "host": "(none)"}
<html><head><title>Error 406 - Not Acceptable</title><head><body><h1>Error 406 - Not Acceptable</h1><p>Generally a 406 error is caused because a request has been blocked by Mod Security. If you believe that your request has been blocked by mistake please contact the web site owner.</p></body></html
```

Is it possible to use something else instead of curl? Perhaps wget as a workaround? My server error logs reported: `[21/May/2017:04:37:08 -0400] "POST /test.php HTTP/1.1" 406 300 "-" "curl/7.37.1"`

Clem-, 2017-10-10:
It's because your **webpage_to_accept_json** is not receiving a JSON request. 

You may want to change your notification script with something else or try to add parameter `-H "Content-type: application/json"` to the CURL call.