fritz-smh/yi-hack #180: FTP upload

theflarenet, 2017-05-22:
I can connect to the FTP successfully and upload files however reconnecting to the FTP doesn't display the files uploaded. Instead of showing me the contents on the SD card or whatever path it directs to, nothing is shown. Although the files uploaded are located on the SD card... just not accessible through FTP. 

Is this intentional or a known bug?

hyptechdev2015, 2017-06-07:
@theflarenet 
it looked like the lastest firmware version does not have permission to access ftp upload script

hyptechdev2015, 2017-06-07:
Hi @theflarenet 

try this way.

Telnet to your camera, using putty.ext or other client.
>telnet "your camera  IP"

once in Telnet
#cd /tmp/hd1/test/scripts/
#./copy_to_ftp.sh

to view the processes
#ps -w | grep sh

[image]

chuckjuhl, 2017-06-12:
I have a brand new yI cam. Trying to hack it for using the BlueIris. I cannot telnet the camera. Just get a connection rejected error. Any ideas?

chuckjuhl, 2017-06-12:
Can't ftp either.

theflarenet, 2017-06-14:
@hyptechdev2015 `copy_to_ftp.sh` does not exist in `/tmp/hd1/test/scripts/`

MrCook1es, 2021-02-14:
Hi, if i manually run copy_to_ftp.sh script it works fine.
But if I type_crontab -e_ and put @reboot /home/hd1/test/scripts/copy_to_ftp.sh
it does nothing? what am I wrong?