fritz-smh/yi-hack #182: Hack for Xiaomi Mijia (not Fang or 360)?

AdolarB, 2017-05-30:
Hallo @all,
is there a hack for Mijia like this http://www.gearbest.com/ip-cameras/pp_615275.html or can somebody tell me the adress (http or rstp) for watching in iSpy or ipcamViewer?

Many Thanks,
Adolar

bronzerain, 2017-05-30:
I own the Xiaomi mijia 1080P Smart IP Camera. It compares to the US version of the YI home camera 2. But I think the Xiaomi mijia 1080P Smart IP Camera is more advanced. It can even utilize the Xiaomi router with built in NAS. This is not turned on yet. I would love to get RTSP working on this camera.

AdolarB, 2017-05-30:
Yes, it is more advanced. Not only with Xiaomi Router it can copy the video files to FTP. I have a asus router with a usb drive. The cam put all videos to the usb drive. also you can copy to cloud storage - i think so.
Nobody a idea how i can see a stream on PC? HTTP, RSTP?

bronzerain, 2017-05-31:
The motion detection on the Xiaomi mijia 1080P Smart IP Camera seems pretty good with having the different motion zone features. Would it be possible to FTP the motion alerts from the camera to a Blu Iris server?

AdolarB, 2017-05-31:
sorry, i don't know this server. How i say, the cam put the video files to an asus router, but it seems me, that not the motion detection files will put to there. It seems to be the continius recording files. The menue in the app is confused, i must some times test it.
sorry for my english, the scool is a long time ago...

ilostthegiraffe, 2017-06-02:
I also have this camera, the current hacks don't seam to work.

LePeR1898, 2017-07-07:
I'm looking for hacks for this cam also

Filipowicz251, 2017-07-15:
I'm waiting for hack for 1080p Mijia camera too :) But i've searched everywhere - and there is nothing till now. 

On another github project (fang hacks https://github.com/samtap/fang-hacks) one of contributors asked me if i can send him hardware pictures - so that he can watch it and compare to fang 1080p camera and maybe propose hack for mijia too. 

But i'm no hardware guy and in this model there is no "easy" dismantle way. So if someone has the "Mijia 1080p camera", which he can spare (maybe bricked) and can turn it to pieces and make photos - we would be most grateful. 

Here is thread where such photos can be added:

https://github.com/samtap/fang-hacks/issues/190

niighthawk, 2017-08-01:
I also want a hack for the sxj01zm ( Mijia 1080p ) to get acces to RTSP

tobbegutt, 2017-08-07:
OMG! yes! Can we please get this to rtsp or to be able to do something more with it?
I need it to record more than 10 seconds clip and i dont want the other alternative wich is continous record 24/7 to my NAS.
You see, i have sdcard installed, and on motion detection it records only 10 seconds. 
There is no option to have it record as long as its detect motion. it cuts out after 10 seconds every time.
The other possibility is just to let it record all day and night to my NAS but i cannot imagine the stress on the NAS and harddrive to be continously recording all day.
Its really not necessary. 
Its a great camera and its cheap and you can acces it from all over the world as long as its connected to wifi.
Some people have problems my self included to teach it the wifi using QRcode after resets but just bypass it and let it connect to Bluetooth instead of QR and you will have no problem during set up.

But yeah, im running the MIHOME application on my phone.
I used to have Foscam and had ISPY on my laptop wich recorded movement and uploaded it to Gooogle drive, This is not possible now as it will not let you select spesific folder inside the NAS strangely enough.
Very sad.
Ill donate to whoever fix this as i like the cam i want to have even more of them. I have 4 at the time beeing.

LePeR1898, 2017-08-07:
@tobbegutt

The device records more than 10 secs at a time but for the notification alerts it sets it to 10 secs to limit bandwidth and let you quickly see what set it off. If you goto manage files or view them directly either on the camera or nas you'll notice videos are alot longer than 10 secs.

Also just set it to "record and save alarm video to sdcard when motion is detected" so it doesn't stress anything.

tobbegutt, 2017-08-08:
Aha. Ok. But, do you know if i can make a nasfolder it will recognize so i can have a Google drive folder or Dropbox folder it can download to? Its better to have detection files to be uploaded in case i do have a burglary and they steal my NAS. Or can i use like a external hard drive and connect it using usb on my router and have a folder thats synced via internet? Im looking for solutions for having my files automatically upload to internet. Just in case. A sdcard is of no use if someone take the camera either so its all gotta be uploaded..

LePeR1898, 2017-08-08:
Well if you setup the nas to say a Windows PC share you could install google backup and sync and set it to backup the folder .

sendorm, 2017-08-08:
I would like to see this work too.

Filipowicz251, 2017-08-10:
Hi

I've created github project for hacks for Mijia 1080p. 

https://github.com/Filipowicz251/mijia-1080P-hacks

I've created also a topic for this:
https://github.com/Filipowicz251/mijia-1080P-hacks/issues/1

ChiWestSider, 2017-08-12:
I finally got my Xiaomi Mijia 1080p  camera with NAS storage working. The Live/remote viewing works and the motion detection with 1 minute clips also works!  I think this covers all the things most people like myself were trying to do.  The remote viewing is done through the  Mi Home app as well as other features.  I could not get it set up to work with my WD MyCloud network drive, I kept getting an error immediately after getting a message saying setup successfully completed.  I used a drive connected to my Asus AC87R by USB and it let me log in to the drive, then shared files folder,  and also go into a specific sub-folder called Videos to send my files from the Mijia.

tobbegutt, 2017-08-12:
? any chance you could screenvideo your setup? im not 100% in English​ language so i would love if you either write a tutorial not or video it. thank you my brother!!

Sendt fra min Mi-telefon

sendorm, 2017-08-12:
The feature I need is jpg streaming. So that I can use the camera on home assistant front end. Also it would be good to disconnect the chinese server connection. The camera still produces images/video even if it is in sleep mode. The night ir lights are on when the light level drops-> image sensor still working during sleep. 
Anything through mi app is not wanted for me.

ChiWestSider, 2017-08-12:
[image]
[image]

tobbegutt, 2017-08-12:
> [screenshot_20170812-150451]<https://user-images.githubusercontent.com/30639807/29243990-c860c00e-7f72-11e7-880c-4bc95bd4a788.png>
> [screenshot_20170812-150459]<https://user-images.githubusercontent.com/30639807/29243991-c861ee7a-7f72-11e7-91f0-810a5fadadd1.png>

hmm. i have only 2 gb cards in my cameras and i even tried turning off save to sdcard but it still records to sd and fill them.up in few hours time and then its all gonna be errors. i cannot get it to stream back from my NAS using mihome. its all just missing something. do i need to set password to my NAS or something? I can make a screenvideo if something is unclear..

Sendt fra min Mi-telefon

ChiWestSider, 2017-08-12:
I don't know what the difference was, but trying to use my WD (Western Digital) MyCloud drive,  attached through ethernet, would not work.  Using my Network attached storage connected to my Asus 87R by USB works fine now.

ChiWestSider, 2017-08-12:
NAS storage will NOT become a choosable option until MicroSD card storage is added.  I have added it but "Record and save alarm video to SD..."  is not being used by me (the slider is set to Off)

ChiWestSider, 2017-08-12:
@tobbegutt 
 Do you see the 'pause transmission' button in my screen shot?  Do you see that I am using a subfolder  (Videos) of my 'Shared folder' root file as my NAS/Network drive?

tobbegutt, 2017-08-13:
[cid:_wpsmail@a5uud6lq78lun7jr0o0049p3.1502586015896]
So. If i set it to pause it will just stop sending files to my NAS all the time? Just the motion detection files?
Sorry, much questions

Sendt fra min Mi-telefon

ChiWestSider, 2017-08-13:
I just checked, and since I had that camera in sleep mode since I tested it while responding to you there are no files present.  I had not 'paused transmission' (the option under NAS settings) yet it had no files recorded, but did populate blank tiles with times by hours.  While typing this I also tested the 'Pause transmission' function.  It functioned the same, even using data.  I originally thought it had not started back recording to NAS, but once I used the 'pause transmission' function, the video filed appeared in the Mi Home Gallery.

You are definitely in the right place to ask questions.  You are not a bother one bit.  This is where I come to learn myself, and if my minute knowledge can help someone else in the process... Great!
[image]
[image]

tobbegutt, 2017-08-13:
Im so confused, let me try clear this out.
I have 2 GB cards and they fill up even if i have set the "save to sdcard" to off.
I can only find 8MB files on my NAS wich translates to 1 minute video.
I jsut want the camera to not use the sdcard so it fills up but instead just save motion detection files to my NAS and not make folder for every hour with a lot of files in each.
It seems like as soon as my sdcard are filled up the videoes on my NAS are just black screens and for no good use.
I tried reformatting all cards now, and set the files for save only 1 day.
maybe that was the issue?
I also set "pause transmission" just to se what it does.

It was so much easier to have foscam and use Ispy surveillance program and rtsp so i could have my computer as acces and file to be saved directly to my google drive folder.

On my NAS i have 4 hdds, and have just set them all to open so i can access all folders from any device.
and the video from these cameras as i understand will not allow me to select a network folder inside the NAS but it just make a xioami video files folder with the mac adress and i cannot have them to be synced with google drive.

I dont either have all the bandwidt to be uploading several GB of data every day from all the files that are not from motion detection.
Sorry, im just confused to how to make this work at least while we wait for it to be hacked with FANG.

Thank you, i know its confusing to read my writing but English is not my native language.

simwe, 2017-08-21:
I disassembled the camera...its hard but possible. Basically use a credit card or small screwdriver and wedge it up along the edge. Be extremely careful when lifting it up, go around the edge with the screw/credit card first and open it slowly, or else the ribbon will be damaged.

on the back of the pcb, there are 4 connections (not the ones cloest to the USB, not sure what they are).
I think it was:

bottom one: 3.3V
second: TX
third: RX 
and at the top: GND

But i can't really do anything with the console, seems to be locked. Can't get a shell, so I don't know. 
Any tips?
Would be nice with rtsp....and I upgraded....dang it. 

I googled the output, someone has uploaded some of the output here
https://pastebin.com/53hdb4x0

Heres my output: https://pastebin.com/D6iHb8ef (I couldn't get beginning of it. can't start with uart adapter connected for some reason)

Thats all for now....

tigerhil99, 2017-10-21:
hi
i have a Xiaomi Mijia 1080P security camera, have 2 issues, the camera seems not to work if i toggle between wifi and 4g, somehow it picks the wrong ip when coming back to wifi from 4G. secondly i cannot get it to work on recording on my NAS WDMYCloud....anyhelp? thnx

ChiWestSider, 2017-10-21:
Are you using a SD card in combination with NAS? I couldn't get it to work without using both.  I have no issues switching between Wi-Fi and 4G, so I can't help you there.   I can't get my 2nd camera to utilize my NAS at the same time as my other Mijia camera tho.

tigerhil99, 2017-10-22:
thanks ChiWestSider, it works in comibination. have 3 cameras but only 1 works on 4G at a time..anyone else has this problem? cant get to work all 3 on cellular? strange ,,,,anyhelp

tigerhil99, 2017-10-27:
@ChiWestSider ..thanks...4G working on all 3 cameras, but Nas working only in 1 @ChiWestSider were you able to figure out your second camera to utilize NAS. I see that folders are made for each camera but only 1 has files transmitted from Sdcard the the two camera folders are empty.

ChiWestSider, 2017-10-27:
You're welcome. The latest update screwed me up even more.   I have to log in again to the app and I don't know my credentials.  I will play with it again when I have time.  Also,  only 1 of my 2 cameras is password protected,  the other isn't and I don't know how to make it.  No,  I still only have 1 camera on NAS  (The one that's password protected)

iisworks, 2017-10-29:
Has anyone managed to save files to a Windows Share? The app sees my PC's, but no shares are visible in the next step after I enter the credentials. The only device I can actually save files to is a WD WirelessPlus wifi drive.

subashp, 2017-10-30:
How did you guys get the camera stream(even in xiaomi app) outside LAN?

ChiWestSider, 2017-10-30:
Within the app. Nothing special to setup.  I have it working on my phone via 4g

tobbegutt, 2017-11-04:
i must be dumb or something. or i missed a step.
I tried downgrade to the recovery img. and then i installed version 6 but i cannot acces the stream in vlc. and i tried other programs as Ispy connect to search for the rtsp but no result.
how can i determine that rtsp is working?
Can i acces it from a webbrowser?
here is my prosedyre..
my cam was at 39 firmware.
installed recovery img, then after boot put in sdcard with 0.6 hack.
it booted but no picture in mihome and it got really really hot after a while.

now i will try this.
1. pair the camera with mihome.
2. downgrade with recovery to earlier version.
3. install 0.7 on the sdcard i have reformatted with fat32 and reboot.
check network ip address and try the rtsp using vlc again.

did i miss any step? or can i confirm working rtsp without vlc?

tobbegutt, 2017-11-04:
i cant get it to downgrade? ive renamed it removing .bak and just turn it on right? or do i need to reset ut with img file?

AdolarB, 2017-12-03:
Hallo Friends,

here was a lot of action since my first question. You are heros :).

Just i will test it. Many thanks about your work.

-The rtsp stream stop after a few seconds but the record at sd card goes on. 
-How can i connect the cam by smb or FTP?
-upload to FTP i will test next week

dannythegreat, 2018-05-27:
hello, someone managed to use the camera in http mode? A jpg every few seconds? I need to conn3ct the camera in octoprint on my raspberry.

pareklinon, 2018-05-30:
Hey there great effort!!! I also want to know If we can use the camera in http mode. One more question. How can I change timestamp in order to see video clips in my timezone (Europe) and not China's?