fritz-smh/yi-hack #185: Can't connect to Wi-Fi

juan22a91, 2017-06-07:
Hi. I have been struggling with this hack since I tried to go back to original firmware. When I installed it first, everything worked fine. I was able to get the RTSP streaming without any problems. But after I decided to go back to orinal firmware, I can´t connect to Wi-Fi and it sends after some time the Wi-Fi password error, even if I know the password is OK and double checked it.

After that, I tried to go back again to the RTSP hack and I was not able to connect to Wi.Fi either. The LED starts blinking orange when flashing the firmware, then it flashes blue when finding Wi-Fi connection, it stays blue for a few seconds and then it starts blinking orange again.

In the log file I found something related to the ra0 interface.

Help please

hyptechdev2015, 2017-06-08:
@juan22a91 
please share your log files

juan22a91, 2017-06-09:
hi @hyptechdev2015 here is my log file

Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=255  (0 is ok)
Wifi configuration answer: Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
Could not set interface ra0 flags (UP): Operation not permitted
WEXT: Could not set interface 'ra0' UP
ra0: Failed to initialize driver interface
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Wed Dec 31 17:00:24 GMT 1969
Done
New datetime is Wed Dec 31 17:00:25 GMT 1969
Start blue led on
Start ftp server...
Check for ftp process : 
 1227 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1254 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1318 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  623 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  682 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  947 root         0 SW   [hidog]
  985 root      1164 S    ./peripheral
  986 root      1144 S <  ./dispatch
  987 root      1064 S    ./exnet
 1227 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1254 root      1408 S    ./serverM 80
 1272 root      3360 S    ./record_event
 1273 root     14728 S    ./mp4record 60
 1302 root      1716 S    /usr/sbin/crond -b
 1312 root       904 S    /home/led_ctl -boff -yfast
 1313 root     97496 S    ./rmm
 1318 root     27252 S N  ./rtspsvrM
 1435 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      3.4M     88.0K  98% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    916.0K  90% /home
tmpfs                    32.0M     40.0K     32.0M   0% /tmp
/dev/hd1                  7.4G      9.4M      7.4G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                  7.4G      9.4M      7.4G   0% /tmp/hd1/test/http/record

juan22a91, 2017-06-09:
And here is the log file for the original firmware (1.8.7.0A_201702081101)

[][12/31/17:0:6:246]: fopen in file fail!
[][12/31/17:0:6:482]: decrypt file fail!
[/home/rmm /tmp/au][12/31/17:0:8:562]: rmm_init_mpp ok!
[/home/rmm /tmp/au][12/31/17:0:8:704]: start do HI_MPI_AI_GetVqeAttr
[/home/rmm /tmp/au][12/31/17:0:8:728]: ai init ok!
[/home/rmm /tmp/au][12/31/17:0:8:729]: rmm_init_ai ok!
[/home/rmm /tmp/au][12/31/17:0:8:733]: rmm_init_adec init ok!
[/home/rmm /tmp/au][12/31/17:0:8:734]: rmm_init_adec ok!
[/home/rmm /tmp/au][12/31/17:0:8:907]: ao init ok!
[/home/rmm /tmp/au][12/31/17:0:8:936]: rmm_init_ao ok!
[/home/rmm /tmp/au][12/31/17:0:8:937]: file=/tmp/audio/welcome.g726
[/home/rmm /tmp/au][12/31/17:0:11:702]: send_ao_beep: read file /tmp/audio/welcome.g726 not equal u32ReadLen(56)
[/home/rmm /tmp/au][12/31/17:0:12:897]: rmm_init_mpp ok!
[/home/rmm /tmp/au][12/31/17:0:13:66]: start do HI_MPI_AI_GetVqeAttr
[/home/rmm /tmp/au][12/31/17:0:13:67]: ai init ok!
[/home/rmm /tmp/au][12/31/17:0:13:67]: rmm_init_ai ok!
[/home/rmm /tmp/au][12/31/17:0:13:68]: rmm_init_adec init ok!
[/home/rmm /tmp/au][12/31/17:0:13:68]: rmm_init_adec ok!
[/home/rmm /tmp/au][12/31/17:0:13:224]: ao init ok!
[/home/rmm /tmp/au][12/31/17:0:13:252]: rmm_init_ao ok!
[/home/rmm /tmp/au][12/31/17:0:13:253]: file=/tmp/audio/start_wait.g726
[/home/rmm /tmp/au][12/31/17:0:15:721]: send_ao_beep: read file /tmp/audio/start_wait.g726 not equal u32ReadLen(48)
[/home/detect_ver][12/31/17:0:23:385]: get_board_ver,  [ hwplatform=2
$, 2 ] 
[/home/detect_ver][12/31/17:0:23:547]: start_rtc, [ hisi rtc ]
[/home/detect_ver][12/31/17:0:23:604]: start_rtc ok.
[/home/detect_ver][12/31/17:0:23:714]: start_cpld_periph ok.[/home/detect_ver][12/31/17:0:23:714]: finished. 
[./dispatch][12/31/17:0:23:877]: cmd: renice -n -5 -p 1013 ok![./dispatch][12/31/17:0:24:484]: GetUiCfgFromFile success[./dispatch][12/31/17:0:24:485]: DISPATCH_init: language is 0, zkIndex 0, /etc/gb2312.zk [./dispatch][12/31/17:0:24:512]: DISPATCH_init: change to /etc/gb2312.zk ok! [./dispatch][12/31/17:0:24:513]: DISPATCH_init: test_result_file(10000) 
[./dispatch][12/31/17:0:24:513]: DISPATCH_init: GetVersionFromFile ok![./dispatch][12/31/17:0:24:678]: DISPATCH_init: do_reset  sn_string(2498KMJRKW7LPJTSHV111ApklgPpeBUS4jVx2Hbe3ef4180d754586f2fadb61fffc4d52) gpmmapinfo->sn(be52e675a2c3b59afb738b535374f9e6) bind_timeout(1800)
[./dispatch][12/31/17:0:24:690]: Dispatch MsgHandle start 
[./rmm][12/31/17:0:25:8]: 
check PO3100K _____RockGetSensorType__line__645___data:0xff_____
[./rmm][12/31/17:0:25:335]: 
check MT9M034_DATA_____RockGetSensorType__line__660___data:0xffff_____
[./rmm][12/31/17:0:25:335]: 
check OV_PIDH_DATA_____RockGetSensorType__line__679___data:0xff_____
[./rmm][12/31/17:0:25:336]: 
check OV9732 RockGetSensorType__line__759___data:0x97_____
[./rmm][12/31/17:0:25:336]: rmm_init_mpp ok!
[./rmm][12/31/17:0:25:337]: start do HI_MPI_AI_GetVqeAttr
[./rmm][12/31/17:0:25:395]: ai init ok!
[./rmm][12/31/17:0:25:396]: rmm_init_ai ok!
[./rmm][12/31/17:0:25:831]: rmm_init_aenc init ok!
[./rmm][12/31/17:0:25:832]: rmm_init_aenc ok!
[./rmm][12/31/17:0:25:863]: rmm_init_adec init ok!
[./rmm][12/31/17:0:25:873]: rmm_init_adec ok!
[./rmm][12/31/17:0:26:76]: ao init ok!
[./rmm][12/31/17:0:26:77]: rmm_init_ao ok!
[./rmm][12/31/17:0:26:78]: start do HI_MPI_AI_GetVqeAttr
[./rmm][12/31/17:0:26:252]: 
system call back libsns_ov9732_720p.so_____________________
[./rmm][12/31/17:0:26:350]: 
____________come [ Rock_COMM_ISP_InitAE ]___________________
[./rmm][12/31/17:0:26:352]: 
_____Gamma test for table 2__________
[./rmm][12/31/17:0:26:353]: 
set auto WB	[./rmm][12/31/17:0:26:355]: rmm_init_isp ok!
[./rmm][12/31/17:0:26:458]: rmm_init_vi ok!
[./rmm][12/31/17:0:26:468]: 
_______VpssChn = 3, stVpssExtChnAttr.u32Width = 640, stVpssExtChnAttr.u32Height = 360 __________________
[./rmm][12/31/17:0:26:469]: 
_______VpssChn = 6, stVpssExtChnAttr.u32Width = 320, stVpssExtChnAttr.u32Height = 240 __________________
[./rmm][12/31/17:0:26:470]: rmm_init_vpss ok!
[./rmm][12/31/17:0:26:471]: frame rate=20 20 700 40
[./rmm][12/31/17:0:26:475]: 
---------Rock_COMM_VENC_BindVpss VencGrp = 0, VpssGrp = 0, VpssChn = 0
[./rmm][12/31/17:0:26:476]: rc set success! venc(0)
[./rmm][12/31/17:0:26:478]: frame rate=20 20 230 40
[... 17 lines cut ...]
[./rmm][12/31/17:0:28:24]: get_data_frame width =352  height =192
[./rmm][12/31/17:0:28:310]: get_data_frame width =352  height =192
[./rmm][12/31/17:0:28:564]: get_data_frame width =352  height =192
[./rmm][12/31/17:0:28:810]: get_data_frame width =352  height =192
[./rmm][12/31/17:0:28:849]: send_ao_beep: read file /tmp/audio/wait.g726 failed u32ReadLen(0)
[./rmm][12/31/17:0:29:249]: get_data_frame width =352  height =192
...
...
...
[./rmm][12/31/17:1:46:244]: decoded QR-Code symbol "b=USnXbreLpoOdhMvi&s=Sk1DX0hPTUU=&p=UkwrKGFYDgF5"
[./rmm][12/31/17:1:46:245]: trans_json info=b=USnXbreLpoOdhMvi&s=Sk1DX0hPTUU=&p=UkwrKGFYDgF5;key=&p=
[./rmm][12/31/17:1:46:246]: trans_json result=UkwrKGFYDgF5
[./rmm][12/31/17:1:46:247]: trans_json info=b=USnXbreLpoOdhMvi&s=Sk1DX0hPTUU=&p=UkwrKGFYDgF5;key=&s=
[./rmm][12/31/17:1:46:248]: trans_json result=Sk1DX0hPTUU=
[./rmm][12/31/17:1:46:266]: trans_json info=b=USnXbreLpoOdhMvi&s=Sk1DX0hPTUU=&p=UkwrKGFYDgF5;key=b=
[./rmm][12/31/17:1:46:268]: trans_json result=USnXbreLpoOdhMvi
[./rmm][12/31/17:1:46:431]: rmm got /tmp/got_wpa ssid(JMC_HOME) pwd(juan22a91) bind(USnXbreLpoOdhMvi)
[./dispatch][12/31/17:1:46:490]: bind_key(USnXbreLpoOdhMvi)
[... cut ...] _ao_beep: read file /tmp/audio/scanok.g726 not equal u32ReadLen(16)
[./rmm][12/31/17:1:52:243]: send_ao_beep: read file /tmp/audio/connectting.g726 not equal u32ReadLen(72)
[/home/monitor_dum][12/31/17:3:32:129]: password is timeout
[./rmm][12/31/17:3:32:748]: get_data_frame width =352  height =192
[./rmm][12/31/17:3:32:843]: get_data_frame width =352  height =192
[./rmm][12/31/17:3:33:131]: get_data_frame width =352  height =192
[./rmm][12/31/17:3:33:318]: get_data_frame width =352  height =192

RiRomain, 2017-08-18:
I have the same problem, I tried multiple wifi access point, it just cannot connect anymore...