fritz-smh/yi-hack #190: 17CNYJXCLF013P160711 accidentally upgrade now it doesnt works

imrich82, 2017-06-26:
I bough this xiomi yi cctv 17CNYJXCLF013P160711 camera 4+ months ago, i'm able to perform https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/ and make it working.

But last 3 days, i forgot and click on update on the firmware now the firmware was 1.8.7.0C_201705091058 and i can't downgrade it to any firmware.

I already try 
1.8.5.1K_201508311131
1.8.6.1B.02
1.8.6.1B_201603181307

all not working, it doesnt want to downgrade, can some please help me.

Thank you

quachtai1959, 2022-06-05:
Dùng bản 1.8.6.1B.02 sẽ ok. Sau đó lấy thẻ nhớ ra chép bản hack. Tôi đã thfnh công. Bạn nên dùng thẻ nhớ tốt nó mới hoạt động.

quachtai1959, 2022-06-05:
Using version 1.8.6.1B.02 will be ok. Then take out the memory card and copy the hack... I succeeded. You should use a good memory card for it to work.