fritz-smh/yi-hack #192: Run copy_to_ftp.sh on startup

tonyrulez, 2017-07-03:
I'm not familiar with the text editor, how can I add it to crontab?
After running crontab-e, I can type in:
`@reboot /scripts/copy_to_ftp.sh`
(yes, I put it in a different folder)
But I can't figure it out how to save :)

tonyrulez, 2017-07-28:
Tried the 2following:
Adding '@reboot /scripts/copy_to_ftp.sh' to crontab via 'crontab -e' - does not start the .sh
Adding
'/scripts/copy_to_ftp.sh || exit 1
exit 0'
to '/etc/rc.local' - does not start the .sh (checked via 'ps -w | grep sh')

Now how the heck can I make it to start automagically?

MrCook1es, 2021-02-14:
same problem here..

if i manually run copy_to_ftp.sh script it works fine.
But if I type_crontab -e_ and put @reboot /home/hd1/test/scripts/copy_to_ftp.sh
it does nothing? what am I wrong?