fritz-smh/yi-hack #193: TIMEZONE not working and ntp also ( time show 1970 ) so no record.. how to fix it ?

niighthawk, 2017-07-16:
as the title said, my livefeed show a timstamp of 1970 how to fix it ? 

Je parle francais également

niighthawk, 2017-07-16:
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr 20:F4:1B:F5:2E:85  
              inet addr:192.168.0.103  Bcast:192.168.0.255  Mask:255.255.255.0
              inet6 addr: fe80::22f4:1bff:fef5:2e85/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:589 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:151831 (148.2 KiB)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server 'temps.nrc.ca
':
Previous datetime is Thu Jan  1 00:00:38 UTC 1970
Done
New datetime is Thu Jan  1 00:00:38 UTC 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1301 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1328 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1496 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  680 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  816 root         0 SW   [RtmpTimerTask]
  817 root         0 SW   [RtmpMlmeTask]
  818 root         0 SW   [RtmpCmdQTask]
  819 root         0 SW   [RtmpWscTask]
  949 root         0 SW   [hidog]
  987 root      1164 S    ./peripheral
  988 root      1144 S <  ./dispatch
  989 root      1060 S    ./exnet
 1147 root      3136 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1301 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1328 root      1408 S    ./serverM 80
 1347 root     14728 S    ./mp4record 60
 1376 root      1716 S    /usr/sbin/crond -b
 1389 root       904 S    /home/led_ctl -boff -bon 
 1491 root     97484 S    ./rmm
 1496 root     30892 S N  ./rtspsvrM
 1587 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    868.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    920.0K  90% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                 29.9G      9.8M     29.9G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 29.9G      9.8M     29.9G   0% /tmp/hd1/test/http/record

tonyrulez, 2017-07-28:
I have the same issue, I have to manually adjust the time after every reboot. `date --help` will tell you how you can do it. Spoiler: `date -s 2017-07-08 22:57:15`

RiRomain, 2017-08-18:
I have the same issue, small precision, you need to put quote around the date:
`date -s "2017-08-18 14:14:15"`

dgncsk, 2017-10-01:
my all three yi cameras shows 1970s as time stamp too. any comments?

Clem-, 2017-10-10:
Hi!
I found out that the problem is related to a bad DNS configuration.
You need to put a correct DNS server after the variable **NAMESERVER** of the _yi-hack.cfg_ file.
You can for instance put `NAMESERVER=8.8.8.8` to connect to Google's DNS.

The reason behind this is that the npd server cannot be resolve/reached and the correct timestamp is not updated.

Let me know if it works for you ;)

jjongyeonglee, 2018-04-30:
I wrote name server 8.8.8.8
Then type the command '$ reboot'
And
Completely remove power line and reconnect
It works !!

LOLCHERT, 2018-08-29:
Just use a numberic ip adress of ntp server

 You'll find the servers at time.google.com – which resolves to 216.239.35.0, a rather less pretty IP address than the 8.8.8.8 and 8.8.4.4 Google uses for its public domain-name system (DNS) servers. There's also time2.google.com at 216.239.35.4, time3. at .8 and time4. at .12.

oleg-sh, 2018-11-25:
> Just use a numberic ip adress of ntp server

I tried that and now it updates my time (yay, thank you!) but doesn't set the time zone correctly. I tried a few different time zones from this [list](http://svn.fonosfera.org/fon-ng/trunk/luci/modules/admin-fon/root/etc/timezones.db) but my camera always sets the GMT (0) time. 
Any ideas?

niighthawk, 2019-01-06:
In telnet type this
**echo "GMT+5" > /etc/TZ**
it work for my timezone just change the number until you get the good time.

oleg-sh, 2019-01-06:
Thanks. I figured the problem eventually. It was custom DNS (I think OpenDNS) set up on my router that caused it.

diegochentw, 2019-05-27:
It seem the only way I can change the date, thanks

zestni, 2020-02-04:
I had the same problems. I installed the DNS of my provider.

SAOPP, 2020-03-15:
Guys any script for this issue? I have this issue on my one camera  utc+0 always.

Clem-, 2020-03-15:
Hi,
I [mentioned a while ago](https://github.com/fritz-smh/yi-hack/issues/193#issuecomment-335563009) that this issue could be solved by specifying a valid NAMESERVER IP address (that is the DNS address). 
What IP address do you have in your configuration file?

SAOPP, 2020-03-16:
Hi @Clem-

Both of my yi cameras use ip address of my router as name server. But few weeks ago one of my camera start to showing utc+0 as a timezone, I'm didn't do any changes in config before this moment... Yesterday evening when I saw the topic about this issue, I trying to set instead of my router address - ip of google, and camera didn't started at all, so I switched back, and set tz manually via telnet, but, I still researching the seolving method of this issue.