fritz-smh/yi-hack #195: Video not working

autobottrixter, 2017-08-05:
I'm able to get my camera connected to the network and I can access it via IP address, but I'm unable to view the video through RTSP. I've attempted it using VLC on multiple computers on the network by right-clicking on the link on the access page and copying the link address, then pasting it into the network stream, but I get an error saying it can't be accessed. I've added an exception for VLC to the Windows Firewall and I've also messed around with RTSP plugins for Chrome and various other things, but no luck. I'm using a DD-WRT router that has a Video tab in the UI and that's also not detecting a camera, though I'm not certain if it should work with this. Any suggestions?

[image]

buZybaZzz, 2017-08-13:
You could check the log file that's created by the script (it's located in the root of your camera) for clues as to what is going on, or you can Telnet into your camera and enter the following command:
`ps | grep rtspsvr | grep -v grep`
If the RTSP server is running, it should show something like this:
` 1393 root     30956 S N  ./rtspsvrM`
If it isn't running, it will show nothing. If it is running, then something in your network is probably blocking it. Most like it isn't running (since you've tried multiple computers). Try to run the service manually by entering:
`cd /home/hd1/test/
./rtspsvrM`
(The 'M' could be an 'I' or a 'K', depending on the firmware that you use.)
You should be able to tell from the output that you'll get whether it's working or not and if it isn't, then hopefully it'll give you a clue as to what's going on. If it is working and you connect to the stream, a lot more logging will appear on the screen. (Indicating that a connection has been made with the stream.)

autobottrixter, 2017-08-28:
I gave that a try but I'm still getting the same error in VLC. I'm attaching a log of the errors that popped up in Telnet when VLC tried to connect. This time I also tried doing it from my Kubuntu install and through a different router with no luck. I appreciate the help so far!

[Camera Telnet Error Log.txt](https://github.com/fritz-smh/yi-hack/files/1254794/Camera.Telnet.Error.Log.txt)

flyrodkid, 2017-11-03:
I'm having the same issue and found based on buZybaZzz's suggestions that my rtsp server was not running. When I start it manually, it appears to start up correctly. When I attempt to connect to it using VLC,  I get a massively repeated error of "[597d1d4e][RTSP] handleCmd_DESCRIBE: get new sps fail" followed by
[597d1d4e][RTSP] DispatchPrintMsg: Print msg length is -4=<0
terminate called after throwing an instance of 'std::bad_alloc'
   wat():  std::bad_alloc
Aborted

I have verified that I have firmware 'M' and started the rtspsvrM

Thoughts?

Clem-, 2017-11-04:
Did you change something in any file?
Maybe you should print us out the complete camera log ?