fritz-smh/yi-hack #196: Quitar marca de tiempo definitvamente

yokero, 2017-08-12:
Hola, he leido todos los hilos y comentarios acerca de este tema pero no logro sacar una solución, por favor necesito que alguien me explique qué pasos hay que realizar para eliminar la marca de agua del video en directo. Quiero añadir la cámara a un servidor NAS de Synology y no quiero que me muestre la cámara la horas (que por cierto la hora que muestra es de 2 horas menos de lo que le corresponde... ejemplo si son las 12.00 pm la camara muestra las 10.00 pm ... he provado a cambiar la zona horaria, el servidor ntp, todo!! pero nada funciona) 

Neceist soluciones por favor, agradecería sus comentarios, es muy urgente. Gracias.

Post data, si alguien ha conseguido poner bien la hora, que me explique cómo.

Por favor, no me remitan a otros hilos, he revisado todo y no he encontrado solucion, quiero respuestas de ahora.

Atentamente, Yokero.

TheRealGitCub, 2017-09-06:
Translation: 

Hello, I have read all of the threads and comments about this issue, but I haven't found a solution. Can someone please explain how I can remove the watermark from the live video? I want to add the camera to a Synergy NAS server and I don't want to see the time (or hours). (the time is two hours off, for example at 12:00 pm the camera shows 10:00 on. I have tried to change the time zone, the NTP server, everything!! But nothing works)

I need a solution please, I would appreciate your comments as it is very urgent. Thanks. 

Please, don't refer me to other threads, I have reviewed them all and I wasn't able to find a solution.

dmolner, 2017-12-18:
Supongo que has probado a borrar el contenido de unos archivos que se indica en un hilo bastante viejo, no?

jafocol, 2021-09-05:
Buenas noches, no he podido cuadrar la hora, por favor me indicas el hilo o que archivos se borran, muchas gracias!!.