fritz-smh/yi-hack #198: "Waiting to connect"

lastb0isct, 2017-08-30:


acoushawk, 2017-09-01:
I'm also having the same problem. I have done a few of these cameras in the past and have not had an issue but this seems to not be working now. Have they patched this?

lastb0isct, 2017-09-01:
Looks like its an issue with the firmware on newer cameras.  This hack doesn't work on the newer firmware.  We have to wait for it to be updated or try a different hack, sad thing is that the other hacks don't support RTSP which I require for my software =(

tube0013, 2017-09-02:
Ugh, I'm just running into this now. I have 1 of these I bought a few months back that the hack works fine on, now the new one I can't get working.

kruschman, 2017-09-06:
Does anyone know which firmware version the newer ones have? Is it possible to downgrade?