fritz-smh/yi-hack #199: Yi Home Camera won't pair after firmwarware update

Deadteeth, 2017-09-23:
I am new here - yet I found this thread and I figured if you guys can't help me - no one can!
I just received a Yi Home Camera (version1) - and am having issues after the firmware update.

It is a International  Edition   (27USY5SIE0)  model YHS-113-UR
When I first started the camera (out of box) it fired up   scanned the QR - connected to wifi and paired.
Yahoo.   BUT - then the app told me to upgrade the firmware - so I did  (through the app).
Now the camera will scan the QR code - connect to WIFI - but will Timeout (never pairs).

I reset the camera - nope.
I spend hours sending Yi the logs and passing details to them - nope
The firmware that appears to be on it is the same current version as on their site (1.8.7.OD_201708091510(USA) from reading it in the log file.  

Sine this appears to be **brand new firmware** - I assume it might be buggy.
I have tried to read the logs yet can't find a way to delineate them so it is a mess to read. 

I asked Yi to give me the firmware that would have initially been on the camera  but they won't do it. 
I have sent them  screen shots / logs files and tested with other routers  - yet they offer no help or ideas.

Does anyone have a firmware version that was a preamble to this one that I might try ?  I assume this would be the logical next step since the camera worked before the FW upgrade.
Right now the camera is a very shiny and cool paperweight....

Thanks in advance

dt  (Canada)

uraharakisuke12, 2017-09-23:
I have same problem. This firmware unbrick my cam https://github.com/hizel/yi_dump . Just copy home_y18 and rootfs_y18 on root sd card. This work for me.

Deadteeth, 2017-09-23:
Thank you so much for the quick reply.  I can't seem to fid a way to get the files downloaded and copy to my PC so I will work away at that.  (hoping you may have some suggestions).   I also found that by searching around for firmware on this site that there are some older firmware pages so if I can't get the recovery files that you mentioned to download - I might try some of these.

Thanks so much for the reply.

dt

Deadteeth, 2017-09-23:
Update:
THANKYOU THANKYOU THANKYOU!!!!!
I got the files downloaded and it worked on the first try!
Pairing is successful!

You solved this for me in 1 post - whereas Yi support would not help me at all.

Thanks again - this is fantastic.

dt

notyouraveragehaker, 2018-01-09:
Ok Im having a similar issue Deadteeth. Out of the box i plug it in and go through the command prompts to set up the camera and connect to WiFi and while its connecting to wifi is says connection successful then my phone says pairing may take 1 to 2 minutes after about a minute the camera says pairing has timed out.

Im about to pull my hear out any help would be greatly appreciated!!!

Deadteeth, 2018-01-12:
uraharakisuke12   provided the link  https://github.com/hizel/yi_dump 
(see above)

Just copy home_y18 and rootfs_y18 on root of an SD  card - out it in the camera and reboot the camera.  The LED on the front will blink slowly while the firware loads.   This worked for me.

Please let me know if this worked.
d

PS:  If it does work and the camera prompts you to downnload new firmware - Don'T  - or it will override the process above.

closed 2018-01-12 by Deadteeth

notyouraveragehaker, 2018-01-12:
This worked then you so much Deadteeth!!! I really appreciate it I was killing me!!! Thank you!!!!

Closed #199<https://github.com/fritz-smh/yi-hack/issues/199>.

Deadteeth, 2018-01-12:
Glad it worked - I know how frustrating it was for me to know that the firmware update broke it.  I spent more time fretting over this $27 camera than anything I have ever bought.  It was user uraharakisuke12  who saved the day for us. 

Keep in mind that you will periodically get prompted to update the firmware.  the prompt will randomly show up on all devices that are synced to the camera to get alerts.    If any one ever selects "Update Firmware"  instead of cancelling - the device will get smoked again.   At least you know how to get it back.   Who knows - maybe the latest version of firmware would actually work - but I'm to chicken to give it a try.  The camera seems to work OK as is - so I'm leaving it alone.

The camera is pretty cool.   Have fun with it!
dt

JessPorquez, 2018-03-25:
Hi there,
I'm having issues with the Yi home camera 1080p: even after manually selecting my 2.4 gHz connection and separate attempts using the (manually) updated and then the firmware provided above I am still unable to connect the camera to the app. It is a brand new device, and I have yet to connect the camera successfully. Perhaps I am just missing some key step altogether? I am attempting to pair the camera to an older iPhone (5s), but according the the app details, I am operating at an acceptable iOS (at least 8.0; iPhone = 10-something). Does anyone have any other suggestions?

Thanks!