fritz-smh/yi-hack #203: Can't get this to work at all.

8bitbirdie, 2017-10-05:
I have the Yi home camera and I want to use it for Motion Eye OS. 

I formatted a 8gb sd card for fat32, copied the contents of the sd file onto it, changed the relevant information, but it seems to completely ignore the SD card and goes straight into it's normal boot. Can't access it through IP. 

Does anyone have any ideas as to what I'm missing? Walkthrough on youtube to confirm I'm not missing some blaringly obvious step? Anything?

Clem-, 2017-10-10:
Hi,
Just to make sure, did you copy the CONTENT of the directory `yi-hack/sd` and not the directory itself? You must have a file "home" in the root of SD Card.
Did you hear any sounds/voice from the camera during boot?

Clément

kruschman, 2017-10-11:
This hack does not work with current hardware versions. Unless you camera is a year old, your SOL.