fritz-smh/yi-hack #204: Blink Yellow light issue again

brookeschris, 2017-10-11:
so I done my best to follow the guide but being new to all this I guessing there is a basic error / mistake in something I done.

files in top level of Sd card 

reviewing the log.txt - extract shown below I can see

line 4: failed to parse ssid '“everythingrosey”'.

but I have no idea why - tried redoing "quotes" - still get same issue.

Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=255  (0 is ok)
Wifi configuration answer: Successfully initialized wpa_supplicant
Line 4: failed to parse ssid '“everythingrosey”'.
Line 4: failed to parse ssid '“everythingrosey”'.
Line 11: failed to parse network block.
Failed to read or parse configuration '/home/wpa_supplicant.conf'.
Do network configuration 1/2 (ip and gateway)
Done

Here the wpa_supplicant.conf file

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid=“everythingrosey”
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="GCHQwatchingyou"
}

can someone guide an newbie on what I've done wrong please

Clem-, 2017-10-11:
It seems that your quotes are incorrect. Try copy/paste those of your PSK variable in order to have : 
`"everythingrosey"`

brookeschris, 2017-10-11:
thank you clem - urns out using textedit on my macos was doing something strange to the quotes marks. 
Went old school and used vi on the file and it sorted it - 

now  to configure vlc to watch stream etc  

great hack and many thanks for your help and prompt to revisit checking quotes ....

closed 2017-10-11 by brookeschris