fritz-smh/yi-hack #206: Cannot connecto to camera through ip/http

webberd1wei, 2017-11-19:
I am new to this repo. I configured the sd card and put it in Yi home camera. The LED became solid blue. But when I tried to connect to the camera through http://mycameraip/ on my browser, I cannot access it. My firmware is 1.8.7.0D_201708091510. I read a couple of open issues regarding firmware upgrades, does this hack work with my version of firmware? If not directly, are there workarounds?

Clem-, 2017-11-19:
Hi,
When you say `mycameraip`, is that the *local* ip address?

webberd1wei, 2017-11-20:
@Clem- yep, 192.168.1.X.

Clem-, 2017-11-25:
Are your sure that your router's assigning this very ip address to your camera?

webberd1wei, 2017-11-27:
Yes I am sure. I assigned static ip to it. Does this hack work with the version of firmware I have?

Clem-, 2017-11-27:
Im personally on 1.8.5.1M_201512011815
Can’t say much about your version. 
Did you find errors in your log.txt (in `test` dir)?

webberd1wei, 2017-11-28:
@Clem- Thanks for your patience. I did not find log.txt in `test` dir. Is that maybe the problem?

Clem-, 2017-11-28:
That could explain one thing : the hack didn’t boot because of your firmware version. 

If you really want to benefit this hack, you should downgrade to 1.8.5