fritz-smh/yi-hack #208: Wifi connection fail

Manu68, 2017-12-09:
I followed the tutorial, but when it comes to the wifi connection, the blue LED blinks, and 10 sec after, switches to orange. This sequence comes back again and again.

I change the configuration file to WPA2. Could it be the problem?

Thanx
M

Manu68, 2017-12-09:
I changed WPA2 to WPA and in the log file, I can find the problem....

Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:18 errors:0 dropped:0 overruns:0 frame:0
              TX packets:18 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:1452 (1.4 KiB)  TX bytes:1452 (1.4 KiB)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:53:A4:E4  
              inet6 addr: fe80::b2d5:9dff:fe53:a4e4/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:0 (0.0 B)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 00:00:22 GMT 1970
Done
New datetime is Thu Jan  1 00:00:22 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1261 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1288 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
Start telnet server...
Start telnet server...
Start telnet server...

Clem-, 2018-03-28:
Hi @Manu68,
Did you fill [yi-hack.cfg Network configuration](https://github.com/fritz-smh/yi-hack/blob/7e4145cdb675ea20194cc4256c8a01de556a527d/sd/test/yi-hack.cfg#L4) ?

haicoi311, 2018-10-16:
I have the same problem. the yellow light keep blinking