fritz-smh/yi-hack #211: Implement motion url detection

fufurax, 2018-03-15:
Hi,

I want to implement the GET motion url in SH scipts. Is it possible ?

I've try : 
curl -i -su login:pass -H "Accept: application/json" -X GET  http://192.168.1.110/motion

but the return is not good

HTTP/1.1 200 Ok
Server:
Date: Thu, 15 Mar 2018 12:05:11 GMT
Content-Type: text/html; charset=UTF-8
Content-Length: 1
Last-Modified: Tue, 01 Jan 1980 00:00:00 GMT
Connection: close

Sorry, i'm really not specialist.

Thanks

Clem-, 2018-03-28:
Hi @fufurax,
Are you sure  http://192.168.1.110/motion is returning something when calling it from your browser?
Maybe there's simply nothing more to return because no new motion had been detected.

fufurax, 2018-03-28:
Hi @Clem- 

Nothing is return when call it from the browser. But it's strange because there is new mp4 files generated in record files

Clem-, 2018-03-28:
The script only detects newer motion detection (for the last minute) : `-mmin -1 `

You may want to update your code with this [pull request](https://github.com/fritz-smh/yi-hack/pull/153/commits/ceffb8b8c3a733eb2b853a340ccdd0208d13473d) that intends to improve this motion detection script.

fufurax, 2018-03-28:
@Clem- 

In fact I want to use the motion detection of the camera to trigger events to Domoticz. 

I can probably write a script in Domoticz to scan every 30 secondes to detect if motion has detected something but I think it's not the good solution. 

Have you implement something like that arroud Yi cam ?

Clem-, 2018-03-28:
Great idea! I'm also using Domoticz but never take the chance to develop that kind of feature.

You could simply follow my previous link to the pull request and make a GET request to Domoticz instead of POST for a notification.
See also [Domoticz API](https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s) in order to know which request to make depending on the event you want to trigger.

If you need further help with that feel free to ask.

gitrc, 2019-04-27:
Hello.  The kernel in this firmware supports inotify.  I was able to compile inotifywait and use it to watch for new recordings and trigger an http endpoint using curl.  Maybe you will find this helpful.

#!/bin/sh
#
# poor mans motion sensor
#
#

/tmp/hd1/inotifywait -mrq -e CREATE --format %w%f /tmp/hd1/record_sub | while read FILE
do
case "$FILE" in
  *.tmp*)
     # Do stuff
     /home/curl -s "http://my/endpoint?subject=room&message=motion" > /dev/null
     ;;
esac
done