fritz-smh/yi-hack #217: YI YHS-113-IR unable to flash

RedFoxy, 2018-11-29:
Hi!
Just I got my YI 720 https://amzn.to/2DPaMVr, on the back it say model YHS-113-IR, si I come here and download all, than I get my 4Gb Sd card and format it in FAT32 than put in the root "home" file and "test" directory, after that I edit wpa_supplicant.conf and yi-hack.cfg with my wifi information and a static ip, than I put it on my YI camera and plug the power cord, it start but the yellow led blinks and say "waiting to connect", so I tried to check if the camera goes so I followed the wizard configuration to connect to my wifi, it goes very well, than I tried again with the SD card but nothing.

does that fix still goes or am I doing something of wronge?

dannybres, 2018-12-13:
I have the same camera and have followed the same steps. Hoping for some advice too.

dannybres, 2018-12-13:
> Hi!
> Just I got my YI 720 https://amzn.to/2DPaMVr, on the back it say model YHS-113-IR, si I come here and download all, than I get my 4Gb Sd card and format it in FAT32 than put in the root "home" [...]

Did you many any progress, RedFoxxy?

RedFoxy, 2018-12-13:
Just I find the v3 version of that mod https://github.com/shadow-1/yi-hack-v3 but after I use it I've the web interface but no RTSP :|

dannybres, 2018-12-13:
I did the same. 😔